coolsnowwolf/lede #5056: docker安装失败

tengyuankoo, 2020-07-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：硬盘格式化新装固件，OpenWrt R9 Docker 小白无脑配置教程执行，DOCKER安装失败

### (2) 路由器型号和固件版本

A：X86 

### (3) 详细日志
TTYD执行：
A：root@OpenWrt:/# /etc/docker-web
Unable to find image 'portainer/portainer:latest' locally
latest: Pulling from portainer/portainer
d1e017099d17: Pull complete 
b8084bf83dcf: Pull complete 
Digest: sha256:55c7614b1ad61eabc27214299c42d41bb49e5ef78238c0e5783031f041499284
Status: Downloaded newer image for portainer/portainer:latest
74ed956cf72c5960840e1bf0a67cf18e19aeb408acaf639a7348d2b6508ba044
docker: Error response from daemon: OCI runtime create failed: /var/lib/docker/overlay2/0c9760ea87b73f5a039bbe84badf051d61c42d4f979dd12828136576ad5e336f/merged is not an absolute path or is a symlink: unknown.

系统日志：
Thu Jul  9 18:44:53 2020 daemon.err dockerd[4253]: time="2020-07-09T10:44:53.111667617Z" level=warning msg="Error getting v2 registry: Get \"https://registry-1.docker.io/v2/\": net/http: TLS handshake timeout"
Thu Jul  9 18:44:53 2020 daemon.err dockerd[4253]: time="2020-07-09T10:44:53.111748889Z" level=info msg="Attempting next endpoint for pull after error: Get \"https://registry-1.docker.io/v2/\": net/http: TLS handshake timeout"
Thu Jul  9 18:44:53 2020 daemon.err dockerd[4253]: time="2020-07-09T10:44:53.111856271Z" level=error msg="Handler for POST /v1.40/images/create returned error: Get \"https://registry-1.docker.io/v2/\": net/http: TLS handshake timeout"
Thu Jul  9 18:45:24 2020 daemon.err dockerd[4253]: time="2020-07-09T10:45:24.669932336Z" level=info msg="Attempting next endpoint for pull after error: Get \"https://registry-1.docker.io/v2/portainer/portainer/manifests/latest\": net/http: TLS handshake timeout"
Thu Jul  9 18:45:24 2020 daemon.err dockerd[4253]: time="2020-07-09T10:45:24.670066165Z" level=error msg="Handler for POST /v1.40/images/create returned error: Get \"https://registry-1.docker.io/v2/portainer/portainer/manifests/latest\": net/http: TLS handshake timeout"
Thu Jul  9 18:45:50 2020 daemon.err dockerd[4253]: time="2020-07-09T10:45:50.290128944Z" level=warning msg="Error getting v2 registry: Get \"https://registry-1.docker.io/v2/\": net/http: TLS handshake timeout"
Thu Jul  9 18:45:50 2020 daemon.err dockerd[4253]: time="2020-07-09T10:45:50.290190530Z" level=info msg="Attempting next endpoint for pull after error: Get \"https://registry-1.docker.io/v2/\": net/http: TLS handshake timeout"
Thu Jul  9 18:45:50 2020 daemon.err dockerd[4253]: time="2020-07-09T10:45:50.290258701Z" level=error msg="Handler for POST /v1.40/images/create returned error: Get \"https://registry-1.docker.io/v2/\": net/http: TLS handshake timeout"
Thu Jul  9 18:46:01 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jul  9 18:46:01 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jul  9 18:46:04 2020 daemon.info dnsmasq[11381]: Connected to system UBus
Thu Jul  9 18:46:04 2020 cron.err crond[11415]: crond (busybox 1.31.1) started, log level 5
Thu Jul  9 18:46:05 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jul  9 18:46:05 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jul  9 18:46:08 2020 daemon.info dnsmasq[11787]: Connected to system UBus
Thu Jul  9 18:46:46 2020 kern.info kernel: [  322.753780] docker0: port 1(veth3ce8864) entered blocking state
Thu Jul  9 18:46:46 2020 kern.info kernel: [  322.760225] docker0: port 1(veth3ce8864) entered disabled state
Thu Jul  9 18:46:46 2020 kern.info kernel: [  322.766697] device veth3ce8864 entered promiscuous mode
Thu Jul  9 18:46:46 2020 daemon.err dockerd[4253]: time="2020-07-09T10:46:46.504683900Z" level=info msg="No non-localhost DNS nameservers are left in resolv.conf. Using default external servers: [nameserver 8.8.8.8 nameserver 8.8.4.4]"
Thu Jul  9 18:46:46 2020 daemon.err dockerd[4253]: time="2020-07-09T10:46:46.504712020Z" level=info msg="IPv6 enabled; Adding default IPv6 external servers: [nameserver 2001:4860:4860::8888 nameserver 2001:4860:4860::8844]"
Thu Jul  9 18:46:46 2020 daemon.err dockerd[4253]: time="2020-07-09T10:46:46.641872314Z" level=info msg="shim containerd-shim started" address="/containerd-shim/moby/74ed956cf72c5960840e1bf0a67cf18e19aeb408acaf639a7348d2b6508ba044/shim.sock" debug=false pid=12180
Thu Jul  9 18:46:46 2020 daemon.err dockerd[4253]: time="2020-07-09T10:46:46.894405905Z" level=info msg="shim reaped" id=74ed956cf72c5960840e1bf0a67cf18e19aeb408acaf639a7348d2b6508ba044
Thu Jul  9 18:46:46 2020 daemon.err dockerd[4253]: time="2020-07-09T10:46:46.909724494Z" level=error msg="stream copy error: reading from a closed fifo"
Thu Jul  9 18:46:46 2020 daemon.err dockerd[4253]: time="2020-07-09T10:46:46.913714659Z" level=error msg="stream copy error: reading from a closed fifo"
Thu Jul  9 18:46:46 2020 kern.info kernel: [  323.260119] docker0: port 1(veth3ce8864) entered disabled state
Thu Jul  9 18:46:47 2020 kern.info kernel: [  323.267985] device veth3ce8864 left promiscuous mode
Thu Jul  9 18:46:47 2020 kern.info kernel: [  323.273309] docker0: port 1(veth3ce8864) entered disabled state
Thu Jul  9 18:46:47 2020 daemon.err dockerd[4253]: time="2020-07-09T10:46:47.034854207Z" level=error msg="74ed956cf72c5960840e1bf0a67cf18e19aeb408acaf639a7348d2b6508ba044 cleanup: failed to delete container from containerd: no such container"
Thu Jul  9 18:46:47 2020 daemon.err dockerd[4253]: time="2020-07-09T10:46:47.034893174Z" level=error msg="Handler for POST /v1.40/containers/74ed956cf72c5960840e1bf0a67cf18e19aeb408acaf639a7348d2b6508ba044/start returned error: OCI runtime create failed: /var/lib/docker/overlay2/0c9760ea87b73f5a039bbe84badf051d61c42d4f979dd12828136576ad5e336f/merged is not an absolute path or is a symlink: unknown"

keeno1982, 2020-07-09:
我也是，升级到5.4后docker装不了了，同样是这个错误。比较可怕的是执行/etc/docker-init后，竟然自动把我的一块未挂载的硬盘给格掉了，还好只是下载电影用的，要是格掉我的家庭资料盘，那是真的要哭死

zfh135072, 2020-07-09:
同问题

lexin8, 2020-07-09:
4.14内核编译不就行了，非要搞高难度的事情，之前我还尝试官方openwrt编译SSR等第三方插件，多次失败放弃还是用别人修改好的源来得快，毕竟我们不是大神。

chongshengB, 2020-07-10:
进入路由/etc/docker目录，
新建一个daemon.json文件，内容如下：

`{
    "data-root": "/opt/", #这个改成你挂载的docker数据分区目录
    "log-level": "warn"
}`

最后重启docker或者直接重启路由即可

tengyuankoo, 2020-07-10:
> 进入路由/ etc /
> docker 目录，新建一个daemon.json文件，内容如下：
> 
> > { 
> > “ data-root”：“ / opt /”，＃这个改成你挂载的docker数据分区目录
> > “ log-level”：“ warn” 
> > }
> 
> 最后重启docker或者直接重启路由即可

按这样操作
docker-ce容易显示未运行，输入命令错误
root@OpenWrt:/# /etc/docker-web
docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.
See 'docker run --help'.

tengyuankoo, 2020-07-10:
我已切换到4.19，你们试下进入路由/ etc /docker 目录，新建一个daemon.json文件，内容如下：

{
    "data-root": "/opt/docker/",
    "log-level": "warn"
}
最后重启docker或者直接重启路由即可
感谢chongshengB

wushenghn, 2020-07-10:
我 的r2s 5.4运行docker正常,跟docker挂载点没关系

cntink, 2020-07-14:
> 进入路由/etc/docker目录，
> 新建一个daemon.json文件，内容如下：
> 
> > {
> > "data-root": "/opt/", #这个改成你挂载的docker数据分区目录
> > "log-level": "warn"
> > }
> 
> 最后重启docker或者直接重启路由即可

用你的方法终于搞定了！这算是个bug么？

xx10n31y, 2020-07-16:
> 4.14内核编译不就行了，非要搞高难度的事情，之前我还尝试官方openwrt编译SSR等第三方插件，多次失败放弃还是用别人修改好的源来得快，毕竟我们不是大神。

求问怎么编译4.*的内核固件, 现在新的代码貌似都是5.4的内核版本, 也是导致docker安装不了

zerofo, 2020-07-17:
docker pull  报错  Error response from daemon: Get "https://registry-1.docker.io/v2/": net/http: TLS handshake timeout
docker info:
```
Client:
 Debug Mode: false

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 0
 Server Version: 19.03.12
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 7ad184331fa3e55e52b890ea95e65ba581ae3429
 runc version: dc9208a3303feef5b3839f4323d9beb36df0a9dd
 init version: 
 Kernel Version: 5.4.50
 Operating System: OpenWrt SNAPSHOT
 OSType: linux
 Architecture: x86_64
 CPUs: 6
 Total Memory: 11.71GiB
 Name: OpenWrt
 ID: GUCW:PP6A:CTQC:VDRL:JVDU:SZNR:3DQT:E3LR:GP2H:AJPV:LWKU:KUFM
 Docker Root Dir: /mnt/sdb1/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Registry Mirrors:
  https://registry.docker-cn.com/
 Live Restore Enabled: false

WARNING: No swap limit support
WARNING: No cpu cfs quota support
WARNING: No cpu cfs period support

```

281677160, 2020-08-07:
> > 4.14内核编译不就行了，非要搞高难度的事情，之前我还尝试官方openwrt编译SSR等第三方插件，多次失败放弃还是用别人修改好的源来得快，毕竟我们不是大神。
> 
> 求问怎么编译4.*的内核固件, 现在新的代码貌似都是5.4的内核版本, 也是导致docker安装不了

用WinSCP工具直接连接乌班图系统，找到源码，target文件夹里面的linux文件夹再到相对于机型文件夹，比如x86，然后找到Makefile这个文件，修改里面的5.4为4.19就可以了，可以用命令直接改的，我不会本地编译，没了解，我直接学的云编译

云编译直接两行命令就搞定了
sed -i 's/KERNEL_PATCHVER:=5.4/KERNEL_PATCHVER:=4.19/g' ./target/linux/x86/Makefile
sed -i 's/KERNEL_TESTING_PATCHVER:=5.4/KERNEL_TESTING_PATCHVER:=4.19/g' ./target/linux/x86/Makefile

我的云编译地址
https://github.com/281677160/Lede-OpenWrt.git

closed 2021-02-04 by coolsnowwolf