coolsnowwolf/lede #5059: verysync-linux-amd64-v1.4.3.tar.gz 包错误

ptsa, 2020-07-09:
verysync-linux-amd64-v1.4.3.tar.gz
这个包 下载以后打不开 导致编译错误

mjyhj, 2020-07-09:
1.4.4啊

ptsa, 2020-07-10:
编译时候下载的就是这个包

hue715, 2020-07-11:
verysync的Makefile中有获取最新版本的指令，但指令执行失败（ubuntu20.04测试失败），无法下载最新的代码包。
[image]

### 最新版本失败报错：
[image]

EnnawYang, 2020-07-12:
> verysync的Makefile中有获取最新版本的指令，但shell指令执行失败（ubuntu20.04测试失败），无法下载最新的代码包。
> [image]
> 
> ### 最新版本失败报错：
> [image]

sudo apt-get install curl

closed 2021-02-04 by coolsnowwolf