coolsnowwolf/lede #5064: 在使用的节点正常没有问题时，5.4内核SSR+会出现频繁切换节点的情况

willaytoun, 2020-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

A：SSR+在节点正常自动切换检查周期设置为默认的情况下，会频繁切换节点。

### (2) 路由器型号和固件版本

A：X86，OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)

### (3) 详细日志

A：

minlang112, 2020-07-10:
SSR 未出现该情况

coolsnowwolf, 2020-07-11:
关闭 IPv6

willaytoun, 2020-07-11:
好的 感谢解答

willaytoun, 2020-07-12:
关闭了IPv6但是偶尔还是会出现这种情况(两天只出现过一次),在看YouTube的时候节点是好的也肯定不会超过11分钟连不上

2020-07-12 17:26:16 Switch to 香港 网飞# proxy!
------------end------------
2020-07-12 17:26:13 Main node:V2Ray 4.26.0 (OpenWrt) Lean (go1.14.4 linux/amd64) Started!
-----------start-----------
2020-07-12 17:26:09 Another server is avilable, now switching server.
2020-07-12 17:26:07 Current server error, try to switch another server.
------------end------------
2020-07-12 15:34:37 Main node:ShadowsocksR 4 Threads Started!
-----------start-----------

ricksuzade-maker, 2020-07-13:
是的，x64默认编译下，还是会出现无故切换节点的问题。

erenyega, 2020-07-13:
对，无故切换了节点之后主节点没有问题也切不回来了。

------------end------------
2020-07-13 17:08:35 Main node:V2Ray 4.26.0 (OpenWrt) Lean (go1.14.4 linux/amd64) Started!
-----------start-----------
2020-07-13 17:07:29 Main server is NOT avilable.Continue using current server.
2020-07-13 17:07:26 Current server is not default Main server, try to switch back.
2020-07-13 16:56:19 Main server is NOT avilable.Continue using current server.
2020-07-13 16:56:15 Current server is not default Main server, try to switch back.
2020-07-13 16:45:08 Main server is NOT avilable.Continue using current server.
2020-07-13 16:45:05 Current server is not default Main server, try to switch back.
2020-07-13 16:33:58 Main server is NOT avilable.Continue using current server.
2020-07-13 16:33:54 Current server is not default Main server, try to switch back.
2020-07-13 16:22:47 Main server is NOT avilable.Continue using current server.
2020-07-13 16:22:44 Current server is not default Main server, try to switch back.
2020-07-13 16:11:37 Main server is NOT avilable.Continue using current server.
2020-07-13 16:11:33 Current server is not default Main server, try to switch back.
2020-07-13 16:00:26 Main server is NOT avilable.Continue using current server.
2020-07-13 16:00:23 Current server is not default Main server, try to switch back.
2020-07-13 15:49:15 Main server is NOT avilable.Continue using current server.
2020-07-13 15:49:12 Current server is not default Main server, try to switch back.
2020-07-13 15:38:05 Main server is NOT avilable.Continue using current server.
2020-07-13 15:38:02 Current server is not default Main server, try to switch back.
2020-07-13 15:26:54 Main server is NOT avilable.Continue using current server.
2020-07-13 15:26:51 Current server is not default Main server, try to switch back.
2020-07-13 15:15:44 Main server is NOT avilable.Continue using current server.
2020-07-13 15:15:40 Current server is not default Main server, try to switch back.
2020-07-13 15:04:33 Main server is NOT avilable.Continue using current server.
2020-07-13 15:04:30 Current server is not default Main server, try to switch back.
2020-07-13 14:53:23 Main server is NOT avilable.Continue using current server.
2020-07-13 14:53:19 Current server is not default Main server, try to switch back.
2020-07-13 14:42:12 Main server is NOT avilable.Continue using current server.
2020-07-13 14:42:09 Current server is not default Main server, try to switch back.
2020-07-13 14:31:02 Main server is NOT avilable.Continue using current server.
2020-07-13 14:30:58 Current server is not default Main server, try to switch back.
2020-07-13 14:19:51 Main server is NOT avilable.Continue using current server.
2020-07-13 14:19:48 Current server is not default Main server, try to switch back.
2020-07-13 14:08:41 Main server is NOT avilable.Continue using current server.
2020-07-13 14:08:38 Current server is not default Main server, try to switch back.
2020-07-13 13:57:30 Main server is NOT avilable.Continue using current server.
2020-07-13 13:57:27 Current server is not default Main server, try to switch back.
2020-07-13 13:46:20 Main server is NOT avilable.Continue using current server.
2020-07-13 13:46:16 Current server is not default Main server, try to switch back.
2020-07-13 13:35:09 Main server is NOT avilable.Continue using current server.
2020-07-13 13:35:06 Current server is not default Main server, try to switch back.
2020-07-13 13:23:59 Main server is NOT avilable.Continue using current server.
2020-07-13 13:23:56 Current server is not default Main server, try to switch back.
2020-07-13 13:12:48 Main server is NOT avilable.Continue using current server.
2020-07-13 13:12:45 Current server is not default Main server, try to switch back.
2020-07-13 13:01:38 Main server is NOT avilable.Continue using current server.
2020-07-13 13:01:35 Current server is not default Main server, try to switch back.
2020-07-13 12:50:27 Main server is NOT avilable.Continue using current server.
2020-07-13 12:50:24 Current server is not default Main server, try to switch back.
2020-07-13 12:39:17 Main server is NOT avilable.Continue using current server.
2020-07-13 12:39:14 Current server is not default Main server, try to switch back.
2020-07-13 12:28:06 Main server is NOT avilable.Continue using current server.
2020-07-13 12:28:03 Current server is not default Main server, try to switch back.
2020-07-13 12:16:56 Main server is NOT avilable.Continue using current server.
2020-07-13 12:16:53 Current server is not default Main server, try to switch back.
2020-07-13 12:05:46 Main server is NOT avilable.Continue using current server.
2020-07-13 12:05:42 Current server is not default Main server, try to switch back.
2020-07-13 11:54:35 Main server is NOT avilable.Continue using current server.
2020-07-13 11:54:32 Current server is not default Main server, try to switch back.
2020-07-13 11:43:25 Switch to 新加坡11 proxy!
------------end------------
2020-07-13 11:43:21 Main node:V2Ray 4.26.0 (OpenWrt) Lean (go1.14.4 linux/amd64) Started!
-----------start-----------
2020-07-13 11:43:14 Another server is avilable, now switching server.
2020-07-13 11:43:12 Current server error, try to switch another server.
------------end------------
2020-07-13 11:09:30 Main node:V2Ray 4.26.0 (OpenWrt) Lean (go1.14.4 linux/amd64) Started!
-----------start-----------
------------end------------
2020-07-13 10:53:22 Main node:V2Ray 4.26.0 (OpenWrt) Lean (go1.14.4 linux/amd64) Started!
-----------start-----------

willaytoun, 2020-07-16:
几天用下来感觉和节点质量有关系，如果节点速度和稳定性都不错的话很少会出问题，但是节点速度和稳定性一般的话就很容易出现这正情况。

toma77777, 2020-07-18:
我也出现了这个情况，节点是正常的并且可以跑满带宽，延迟稳定在150～160ms。但是就是频繁切换节点，并且提示主节点不可用。

closed 2021-02-04 by coolsnowwolf