coolsnowwolf/lede #5065: nanopiR2S

mapestliu, 2020-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

DanielChen008, 2020-07-12:
编译 R2S 时在 make menuconfig 中怎么选？我怎么没找到对应的处理器

mapestliu, 2020-07-13:
> 编译R2S时在make menuconfig中怎么选？我怎么没找到对应的处理器

1、rockchip
2、rockchip_armv8
3、friendlyarm_nanopi-r2s

DanielChen008, 2020-07-14:
> > 编译R2S时在make menuconfig中怎么选？我怎么没找到对应的处理器
> 
> 1、rockchip
> 2、rockchip_armv8
> 3、friendlyarm_nanopi-r2s

感谢！最近刚搞了两台R2S，这小玩意挺不错，一个放公司电脑边，一个出差随身带。

joesdu, 2020-07-28:
> > 编译R2S时在make menuconfig中怎么选？我怎么没找到对应的处理器
> 
> 1、rockchip
> 2、rockchip_armv8
> 3、friendlyarm_nanopi-r2s

这个pi性能怎么样哦，平时温度和CPU占用率一般在多少？

DanielChen008, 2020-07-29:
> > > 编译R2S时在make menuconfig中怎么选？我怎么没找到对应的处理器
> > 
> > 
> > 1、rockchip
> > 2、rockchip_armv8
> > 3、friendlyarm_nanopi-r2s
> 
> 这个pi性能怎么样哦，平时温度和CPU占用率一般在多少？

1：温度较高，不上散热感觉随时可以到70°；
2：性能适合500M以下宽带接入的用户，我用R2S 在千兆宽带下最多只能跑出560，我不知道其他人怎么跑的，有人说用 Friendly 官方固件可以跑出800-900；
3：CPU占用率倒不高，我在AES下跑4KHDR 60FPS的YouTuBe视频时也才15%不到；
4：优点可能就是小巧了吧，可以塞进弱电箱；
5：家用软路由还是推荐x86一步到位吧，比如我用了3年的 J1900稳的一批，或者后来更火的 3865U;

joesdu, 2020-07-29:
> 1：温度较高，不上散热感觉随时可以到70°；
> 2：性能适合500M以下宽带接入的用户，我用R2S 在千兆宽带下最多只能跑出560，我不知道其他人怎么跑的，有人说用 Friendly 官方固件可以跑出800-900；
> 3：CPU占用率倒不高，我在AES下跑4KHDR 60FPS的YouTuBe视频时也才15%不到；
> 4：优点可能就是小巧了吧，可以塞进弱电箱； [...]

我目前用的D525,J1900的我用来当作测试服务器了,这个我昨晚也买了一个,主要是小巧.觉得还是挺不错的,加个散热倒是没啥,淘宝几块钱就搞定了.若是这个比较稳的话.还是挺不错,想把我那D525出了.

joesdu, 2020-07-29:
[image]
编译出来了两个文件,用哪个比较好哦?这两个有啥区别吗?

joesdu, 2020-10-04:
最新代码名称好像有了，温度好像还是不行

closed 2021-02-04 by coolsnowwolf