coolsnowwolf/lede #5066: 新代码的无线功率不能手动设置了

Crazy365, 2020-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

使用新代码（1eb3eac387abd6b0a9148f4d2139a2bee6543a49）编译了K3固件，发现无线功率手动设置失效，不管选择哪一档，传输功率一直都显示为31 dBm。

### (1) 具体问题

A：使用新代码（1eb3eac387abd6b0a9148f4d2139a2bee6543a49）编译了K3固件，发现无线功率手动设置失效，不管选择哪一档，传输功率一直都显示为31 dBm。

### (2) 路由器型号和固件版本

A：K3，commit 1eb3eac387abd6b0a9148f4d2139a2bee6543a49

### (3) 详细日志

A：

Aes64X, 2020-07-10:
K3用Openwrt 啥时候能手动设置功率了？？？

coolsnowwolf, 2020-07-11:
[image]

Crazy365, 2020-07-11:
@coolsnowwolf 我设置“无线电功率”为20 dBm，但上面的“传输功率”仍然为31 dBm。

[image]

closed 2020-07-12 by coolsnowwolf