coolsnowwolf/lede #5067: 今天pull了最新的源码发现make menuconfig菜单里7621下面没有K2P和RM2100了

loonyd, 2020-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新的源码make menuconfig选择型号，发现没有K2P和rm2100了

### (2) 路由器型号和固件版本

A：K2P，RM2100

### (3) 详细日志

A：

wangfeng35, 2020-07-10:
+1 同消失了

forever2019, 2020-07-10:
xy-c5 也没了

wangfeng35, 2020-07-10:
mt7621里仅剩下以下设备型号：
# CONFIG_TARGET_ramips_mt7621_DEVICE_afoundry_ew1200 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_alfa-network_quad-e4g is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_asus_rt-ac57u is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_asus_rt-ac65p is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_asus_rt-ac85p is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_asiarf_ap7621-001 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_asiarf_ap7621-nv1 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_buffalo_wsr-1166dhp is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_buffalo_wsr-2533dhpl is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_buffalo_wsr-600dhp is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_dlink_dir-860l-b1 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_dlink_dir-878-a1 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_dlink_dir-882-a1 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_elecom_wrc-1167ghbk2-s is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_elecom_wrc-1900gst is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_elecom_wrc-2533gst is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_elecom_wrc-2533gst2 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_edimax_rg21s is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_edimax_ra21s is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_firefly_firewrt is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_gehua_ghl-r-001 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_gnubee_gb-pc1 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_gnubee_gb-pc2 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_hiwifi_hc5962 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_iodata_wn-ax1167gr is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_iodata_wn-ax1167gr2 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_iodata_wn-ax2033gr is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_iodata_wn-dx1167r is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_iodata_wn-gx300gr is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_iodata_wnpr2600g is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_jcg_jhr-ac876m is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_jdcloud_re-sp-01b is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_linksys_ea7500-v2 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_linksys_re6500 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_mediatek_mt7621-eval-board is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_mediatek_ap-mt7621a-v60 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_mikrotik_routerboard-750gr3 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_mikrotik_routerboard-m11g is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_mikrotik_routerboard-m33g is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_lenovo_newifi-d1 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_pbr-m1 is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_iptime_a6ns-m is not set
# CONFIG_TARGET_ramips_mt7621_DEVICE_iptime_a8004t is not set

wangfeng35, 2020-07-10:
https://github.com/coolsnowwolf/lede/blob/15b5fbbc0aa532fa68e8529f1d6ad6fa40d96d84/target/linux/ramips/image/mt7621.mk
596行末少了 \ 符号 补充后可以正常出现设备型号了

loonyd, 2020-07-10:
群里有人找到原因了，看下图，少了一个 \ [image]

closed 2020-07-10 by loonyd