coolsnowwolf/lede #5074: ext4硬盘下未知原因transmission频繁重启

evangelion015, 2020-07-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5月9号编译版本挂载ntfs格式硬盘使用tr一直没有大问题，7月12编译新版有ntfs权限问题tr无法更改数据缓存
于是将硬盘改为ext4，之后下载一会就会无提示重启tr，重新格式化无解，更换同为ext4格式的硬盘也出现相同情况
更换原5月9号编译版本也出现相同情况，恩山论坛下载的固件也相同情况，都是lean源码编译版本

### (2) 路由器型号和固件版本

A：树莓派4B，R20，内核4.19&5.4

### (3) 详细日志

A：Sun Jul 12 00:21:03 2020 daemon.err transmission-daemon[3826]: [2020-07-12 00:21:03.962] Couldn't connect socket 149 to 2405:6585:2800:2200:9cab:ae8b:f1c5:14ce, port 44171 (errno 13 - Permission denied) (net.c:290)
Sun Jul 12 00:21:05 2020 daemon.err transmission-daemon[3826]: [2020-07-12 00:21:05.961] Couldn't connect socket 215 to 2408:210:a7a1:8000:40be:8408:8ad3:2e3, port 8999 (errno 13 - Permission denied) (net.c:290)
Sun Jul 12 00:21:05 2020 daemon.err transmission-daemon[3826]: [2020-07-12 00:21:05.961] Couldn't connect socket 215 to 2408:210:a7a1:8000:40be:8408:8ad3:2e3, port 8999 (errno 13 - Permission denied) (net.c:290)

closed 2020-07-11 by evangelion015

Kisoul, 2021-09-06:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [x]  没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：5月9号编译版本挂载ntfs格式硬盘使用tr一直没有大问题，7月12编译新版有ntfs权限问题tr无法更改数据缓存 [...]

问一下你解决了吗？

EYW-015, 2021-09-06:
> 问一下你解决了吗？

换qbittorrent了