coolsnowwolf/lede #5075: 最新版本的编译错误，求助指点

tufu9441, 2020-07-11:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：全新Pull了最新版本的代码后，打开了hello world，之后完全按照步骤操作，Applications中保持默认选项并加选了frpc和几个酸酸乳的子项。单线程编译了几次都没成功，停留在下方的错误提示。

### (2) 路由器型号和固件版本

A：Newifi D2

### (3) 详细日志

A：

```
make[3]: Leaving directory '/home/whm/lede/package/lean/dns2socks'
time: package/lean/dns2socks/compile#0.03#0.02#0.24
make[3]: Entering directory '/home/whm/lede/package/lean/microsocks'
touch /home/whm/lede/build_dir/target-mipsel_24kc_musl/microsocks/.prepared_075897c08bd7e08758d9c5e96d504d84_6664517399ebbbc92a37c5bb081b5c53_check
. /home/whm/lede/include/shell.sh; gzip -dc /home/whm/lede/dl/microsocks.1.0.1.tar.gz | tar -C /home/whm/lede/build_dir/target-mipsel_24kc_musl/microsocks/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [Makefile:42: /home/whm/lede/build_dir/target-mipsel_24kc_musl/microsocks/.prepared_075897c08bd7e08758d9c5e96d504d84_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/whm/lede/package/lean/microsocks'
time: package/lean/microsocks/compile#0.03#0.01#0.13
make[2]: *** [package/Makefile:113: package/lean/microsocks/compile] Error 2
make[2]: Leaving directory '/home/whm/lede'
make[1]: *** [package/Makefile:107: /home/whm/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/whm/lede'
make: *** [/home/whm/lede/include/toplevel.mk:224: world] Error 2
```
[image]

hue715, 2020-07-11:
microsocks.1.0.1.tar.gz文件下载失败了

EnnawYang, 2020-07-12:
错误的包删了重新下啊

minlang112, 2020-07-12:
删除 /home/whm/lede/dl/microsocks.1.0.1.tar.gz 这个文件，重新尝试编译

closed 2020-07-12 by coolsnowwolf