coolsnowwolf/lede #5078: 编译错误 antfs-mount wsdd2 proto-bonding

kktt007, 2020-07-12:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) X64
A：

### (3) 详细日志

A： * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 * 	antfs-mount
 * opkg_install_cmd: Cannot install package automount.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
 * 	wsdd2
 * opkg_install_cmd: Cannot install package autosamba.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-proto-bonding:
 * 	proto-bonding
 * opkg_install_cmd: Cannot install package luci-proto-bonding.

已经执行过以下步骤:
bin文件删除
./scripts/feeds clean
rm -rf ./tmp
 rm -rf .config 
make clean 
然后重新git pull,编译依旧出错

kktt007, 2020-07-12:
使用make distclean编译通过,但是
这样每次未免太效率低了了,有更小成本的解决方案吗.感谢

minlang112, 2020-07-13:
使用make distclean 编译通过，下次应该就不会再出现这种报错了

closed 2020-07-13 by coolsnowwolf