coolsnowwolf/lede #5091: 外网服务器不稳定会影响国内网站也不稳定的情况

aaron021512, 2020-07-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
目前配置只要外网V2RAY服务器不稳定的情况下，会同时连带国内网站的访问也不稳定，甚至钉钉之类的软件都会断链...

想问下SSR Plus+的DNS解析方式，PDNSD具体是怎么运行的呢？ 因为电脑上的NDS全部指向到了路由192.168.3.1，我不确定的就是PDNSD是否会将所有国外、国内的网址都走8.8.4.4解析 （我知道理论上是GFW列表才走），但是没有找到国内DNS的地址设置，所以想询问下，会是什么问题？

[image]
[image]

A：

### (2) 路由器型号和固件版本
路由器：网件R8000
固件版本：OpenWrt R9.10.1
A：

### (3) 详细日志
N/A
A：

coolsnowwolf, 2020-07-13:
DNS设置错误

aaron021512, 2020-07-13:
> DNS设置错误

多谢老大回答，请问有推荐的教程指导下么？

aaron021512, 2020-07-14:
[image]
Lan口的DNS也没有特别设置，内网的电脑DNS都是指向到路由器的。

但是我发现一个奇怪的事情，我这边本机电脑改了NDS到114.114.114.114，刷新本机DNS缓存，发现依然是可以上外网的，我有点不理解了。。求帮助。。

jianyun8023, 2020-07-14:
网络 -> dhcp -> 里面有设置上游dns的地方。把国内dns设置进去就行。

aaron021512, 2020-07-14:
> 网络 -> dhcp -> 里面有设置上游dns的地方。把国内dns设置进去就行。

[image]

谢谢大佬，我这边用的dnsmasq，所以应该是这边设置好了上游DNS服务器就可以了是吗？我发现之前已经设置过了。。。是不是这里也有顺序上的区别？8.8.8.8应该往下放？

Nats-ji, 2020-07-15:
> > 网络 -> dhcp -> 里面有设置上游dns的地方。把国内dns设置进去就行。
> 
> [image]
> 
> 谢谢大佬，我这边用的dnsmasq，所以应该是这边设置好了上游DNS服务器就可以了是吗？我发现之前已经设置过了。。。是不是这里也有顺序上的区别？8.8.8.8应该往下放？

不建议放8.8.8.8有些地方已经被isp劫持

aaron021512, 2020-07-15:
好的，多谢各位大佬，已解决！

closed 2020-07-15 by aaron021512

authon, 2020-08-24:
> 好的，多谢各位大佬，已解决！

你好，你这个是在哪里修改的，我打开hdcp/dns里面没有设置上有dns的地方