coolsnowwolf/lede #5094: 自从上次重大更新5.4内核开始 到现在刚刚PULL的源码编译后发现个问题

chiyu521, 2020-07-13:
玩游戏 街霸5 没办法在线玩了 选完人以后 就掉线 有的时候 没选完呢 就掉了 游戏本身联机是P2P的
测试过小娱和竞斗云  同样的问题

acooler15, 2020-07-14:
Upnp的问题吧，看下这个 #5020

chiyu521, 2020-07-16:
Thu Jul 16 19:30:13 2020 daemon.err uhttpd[2102]: udhcpc: started, v1.31.1
Thu Jul 16 19:30:13 2020 daemon.err uhttpd[2102]: udhcpc: sending discover
Thu Jul 16 19:30:16 2020 daemon.err uhttpd[2102]: udhcpc: no lease, failing
Thu Jul 16 19:30:17 2020 daemon.info dnsmasq[10660]: Connected to system UBus
Thu Jul 16 19:31:22 2020 authpriv.info dropbear[9092]: Exit (root) from <192.168.0.2:62343>: Exited normally
Thu Jul 16 19:31:34 2020 authpriv.info dropbear[7509]: Exit (root) from <192.168.0.2:62339>: Exited normally
Thu Jul 16 19:31:34 2020 authpriv.info dropbear[9073]: Exit (root) from <192.168.0.2:62342>: Exited normally
这个错误 和UPNP貌似没关系

acooler15, 2020-07-16:
> Exited normally

这段日志，没发现什么错误啊？

acooler15, 2020-07-16:
> Thu Jul 16 19:30:13 2020 daemon.err uhttpd[2102]: udhcpc: started, v1.31.1
> Thu Jul 16 19:30:13 2020 daemon.err uhttpd[2102]: udhcpc: sending discover
> Thu Jul 16 19:30:16 2020 daemon.err [...]

这段日志，没发现什么错误啊

chiyu521, 2020-07-16:
最后那3个 我一联机失败 就出这个

acooler15, 2020-07-16:
> 最后那3个 我一联机失败 就出这个

。。。我也不知道是啥情况了。。。
dropbear是提供sshd服务的，从日志上来看，是192.168.0.2的ssh客户端，用root账户登录路由器后正常退出

chiyu521, 2020-07-16:
我把UPNPD换到了
PKG_NAME:=miniupnpd
PKG_VERSION:=2.0.20170421
PKG_RELEASE:=3
然后问题依旧 我崩溃了~这次更新的文件太多了 实在是找不到什么地方出的问题

acooler15, 2020-07-16:
> 我把UPNPD换到了
> PKG_NAME:=miniupnpd
> PKG_VERSION:=2.0.20170421
> PKG_RELEASE:=3
> 然后问题依旧 我崩溃了~这次更新的文件太多了 实在是找不到什么地方出的问题

下载个比特彗星，打开看看有没有提示 UPnP NAT端口映射: 失败。
有IPv6的话，先关闭IPv6。

或者。。。设置dmz🤣

chiyu521, 2020-07-22:
最新报告: 小娱C1/C5 (MT7621)用最新的源码 已经没有这个问题了!  X86 和竞斗云 还是这个问题~~不知道到底差在哪里!

chiyu521, 2020-07-24:
终于找到游戏掉线的原因了只要开启SFE 就会在街霸网络对战开始的时候与对手断开连接 关闭了SFE恢复正常 不知道具体原因在哪 等大屌有时间看看吧

closed 2020-07-25 by chiyu521