coolsnowwolf/lede #5095: 5.4内核，K2P的ssr plus+不稳定

fjc0414, 2020-07-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
5.4内核，K2P的ssr plus+不稳定
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
无
## 2. 详细叙述

### (1) 具体问题

A：表现为gfw list模式启动一段时间后就会无法出墙。切换为绕过国内模式恢复正常，但是在此模式下如果有外网下载时插件崩溃，路由器luci也会无法打开。

### (2) 路由器型号和固件版本

A：K2P

### (3) 详细日志

A：Mon Jul 13 22:40:45 2020 kern.notice kernel: [    0.000000] Linux version 5.4.50 (fjc@fjc-K43SV) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2748-db12a9ccb)) #0 SMP Mon Jul 6 08:13:07 2020
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] MIPS: machine is Phicomm K2P
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Mon Jul 13 22:40:45 2020 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Mon Jul 13 22:40:45 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mon Jul 13 22:40:45 2020 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Zone ranges:
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000]   HighMem  empty
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Early memory node ranges
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Mon Jul 13 22:40:45 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Mon Jul 13 22:40:45 2020 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Mon Jul 13 22:40:45 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon Jul 13 22:40:45 2020 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
Mon Jul 13 22:40:45 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
Mon Jul 13 22:40:45 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Mon Jul 13 22:40:45 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00006cd0
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00006cd0
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] Memory: 120976K/131072K available (5641K kernel code, 198K rwdata, 1224K rodata, 1244K init, 230K bss, 10096K reserved, 0K cma-reserved, 0K highmem)
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Mon Jul 13 22:40:45 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.015484] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.055841] pid_max: default: 32768 minimum: 301
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.065193] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.079600] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.097821] rcu: Hierarchical SRCU implementation.
Mon Jul 13 22:40:45 2020 kern.info kernel: [    0.107964] smp: Bringing up secondary CPUs ...
Mon Jul 13 22:40:45 2020 kern.warn kernel: [    7.300280] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Mon Jul 13 22:40:45 2020 kern.warn kernel: [    7.300290] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mon Jul 13 22:40:45 2020 kern.warn kernel: [    7.300303] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Mon Jul 13 22:40:45 2020 kern.info kernel: [    7.300408] CPU1 revision is: 0001992f (MIPS 1004Kc)
[... 614 lines cut ...]
Mon Jul 13 23:18:45 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 13 23:18:45 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 13 23:18:49 2020 daemon.info dnsmasq[18364]: Connected to system UBus
Mon Jul 13 23:19:05 2020 authpriv.info dropbear[18443]: Child connection from 192.168.1.173:54897
Mon Jul 13 23:19:10 2020 authpriv.notice dropbear[18443]: Password auth succeeded for 'root' from 192.168.1.173:54897
Mon Jul 13 23:19:18 2020 user.notice shadowsocksr: ssr redir tcp error.restart!
Mon Jul 13 23:19:21 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 13 23:19:21 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 13 23:19:24 2020 daemon.info dnsmasq[18640]: Connected to system UBus
Mon Jul 13 23:19:24 2020 cron.err crond[18674]: crond (busybox 1.31.1) started, log level 5
Mon Jul 13 23:19:58 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 13 23:19:58 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 13 23:20:02 2020 daemon.info dnsmasq[19043]: Connected to system UBus
Mon Jul 13 23:20:35 2020 user.notice shadowsocksr: ssr redir tcp error.restart!
Mon Jul 13 23:20:38 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 13 23:20:38 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 13 23:20:42 2020 daemon.info dnsmasq[19318]: Connected to system UBus
Mon Jul 13 23:20:42 2020 cron.err crond[19352]: crond (busybox 1.31.1) started, log level 5
Mon Jul 13 23:21:34 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 13 23:21:34 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 13 23:21:38 2020 daemon.info dnsmasq[19725]: Connected to system UBus
Mon Jul 13 23:22:05 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 13 23:22:05 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 13 23:22:09 2020 daemon.info dnsmasq[20014]: Connected to system UBus
Mon Jul 13 23:22:10 2020 cron.err crond[20049]: crond (busybox 1.31.1) started, log level 5

acooler15, 2020-07-14:
我没有遇到这种情况，用的是v2ray还是trojan？

fjc0414, 2020-07-15:
> 我没有遇到这种情况，用的是v2ray还是trojan？

v2和trojan都有这种情况

chchia, 2020-07-16:
今天最新commit编译后任然一样

ssr redir tcp error.restart!

每一两分钟重启一次

[image]

acooler15, 2020-07-16:
要不，删了配置文件，重新编译下？🤣

coolsnowwolf, 2020-07-17:
端口冲突

closed 2020-07-17 by fjc0414