coolsnowwolf/lede #5097: do not install libwebsockets-openssl and libwebsockets-full

ntlclxyc, 2020-07-13:
I select mosquitto_ssl,but it is not istalled
x86_64

package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/lcl/lede'
package/Makefile:108: recipe for target '/home/lcl/lede/staging_dir/target-x86_6                                                                                                 4_musl/stamp/.package_install' failed
make[1]: *** [/home/lcl/lede/staging_dir/target-x86_64_musl/stamp/.package_insta                                                                                                 ll] Error 2
make[1]: Leaving directory '/home/lcl/lede'
/home/lcl/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
lcl@lcl-PC:~/lede$ Collected errors:
 * check_data_file_clashes: Package libwebsockets-openssl wants to install file                                                                                                  /home/lcl/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libwebsockets.so
        But that file is already provided by package  * libwebsockets-full
 * check_data_file_clashes: Package libwebsockets-openssl wants to install file                                                                                                  /home/lcl/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libwebsockets.so.15
        But that file is already provided by package  * libwebsockets-full
mos libwebsockets-openssl

acooler15, 2020-07-14:
libwebsockets-openssl包与libwebsockets-full包冲突？选了libwebsockets-full不能再安装 libwebsockets-openssl。应该是这个情况

minlang112, 2020-07-14:
> libwebsockets-openssl包与libwebsockets-full包冲突？选了libwebsockets-full不能再安装 libwebsockets-openssl。应该是这个情况

应该是冲突的

closed 2020-07-19 by coolsnowwolf