coolsnowwolf/lede #5109: 最新的版本编译报错 ，提示json.h: No such file or directory

qwazx, 2020-07-15:
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：完全按照教程出现了报错问题，多次从头下载编译都出现了相同的报错，但是没搞懂是什么原因，已经查过文件应该是全的。

### (2) 详细日志

A：
```
make[6]: Entering directory '/home/zx/桌面/lede/build_dir/hostpkg/libubox-2020-05-25-66195aee'
[ 47%] Building C object CMakeFiles/blobmsg_json.dir/blobmsg_json.c.o
/home/zx/桌面/lede/build_dir/hostpkg/libubox-2020-05-25-66195aee/blobmsg_json.c:21:11: fatal error: json.h: No such file or directory
   21 |  #include <json.h>
      |           ^~~~~~~~
compilation terminated.
make[6]: *** [CMakeFiles/blobmsg_json.dir/build.make:80: CMakeFiles/blobmsg_json.dir/blobmsg_json.c.o] Error 1
make[6]: Leaving directory '/home/zx/桌面/lede/build_dir/hostpkg/libubox-2020-05-25-66195aee'
make[5]: *** [CMakeFiles/Makefile2:226: CMakeFiles/blobmsg_json.dir/all] Error 2
make[5]: Leaving directory '/home/zx/桌面/lede/build_dir/hostpkg/libubox-2020-05-25-66195aee'
make[4]: *** [Makefile:147: all] Error 2
make[4]: Leaving directory '/home/zx/桌面/lede/build_dir/hostpkg/libubox-2020-05-25-66195aee'
make[3]: *** [Makefile:110: /home/zx/桌面/lede/build_dir/hostpkg/libubox-2020-05-25-66195aee/.built] Error 2
make[3]: Leaving directory '/home/zx/桌面/lede/package/libs/libubox'
time: package/libs/libubox/host-compile#2.08#0.56#2.84
make[2]: *** [package/Makefile:111: package/libs/libubox/host/compile] Error 2
make[2]: Leaving directory '/home/zx/桌面/lede'
make[1]: *** [package/Makefile:107: /home/zx/桌面/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zx/桌面/lede'
make: *** [/home/zx/桌面/lede/include/toplevel.mk:224：world] 错误 2

```

linxuankent, 2020-07-15:
跟我一样报错，json.h没连接，不知道怎么处理

linxuankent, 2020-07-15:
把lede/build-dir/hostpkg/json-c-0.13.1目录下的内容复制到/usr/include/json 目录下，如果目录存在全部删除重新复制，所有在hostpkg目录下提示的报错，对应报错文件中的 #include <json.h>改成 #include <json/json.h>，所有在target-x86_64_musl目录下的报错，对应报错文件中的 #include <json.h>改成 #include <json-c/json.h>。

dolphinpaopao, 2020-07-15:
建议每次编译都全新git clone，把dl文件夹保留备用即可。我每次都是全新编译，除开源码本身bug导致编译失败过几次，x86一直没有出现过问题。刚刚出炉全新编译了一个测试固件。
[image]

linxuankent, 2020-07-15:
> 建议每次编译都全新git clone，把dl文件夹保留备用即可，这样可以最大程度地避免更新和旧配置不匹配等等导致的问题。我每次都是全新编译，除开原始码错误导致编译失败过以前，x86一直没有出现过问题。刚刚出炉全新编译了一个测试固件。
> [image]

这个只出现在第一次git的情况中，如果只是更新没有这个问题。

qwazx, 2020-07-15:
> 建议每次编译都全新git clone，把dl文件夹保留备用即可，这样可以最大程度避免更新和旧配置不匹配等等导致的问题。我每次都是全新编译，除开源码本身bug导致编译失败过几次，x86一直没有出现过问题。刚刚出炉全新编译了一个测试固件。
> [image]
> 
> It is recommended that you keep your git clone new each time you compile and keep your dl folder in reserve to minimize the problems caused by updates, old configuration mismatches, etc. . Every time I compile is brand new, except open source itself Bug caused compilation failure several times, X86 has not been a problem. Just came out with an all-new test firmware.

主要我都是从头进行编译的，会删除全部的文件重新git资源库，所以不应该是兼容问题。并且我自己时从头编译了好几次都这样，难道是因为helloworld的问题吗？

qwazx, 2020-07-15:
> 把lede/build-dir/hostpkg/json-c-0.13.1目录下的内容复制到/usr/include/json 目录下，如果目录存在全部删除重新复制，所有在hostpkg目录下提示的报错，对应报错文件中的 #include <json.h>改成 #include <json/json.h>，所有在target-x86_64_musl目录下的报错，对应报错文件中的 #include <json.h>改成 #include <json-c/json.h>。
> 
> Copy the contents of the directory lede/build-dir/hostpkg/json-c-0.13.1 to the/usr/include/json directory. If the directory is deleted and copied again, any errors prompted in the hostpkg directory will correspond to the # include < in the error file. Change h > to # include < json/json. H > , all errors in the target-x86 directory, correspond to # include < JSONIN the error file. Change h > to # include < json-c/json.json. H > .

感谢，我回去试一下，谢谢大佬

dolphinpaopao, 2020-07-15:
> 主要我都是从头进行编译的，会删除全部的文件重新git资源库，所以不应该是兼容问题。并且我自己时从头编译了好几次都这样，难道是因为helloworld的问题吗？

应该不是ssr+，我的测试固件也是包含了最新ssr+的，还有可能是编译环境的原因导致一些莫名其妙的问题，避免使用系统自带的WSL，文件路径避免中文字符，避免使用桌面系统版本和增加编译系统的内存等等。我用的推荐系统18.04 x64server版本，virtualbox 虚拟机。使用体验良好。

qwazx, 2020-07-15:
> > > 建议每次编译都全新git clone，把dl文件夹保留备用即可，这样可以最大程度避免更新和旧配置不匹配等等导致的问题。我每次都是全新编译，除开源码本身bug导致编译失败过几次，x86一直没有出现过问题。刚刚出炉全新编译了一个测试固件。
> > > [image]
> > > It is recommended that you keep your git clone new each time you compile and keep your dl folder in reserve to minimize the problems caused by updates, old configuration mismatches, etc. . Every time I compile is brand new, except open source itself Bug caused compilation failure several times, X86 has not been a problem. Just came out with an all-new test firmware.
> > 
> > 
> > 主要我都是从头进行编译的，会删除全部的文件重新git资源库，所以不应该是兼容问题。并且我自己时从头编译了好几次都这样，难道是因为helloworld的问题吗？
> 
> 应该不是ssr+，我的测试固件也是包含了最新ssr+的，还有可能是编译环境的原因导致一些莫名其妙的问题，避免使用win10系统自带的WSL，文件路径避免中文字符，避免使用桌面系统版本和增加编译系统的内存等等。我用的推荐系统18.04 x64server版本，virtualbox 虚拟机。使用体验良好。

我是使用的虚拟机编译的，那么就两个问题，中文的乌班图路径和高版本，我回去试试英文路径，还不行就试试18的版本，最新的20可能会有兼容问题吧。感谢帮助

linxuankent, 2020-07-15:
> > > > 建议每次编译都全新git clone，把dl文件夹保留备用即可，这样可以最大程度地避免更新和旧配置不匹配等等导致的问题。我每次都是全新编译，除开原始码错误导致编译失败过几次，X86一直没有出现过问题。刚刚出炉全新编译了一个测试固件。
> > > > 建议您新的每次让你的git克隆编译并保持您的DL文件夹中的储备，以减少因更新的问题，老配置不匹配等。我每次编译都是全新的，除了开源本身的Bug几次导致编译失败外，X86都不是问题。刚刚推出了全新的测试固件。
> > > > [image]
> > > 
> > > 
> > > 主要我都是从头进行编译的，会删除全部的文件重新git资源库，所以不应该是兼容的问题。并且我自己时从头进行了好几次都这样，难道是因为helloworld的问题吗？
> > 
> > 
> > 应该不是ssr +，我的测试固件也是包含了最新的ssr +的，还有可能是编译环境的原因导致一些莫名其妙的问题，避免使用win10系统自带的WSL，文件路径避免中文字符，避免使用桌面系统版本和我用的推荐系统18.04 x64服​​务器版本，virtualbox虚拟机。使用体验良好。
> 
> 我是使用的虚拟机编译的，那么就两个问题，中文的乌班图路径和高版本，我回去试试英语路径，还不行就试试18的版本，最新的20可能会有兼容问题吧。感谢帮助

不是乌班图版本问题，18也存在这个问题，我现在用的是20,

qwazx, 2020-07-15:
> 不是乌班图版本问题，18也存在这个问题，我现在用的是20,

那我先试试改英文路径，还不行再试试您上面说的方案，谢谢帮助

zwczou, 2020-07-17:
`cp -r build_dir/target-x86_64_musl/json-c-0.13.1/ipkg-install/usr/include/json-c/* staging_dir/target-x86_64_musl/usr/include/
cp -r build_dir/target-x86_64_musl/json-c-0.13.1/ipkg-install/usr/include/json-c staging_dir/hostpkg/include/json
`

我根据CMakefile的文件
发现需要包含的路径为以上两个
如果还不能解决，可能需要多复制一份到 /include/json

qwazx, 2020-07-17:
> `cp -r build_dir/target-x86_64_musl/json-c-0.13.1/ipkg-install/usr/include/json-c/* staging_dir/target-x86_64_musl/usr/include/`
>
> `cp -r [...]

我这两天测试了，结果就是中文目录的问题好像，我使英文版的乌班图改为中文，但是路径保留英文就没有报错。就很奇怪。之前没有出现过这个问题。

qwazx, 2020-07-17:
> 不是乌班图版本问题，18也存在这个问题，我现在用的是20,

好像就是中文路径的问题，我保留路径为英文的情况就编译成功了，但是如果是中文就不行。但是之前没遇到过这个问题啊

linxuankent, 2020-07-19:
我在编译python相关文件的时候报utf-8错误我就换成了英文目录，我已经编译完成了

0987363, 2020-09-30:
相同问题

用虚拟机ubuntu20编译没问题
换docker ubuntu20，目录映射进去，也报这个错，都是全英文目录

用上面的方式就正常了
cp -r build_dir/target-x86_64_musl/json-c-0.13.1/ipkg-install/usr/include/json-c/* staging_dir/target-x86_64_musl/usr/include/
cp -r build_dir/target-x86_64_musl/json-c-0.13.1/ipkg-install/usr/include/json-c staging_dir/hostpkg/include/json

closed 2021-06-20 by coolsnowwolf