coolsnowwolf/lede #5111: 运行9天出现问题

tengyuankoo, 2020-07-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：运行9天，5G连接不上提示密码错误，实际密码没改过是对的，重启进后台跳错误信息

### (2) 路由器型号和固件版本

A：红米AC2100 5.4.50

### (3) 详细日志

A：/usr/lib/lua/luci/dispatcher.lua:569: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	/usr/lib/lua/luci/dispatcher.lua:569: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:238: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
[image]

tengyuankoo, 2020-07-15:
固件做AP使用，没选服务插件，应该没有冲突

closed 2020-07-15 by tengyuankoo

qiqiandfei, 2020-07-30:
更改5G信道试试