coolsnowwolf/lede #5113: 请问如何扩大K2P固件大小，已经硬改32M

ltactor, 2020-07-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

      最近编译升级k2p固件，发现K2P  flash只有16M，安装完基本固件脸curl都装不了。所以硬改k2p为32M ，
     
      刷入最新的breed，目前可以使用以前的固件。
     
     更改  target/linux/ramips/dts/mt7621_phicomm_k2p.dts 里面 的 label = "firmware";	reg = <0xa0000 0xf60000>;  
     更改  target/linux/ramips/image/mt7621.mk 里面  IMAGE_SIZE := 32000k
     然后编译，出15.7 MB (16,515,845 字节)的固件。 
     breed选择斐讯专区刷入， k2p闪红灯，有线连接K2p无法获取地址。
   
     请问，还需要更改哪里？ 
     感谢。

### (2) 路由器型号和固件版本

A：K2P a2

### (3) 详细日志

A：无法启动没有日志。

acooler15, 2020-07-16:
mt7621_phicomm_k2p.dts里reg 的值要与target/linux/ramips/image/mt7621.mk 里面  IMAGE_SIZE的值相匹配，如IMAGE_SIZE为32448k，则reg为<0xa0000 0x1fb0000>。其次，还需要在mt7621_phicomm_k2p.dts的spi的flash中添加一行`broken-flash-reset;`，可以参照target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts。

acooler15, 2020-07-16:
[image]
修改命令如下：
```
sed -i '/spi-max-frequency/a\\t\tbroken-flash-reset;' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/<0xa0000 0xf60000>/<0xa0000 0x1fb0000>/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/15744k/32448k/g' target/linux/ramips/image/mt7621.mk
```

ltactor, 2020-07-16:
> [image]
> 修改命令如下：
> 
> ```
> sed -i '/spi-max-frequency/a\\t\tbroken-flash-reset;' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
> sed -i 's/<0xa0000 0xf60000>/<0xa0000 0x1fb0000>/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
> sed -i 's/15744k/32448k/g' target/linux/ramips/image/mt7621.mk
> ```

感谢回复，更改后，测试重新编译，不行还是不启动，只亮红灯。

现在重新git     重新编译中。。。。。。。

acooler15, 2020-07-16:
> 感谢回复，更改后，测试重新编译，不行还是不启动，只亮红灯。
>
> 现在重新git 重新编译中。。。。。。。

最好在`make menuconfig`之后再修改mt7621_phicomm_k2p.dts和mt7621.mk

ltactor, 2020-07-17:
正在编译，昨天梯子不好，今天换梯子，重新编译，单核编译，稍慢。

acooler15, 2020-07-17:
> 正在编译，昨天梯子不好，今天换梯子，重新编译，单核编译，稍慢。

可以借助git actions啊🤣

ltactor, 2020-07-17:
> > 正在编译，昨天梯子不好，今天换梯子，重新编译，单核编译，稍慢。
> 
> 可以借助git actions啊🤣

非常感谢回复，已经编译成功，并成功刷写，运行正常。
昨天编译遇到点小问题，一共编译2次才成功。
git actions 不方便看错误日志，编译成功后肯定会用的。感谢。

这次是因为k2p升级wifi固件，导致固件包太大基本15m左右，已经到极限。 因需要安装curl200kb左右，已经无法安装，k2p一直表现不错，扔了可惜，决定换32M。需要基本的电烙铁，稍有经验即可。 然后按照acooler15 提供的步骤更改，编译即可。

感谢acooler15 耐心帮助，祝好运。。。。。。。

closed 2020-07-17 by ltactor