coolsnowwolf/lede #5114: 编译（20G）大容量squashfs-img,实际编译的包大于38g还在继续增长。

linxuankent, 2020-07-16:
给虚拟机分配了200G硬盘都编译满了还不够。以前编译都是正常的，从上周开始编译出现这个问题。
[image]
[image]

EnnawYang, 2020-07-16:
头一次见到设置几十个G大小的固件，按照你这个设置的话img大小正常在30G左右

xiehuayi75, 2020-07-16:
你逗着玩呢

linxuankent, 2020-07-16:
> 你逗着玩呢

> 头一次见到设置几十个G大小的固件，按照你这个设置的话img大小正常在30G左右
这下图才是正常编译完的大小，我软路由本身就120G ssd,要不是空间不够我直接编译120g的固件懒得在安装完系统后在分配空间。
我之前也是这么设置的编译出来就应该不会到
[image]

minlang112, 2020-07-16:
为何要如此操作……直接U盘启动，把ssd全部挂载给openwrt岂不美哉

mapestliu, 2020-07-16:
启用压缩就可以了，我编译64G的也才几十兆
CONFIG_TARGET_IMAGES_GZIP=y

linxuankent, 2020-07-16:
> CONFIG_TARGET_IMAGES_GZIP = y

晚点试下

linxuankent, 2020-07-16:
> 为何要如此操作……直接U盘启动，把ssd全部挂载给openwrt岂不美哉

那不就是相当于浪费个u盘一直用着

hxlls, 2020-07-16:
要升级都要40G的内存，要不升级不了，这是神操作？op 设置的小容量，然后挂载的剩余分区不完事了。

linxuankent, 2020-07-16:
> 启用压缩就可以了，我编译64G的也才几十兆
> CONFIG_TARGET_IMAGES_GZIP = y

直接在config.in中添加就行了吗？

mapestliu, 2020-07-16:
.config中，你看看默认有没有启用

mapestliu, 2020-07-16:
我是写好配置文件去编译，在上面选择不习惯，或者你看看选择项里面有没有。

linxuankent, 2020-07-16:
> 我是写好配置文件去编译，在上面选择不习惯，或者你看看选择项里面有没有。

可以了谢谢

closed 2020-07-16 by linxuankent