coolsnowwolf/lede #5117: 系统日志中一直有网卡错误出现（x520万兆网卡）

lam640, 2020-07-16:
(1) 具体问题
A：系统日志中大量类似错误
kern.warn kernel: [121669.543660] ixgbe 0000:01:00.1: Warning firmware error detected FWSM: 0x00000000
主机名 | OpenWrt

-- | --
主机型号 | Intel(R) Xeon(R) CPU E3-1265L v3 @ 2.50GHz : 4 Core 8 Thread (CpuMark : 122277.416890 Scores)
CPU 信息 | 1502.398 MHz +52.0°C (high = +84.0°C, crit = +90.0°C)
固件版本 | HomeLede v2020.07.07 based on OpenWrt R20.7.7 / LuCI Master (git-20.187.45844-a7450cd)
内核版本 | 5.4.50
本地时间 | Fri Jul 10 13:53:49 2020
运行时间 | 1d 9h 49m 20s
平均负载 | 0.50, 0.18, 0.06
CPU 使用率（%） |  

(3) 详细日志
A：[121766.305260] ixgbe 0000:01:00.1: Warning firmware error detected FWSM: 0x00000000
[121766.529255] ixgbe 0000:01:00.0: Warning firmware error detected FWSM: 0x00000000
[121768.325211] ixgbe 0000:01:00.1: Warning firmware error detected FWSM: 0x00000000
[121768.545230] ixgbe 0000:01:00.0: Warning firmware error detected FWSM: 0x00000000

shenyz, 2020-07-16:
新内核，老网卡驱动的结果

dolphinpaopao, 2020-07-17:
第三方固件需要咨询原作者采用了什么驱动和修改过什么地方。开源的东西，不太好就凭一个修改过的报错来定位问题。

coolsnowwolf, 2020-07-17:
网卡加载的firmware 有问题

lam640, 2020-07-17:
> 网卡加载的firmware 有问题
请问cool大是要通过刷新网卡固件解决吗？谢谢

zpkx, 2021-06-13:
我也出现同样的错误，是不是OP里带的网卡驱动版本低了？

closed 2021-06-20 by coolsnowwolf