coolsnowwolf/lede #5118: 编译过程出错

wdza1234, 2020-07-17:
ake[6]: Leaving directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/zlib'
make[6]: Entering directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/libbacktrace'
make  all-am
make[7]: Entering directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/libbacktrace'
true  DO=all multi-do # make
make[7]: Leaving directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/libbacktrace'
make[6]: Leaving directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/libbacktrace'
make[6]: Entering directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[6]: Leaving directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/libcpp'
make[6]: Entering directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/libdecnumber'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/libdecnumber'
make[6]: Entering directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/gcc'
g++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Ic -I/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/c -I/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/home/sdjahgj1234/lede/staging_dir/host/include -I/home/sdjahgj1234/lede/staging_dir/host/include -I/home/sdjahgj1234/lede/staging_dir/host/include  -I/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o c/c-decl.o -MT c/c-decl.o -MMD -MP -MF c/.deps/c-decl.TPo /home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/c/c-decl.c
In file included from /home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/c/c-decl.c:11573:0:
./gt-c-c-decl.h:2321:2: fatal error: error writing to /home/sdjahgj1234/lede/tmp/cc1EsA1f.s: No space left on device
 };
  ^
compilation terminated.
Makefile:1110: recipe for target 'c/c-decl.o' failed
make[6]: *** [c/c-decl.o] Error 1
make[6]: Leaving directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/gcc'
Makefile:4270: recipe for target 'all-gcc' failed
make[5]: *** [all-gcc] Error 2
make[5]: Leaving directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
Makefile:905: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
Makefile:89: recipe for target '/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/.built' failed
make[3]: *** [/home/sdjahgj1234/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/.built] Error 2
make[3]: Leaving directory '/home/sdjahgj1234/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#5.09#0.43#5.45
toolchain/Makefile:98: recipe for target 'toolchain/gcc/final/compile' failed
make[2]: *** [toolchain/gcc/final/compile] Error 2
make[2]: Leaving directory '/home/sdjahgj1234/lede'
toolchain/Makefile:96: recipe for target '/home/sdjahgj1234/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/sdjahgj1234/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/sdjahgj1234/lede'
/home/sdjahgj1234/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

请问大神们如何解决，

minlang112, 2020-07-17:
No space left on device
你磁盘空间不够啊……

acooler15, 2020-07-21:
No space left on device

closed 2021-06-20 by coolsnowwolf