coolsnowwolf/lede #5122: R2S使用SFE+DNS加速，取消后DNS无法解析

ramondelee, 2020-07-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用SFE+DNS加速，取消DNS加速后DNS无法解析，导致无法访问网络

### (2) 路由器型号和固件版本

A：R2S，OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)，内核：5.4.50

### (3) 详细日志

A：
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Sat Jul 18 00:06:53 2020 kern.notice kernel: [    0.000000] Linux version 5.4.50 (ramonde@ramonde-pc) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2820-6e2f7ac72)) #0 SMP PREEMPT Fri Jul 17 02:54:46 2020
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] Machine model: FriendlyARM NanoPi R2S
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] cma: Reserved 16 MiB at 0x000000003f000000
Sat Jul 18 00:06:53 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 261632
Sat Jul 18 00:06:53 2020 kern.debug kernel: [    0.000000]   DMA32 zone: 4088 pages used for memmap
Sat Jul 18 00:06:53 2020 kern.debug kernel: [    0.000000]   DMA32 zone: 0 pages reserved
Sat Jul 18 00:06:53 2020 kern.debug kernel: [    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s48280 r8192 d29544 u86016
Sat Jul 18 00:06:53 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s48280 r8192 d29544 u86016 alloc=21*4096
Sat Jul 18 00:06:53 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
Sat Jul 18 00:06:53 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS2,1500000 console=tty1 earlycon=uart8250,mmio32,0xff130000 swiotlb=1 root=PARTUUID=5452574f-02 rw rootwait loglevel=9
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] Memory: 995504K/1046528K available (8190K kernel code, 612K rwdata, 2652K rodata, 1792K init, 717K bss, 34640K reserved, 16384K cma-reserved)
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] rcu: 	RCU event tracing is enabled.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] 	Tasks RCU enabled.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Sat Jul 18 00:06:53 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x398 with crng_init=0
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.001382] Console: colour dummy device 80x25
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.001853] printk: console [tty1] enabled
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.002242] printk: bootconsole [uart8250] disabled
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.002745] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.002781] pid_max: default: 32768 minimum: 301
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.003037] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.003071] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.025615] ASID allocator initialised with 32768 entries
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.033597] rcu: Hierarchical SRCU implementation.
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.049688] smp: Bringing up secondary CPUs ...
Sat Jul 18 00:06:53 2020 kern.info kernel: [    0.081936] Detected VIPT I-cache on CPU1
[... 411 lines cut ...]
Sat Jul 18 00:10:46 2020 daemon.notice miniupnpd[4578]: HTTP IPv6 address given to control points : [fdda:c19c:ff1d::1]
Sat Jul 18 00:10:46 2020 daemon.notice miniupnpd[4578]: Listening for NAT-PMP/PCP traffic on port 5351
Sat Jul 18 00:10:46 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Sat Jul 18 00:10:57 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sat Jul 18 00:11:01 2020 daemon.notice netifd: Interface 'Modem' is enabled
Sat Jul 18 00:11:01 2020 daemon.notice netifd: Interface 'Modem' has link connectivity
Sat Jul 18 00:11:01 2020 daemon.notice netifd: Interface 'Modem' is setting up now
Sat Jul 18 00:11:01 2020 daemon.notice netifd: Modem (4952): udhcpc: started, v1.31.1
Sat Jul 18 00:11:01 2020 daemon.notice netifd: Modem (4952): udhcpc: sending discover
Sat Jul 18 00:11:01 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Sat Jul 18 00:11:01 2020 daemon.info dnsmasq[5015]: Connected to system UBus
Sat Jul 18 00:11:03 2020 daemon.notice netifd: Modem (4952): udhcpc: sending select for 192.168.1.3
Sat Jul 18 00:11:03 2020 daemon.notice netifd: Modem (4952): udhcpc: lease of 192.168.1.3 obtained, lease time 259200
Sat Jul 18 00:11:03 2020 daemon.notice netifd: Interface 'Modem' is now up
Sat Jul 18 00:11:13 2020 user.notice firewall: Reloading firewall due to ifup of Modem (eth0)
Sat Jul 18 00:11:40 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jul 18 00:11:40 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jul 18 00:11:43 2020 daemon.info dnsmasq[5864]: Connected to system UBus
Sat Jul 18 00:11:43 2020 daemon.err dnscache[5869]: * 07/18 00:11:43| pdnsd: info: pdnsd-1.2.9b-par starting.
Sat Jul 18 00:11:43 2020 daemon.err dnscache[5869]: * 07/18 00:11:43| pdnsd: warning: EOF encountered while reading entry count from disk cache file.
Sat Jul 18 00:12:29 2020 kern.alert kernel: [  349.942731] fast-classifier: shutting down
Sat Jul 18 00:12:29 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jul 18 00:12:29 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jul 18 00:12:33 2020 daemon.info dnsmasq[6449]: Connected to system UBus
Sat Jul 18 00:12:33 2020 daemon.err dnscache[5869]: * 07/18 00:12:33| pdnsd: warning: Caught signal 15. Exiting.

Ljzkirito, 2020-07-18:
是不是取消DNS加速后，DHCP/DNS设置中DNS转发中127.0.0.1#5333未删除？

Cathgao, 2020-07-18:
同样问题

forever2019, 2020-07-18:
这个问题  我也遇见了   xy-c5  DNS加速用了下  然后关闭了  就打不开网页了  
没法子又刷的固件才恢复

ramondelee, 2020-07-18:
> 是不是取消DNS加速后，DHCP/DNS设置中DNS转发中127.0.0.1#5333未删除？

是没删除，但删除后还是不能访问网站

ramondelee, 2020-07-18:
> 这个问题 我也遇见了 xy-c5 DNS加速用了下 然后关闭了 就打不开网页了
> 没法子又刷的固件才恢复

不用刷，重置回出厂设置就好了

forever2019, 2020-07-18:
有一次恢复出厂后    管理页面都进不去  提示页面脚本错误什么的    具体不清楚  
下次遇见了  提交bug  问问大佬怎么回事

ramondelee, 2020-07-18:
希望能忙修复

Mini-R, 2020-07-20:
我也遇到了同样的问题，暂时先开着DNS加速用着了，等下个版本修复。

exxtremecn, 2020-07-21:
在J1900/螃蟹双网卡上遇到了同样的问题，不开一切ok，一旦开加速，就会无法访问网络，关闭也不会自动恢复，必须重启才恢复

ostrichii, 2020-07-23:
[https://github.com/coolsnowwolf/lede/issues/5158](url)
应该是这个issue

ramondelee, 2020-07-23:
> #5158
> 应该是这个issue

是这个，昨晚编译的新版已经修复

closed 2020-07-23 by ramondelee

fhbyljj, 2021-04-22:
> 是不是取消DNS加速后，DHCP / DNS设置中DNS转发中127.0.0.1＃5333未删除？

感谢你这个回复，我现在发现打开DNS加速后，关闭了DNS后就出问题