coolsnowwolf/lede #5123: MTK 5.4 内核交换机功能和 HWNAT 功能测试反馈

coolsnowwolf, 2020-07-17:
来自 PPTP52 和 暗云的补丁提交，取消某些机型的 DSA 而重新添加 MT7530 交换机的支持和 HWNAT  硬件转发

从此 AC2100 之类的流量转发性能和老毛子系列完全持平

取消 SFE 而使用 Flow 并硬件加速即可

有任何使用体验欢迎在下面反馈

mjyhj, 2020-07-17:
牛逼，消灭0评论

monsterzzzz, 2020-07-18:
新3试了一下，发现有线网lan不通，不能ping通，恢复出厂设置也不行
wifi是通的，可以获得ip，可以进入luci设置

coolsnowwolf, 2020-07-18:
需要 make clean 重新生成带新的 dts 的内核

coolsnowwolf, 2020-07-18:
@AmadeusGhost

AmadeusGhost, 2020-07-18:
可能是旧的 config 的问题吧，我昨天全新编译是正常的
[image]
[image]

afeiy, 2020-07-18:
红米2100在breedz中更新initramfs-kernel.bin就不成功，目前回退到上一版本使用中

EnnawYang, 2020-07-18:
1.make clean之后再编译，2.升级不要保存配置，不然可能会出现不能联网的现象

monsterzzzz, 2020-07-18:
删除.config重新配置就可以了

afeiy, 2020-07-18:
> 1.make clean、重新生成config之后再编译，2.升级不要保存配置，不然可能会出现不能联网的现象

严格按上述步骤编译成功，在breed中仍不能更新initramfs-kernel.bin。又刷回老版本了。

AmadeusGhost, 2020-07-18:
> 严格按上述步骤编译成功，在breed中仍不能更新initramfs-kernel.bin。又刷回老版本了。

直接用旧版本的 initramfs-kernel.bin 启动后不保留配置刷现在的 squashfs-sysupgrade.bin 试试

afeiy, 2020-07-18:
> > 严格按上述步骤编译成功，在breed中仍不能更新initramfs-kernel.bin。又刷回老版本了。
> 
> 直接用旧版本的 initramfs-kernel.bin 启动后不保留配置刷现在的 squashfs-sysupgrade.bin 试试

web升级提示“不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件。”晚一点儿再在breed试试您说的方法。

AmadeusGhost, 2020-07-18:
> 严格按上述步骤编译成功，在breed中仍不能更新initramfs-kernel.bin。又刷回老版本了。

不是有个 factory.bin 嘛，breed 刷那个试试吧

kisaragychihaya, 2020-07-18:
[syslog.txt](https://github.com/coolsnowwolf/lede/files/4941657/syslog.txt)
[krllog.txt](https://github.com/coolsnowwolf/lede/files/4941658/krllog.txt)
RM AC2100 无法拨号 报错如上

afeiy, 2020-07-18:
> > 严格按上述步骤编译成功，在breed中仍不能更新initramfs-kernel.bin。又刷回老版本了。
> 
> 不是有个 factory.bin 嘛，breed 刷那个试试吧
问题依旧。已采取的步骤：
1，breed中刷入 factory.bin，无限重启；
2，breed中刷入旧版本initramfs-kernel.bin，后进入web升级仍提示“不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件。”
仍退回老版本使用中。

xcf13363175, 2020-07-18:
最新编译的固件，刷入openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100-squashfs-sysupgrade.bin 同样无限重启

AmadeusGhost, 2020-07-19:
> RM AC2100 无法拨号 报错如上

```
Sat Jul 18 01:04:12 2020 daemon.notice netifd: wan (4676): uci: Entry not found
Sat Jul 18 01:04:12 2020 daemon.notice netifd: wan (4676): sh: out of range
Sat Jul 18 01:04:12 2020 daemon.info pppd[4699]: Plugin rp-pppoe.so loaded.
Sat Jul 18 01:04:12 2020 daemon.info pppd[4699]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sat Jul 18 01:04:12 2020 daemon.notice pppd[4699]: pppd 2.4.8 started by root, uid 0
Sat Jul 18 01:04:27 2020 daemon.warn pppd[4699]: Timeout waiting for PADO packets
Sat Jul 18 01:04:27 2020 daemon.err pppd[4699]: Unable to complete PPPoE Discovery
Sat Jul 18 01:04:27 2020 daemon.info pppd[4699]: Exit.
Sat Jul 18 01:04:27 2020 daemon.notice netifd: Interface 'wan' is now down
Sat Jul 18 01:04:27 2020 daemon.notice netifd: Interface 'wan' is setting up now
```
看日志应该跟换回4.14驱动没有关系，可能 ppp 本身的问题
话说红米ac2100到底能不能用啊，有人无限重启有人正常启动...

xcf13363175, 2020-07-19:
> ```
> Sat Jul 18 01:04:12 2020 daemon.notice netifd: wan (4676): uci: Entry not found
> Sat Jul 18 01:04:12 2020 daemon.notice netifd: wan (4676): sh: out of range
> Sat Jul 18 01:04:12 2020 daemon.info [...]

pb-boot下面刷5.4内核无限重启（padavan正常），准备换breed再刷试试。

AmadeusGhost, 2020-07-19:
> pb-boot下面刷5.4内核无限重启（padavan正常），准备换breed再刷试试。

先不用试了，我套个东西上去试试看

Aes64X, 2020-07-19:
> ```
> Sat Jul 18 01:04:12 2020 daemon.notice netifd: wan (4676): uci: Entry not found
> Sat Jul 18 01:04:12 2020 daemon.notice netifd: wan (4676): sh: out of range
> Sat Jul 18 01:04:12 2020 daemon.info [...]

R3G正常，5.4.52内核，只是没有交换机选项了，还是用的DSA？不过这无所谓了。。。。

xiaoguoyi, 2020-07-19:
> ```
> Sat Jul 18 01:04:12 2020 daemon.notice netifd: wan (4676): uci: Entry not found
> Sat Jul 18 01:04:12 2020 daemon.notice netifd: wan (4676): sh: out of range
> Sat Jul 18 01:04:12 2020 daemon.info [...]

新三也是这样的报错，无法拨号，提示 rp-pppoe 冲突

Aes64X, 2020-07-19:
> > pb-boot下面刷5.4内核无限重启（padavan正常），准备换breed再刷试试。
> 
> 先不用试了，我套个东西上去试试看

我的R3G没测PPPOE啊，我测的DHCP

AmadeusGhost, 2020-07-19:
> DHCP测速，300M带宽，在概况仍然看到CPU占用30%左右。。。。不知道是否起作用了 --，有时是3%

DSA 目前不支持 hwnat 的，有空我把  r3g 也改下

Aes64X, 2020-07-19:
> > DHCP测速，300M带宽，在概况仍然看到CPU占用30%左右。。。。不知道是否起作用了 --，有时是3%
> 
> DSA 目前不支持 hwnat 的，有空我把 r3g 也改下

呃。。。。我抄newifi的先试试

kisaragychihaya, 2020-07-19:
对了，还有一个问题就是有线的LAN没法获取IP了

AmadeusGhost, 2020-07-19:
> 对了，还有一个问题就是有线的LAN没法获取IP了

我写反了 Orz I'm sorry about this.

Aes64X, 2020-07-19:
> > 对了，还有一个问题就是有线的LAN没法获取IP了
> 
> 我写反了 Orz I'm sorry about this.

啊，我抄了newifi的写法在R3G上，他工作正常，因为是远程R3G，我只能测试DHCP。300M带宽WAN--LAN CPU占用不超过5%

AmadeusGhost, 2020-07-19:
> 啊，我抄了newifi的写法在R3G上，他工作正常，因为是远程R3G，我只能测试DHCP。300M带宽WAN--LAN CPU占用不超过5%

就 ac2100 的反了，顺便套了个 lzma 上去看看能不能解决不启动的问题（

AmadeusGhost, 2020-07-19:
@afeiy @xcf13363175 @kisaragychihaya pr #5136 试试

Ozzy-hzy, 2020-07-19:
刷进去lan口无法获取ip，lan口物理设置改为eth0.2后能获取到了

iamfuhui, 2020-07-19:
newifi D2 刷完以后 WAN LAN 都无法获取 ip 地址

AmadeusGhost, 2020-07-19:
> newifi D2 刷完以后 WAN LAN 都无法获取 ip 地址

不能用旧 config，不能保留配置升级

kisaragychihaya, 2020-07-19:
AC2100可以正常使用了

iamfuhui, 2020-07-19:
> > newifi D2 刷完以后 WAN LAN 都无法获取 ip 地址
> 
> 不能用旧 config，不能保留配置升级

干掉 config 以后，重新编译搞定了。谢谢。

xiaoguoyi, 2020-07-19:
删除config 不行，make distclian重新编译仍然不能拨号（新三）
Sun Jul 19 19:14:17 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sun Jul 19 19:14:17 2020 daemon.err insmod: module is already loaded - slhc
Sun Jul 19 19:14:17 2020 daemon.err insmod: module is already loaded - ppp_generic
Sun Jul 19 19:14:17 2020 daemon.err insmod: module is already loaded - pppox
Sun Jul 19 19:14:17 2020 daemon.err insmod: module is already loaded - pppoe
Sun Jul 19 19:14:17 2020 daemon.notice netifd: wan (4549): uci: Entry not found
Sun Jul 19 19:14:17 2020 daemon.notice netifd: wan (4549): sh: out of range
Sun Jul 19 19:14:17 2020 daemon.info pppd[4594]: Plugin rp-pppoe.so loaded.
Sun Jul 19 19:14:17 2020 daemon.info pppd[4594]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Jul 19 19:14:17 2020 daemon.notice pppd[4594]: pppd 2.4.8 started by root, uid 0

afeiy, 2020-07-20:
make clean,make distclean编译出来的都没成功，删除lede目录重新clone编译出来的可用，目前正常使用中。

Aes64X, 2020-07-20:
> > newifi D2 刷完以后 WAN LAN 都无法获取 ip 地址
> 
> 不能用旧 config，不能保留配置升级

额。。。那个R3G的似乎有问题呀，保留配置刷进去交换机页面显示5个网口。。。。不保留配置刷进去通电3网口灯全亮，无法启动 - -

xcf13363175, 2020-07-20:
实验证明RM2100不能用pb-boot，换breed可以正常启动openwrt

coolsnowwolf, 2020-07-20:
pb-boot 只能刷 factory 固件

AmadeusGhost, 2020-07-20:
> 额。。。那个R3G的似乎有问题呀，保留配置刷进去交换机页面显示5个网口（提示存在未知的拓扑）。。。。不保留配置刷进去通电3网口灯全亮，无法启动 - -

不应该啊，https://github.com/coolsnowwolf/openwrt/blob/lede-17.01/target/linux/ramips/mt7621/base-files/etc/board.d/02_network 这里也是这样写的
```
xiaomi,mir3g)
	ucidef_add_switch "switch0" \
		"2:lan:2" "3:lan:1" "1:wan" "6t@eth0"
	;;
```

Aes64X, 2020-07-20:
> 不应该啊，https://github.com/coolsnowwolf/openwrt/blob/lede-17.01/target/linux/ramips/mt7621/base-files/etc/board.d/02_network 这里也是这样写的
>
> ```
> xiaomi,mir3g)
> 	ucidef_add_switch "switch0" \
> 		"2:lan:2" [...]

我看了下。。。其他的部分都一样

在  https://github.com/coolsnowwolf/lede/commit/1e8536f1175d3fdf1d7f6b39e56673d7df5e6db3
 target/linux/ramips/mt7621/base-files/etc/board.d/02_network  第150行 
150      	xiaomi,mir3g|\
                xiaomi,mi-router-ac2100|\
	        xiaomi,redmi-router-ac2100)
		lan_mac=$(cat /sys/class/net/eth0/address)

这里我当时。。。忘记抄上了，但他工作正常-- ，交换机页面端口也对，  我也不知道这啥意思。。。。

AmadeusGhost, 2020-07-20:
> 这里我当时。。。忘记抄上了，但他工作正常-- ，交换机页面端口也对， 我也不知道这啥意思。。。。

这个只是指定 lan/wan mac address

jxph1123, 2020-07-21:
xiaomi R3G 全部重新clone代码后编译，选用的是闭源驱动编译。刷机后用电脑连接，不动态分配IP地址，电脑设置成静态IP，也不行。就进不了管理界面。表面看指示灯都是正常。

openwrt-1983, 2020-07-23:
我一直觉得吧，应该弄两套  一套跟着官方的内核走 一套就是测试版    这样才有更多的选择

AmadeusGhost, 2020-07-26:
@Aes64X @jxph1123 Fixed, just run `git pull`.

Aes64X, 2020-07-26:
> @Aes64X @jxph1123 Fixed, just run `git pull`.

thk

VergilGao, 2020-07-27:
小米ac2100，在定时重启之后会出现wifi和有线均无法连接外网的情况，但是路由器本身ping baidu.com之类的是可以ping通的。
windows网络检测反馈说是远程计算机拒绝服务。
最后网络设置里重连wan口可以解决。
现在在公司，无法上图，晚上回家再测试一下，会不会复现这个问题。

Aes64X, 2020-07-28:
提醒一下，R3G 的5.4内核 initramfs-kernel 是不能用的，可以通过一个4.14内核 initramfs-kernel刷入5.4内核的固件。
直接使用5.4内核 initramfs-kernel 上电启动直接卡死

maxmib, 2020-07-30:
> 小米ac2100，在定时重启之后会出现wifi和有线均无法连接外网的情况，但是路由器本身ping baidu.com之类的是可以ping通的。
> windows网络检测反馈说是远程计算机拒绝服务。
> 最后网络设置里重连wan口可以解决。个人认为是交换机功能的故障。
> 现在在公司，无法上图，晚上回家再测试一下，会不会复现这个问题。

极路由hc5661有类似问题，看了一下是路由重启后wan口dhcp获得的ip地址不对，重连wan口后分配了正确的地址就正常了

1715173329, 2020-08-05:
X-Wrt 那边似乎新引入了有关无线的 HWNAT 驱动，据称使用开源无线驱动可以达到 600 Mbps。

VergilGao, 2020-08-06:
> > 小米ac2100，在定时重启之后会出现wifi和有线均无法连接外网的情况，但是路由器本身ping baidu.com之类的是可以ping通的。
> > windows网络检测反馈说是远程计算机拒绝服务。
> > 最后网络设置里重连wan口可以解决。个人认为是交换机功能的故障。
> > 现在在公司，无法上图，晚上回家再测试一下，会不会复现这个问题。
> 
> 极路由hc5661有类似问题，看了一下是路由重启后wan口dhcp获得的ip地址不对，重连wan口后分配了正确的地址就正常了

目前有自动化的解决的办法吗？比如设置一个开机脚本之类的？因为这个问题我现在关掉了路由器的自动重启功能。但是电信网还是需要定期重启否则会有问题。

AmadeusGhost, 2020-08-06:
> X-Wrt 那边似乎新引入了有关无线的 HWNAT 驱动，据称使用开源无线驱动可以达到 600 Mbps。

那个是用 pt52 大佬他自己的 natflow 实现的，不是用的 flow offload

maxmib, 2020-08-06:
> > > 小米ac2100，在定时重启之后会出现wifi和有线均无法连接外网的情况，但是路由器本身ping baidu.com之类的是可以ping通的。
> > > windows网络检测反馈说是远程计算机拒绝服务。
> > > 最后网络设置里重连wan口可以解决。个人认为是交换机功能的故障。
> > > 现在在公司，无法上图，晚上回家再测试一下，会不会复现这个问题。
> > 
> > 极路由hc5661有类似问题，看了一下是路由重启后wan口dhcp获得的ip地址不对，重连wan口后分配了正确的地址就正常了
> 
> 目前有自动化的解决的办法吗？比如设置一个开机脚本之类的？因为这个问题我现在关掉了路由器的自动重启功能。但是电信网还是需要定期重启否则会有问题。

重新编译固件和重新配置后，发现是删除了wan6接口后就会出现这种情况

VergilGao, 2020-08-07:
> 重新编译固件和重新配置后，发现是删除了wan6接口后就会出现这种情况

谢谢，我晚上下班回去试一下。

jdjingdian, 2020-08-09:
小娱c1/c5什么时候才能用上呢

Aes64X, 2020-08-09:
> 小娱c1/c5什么时候才能用上呢

已经支持了。我这已经测试完成了，不过看最近小娱涨的挺快。算了，就不提交了。也建议各位大佬，看哪个设备涨得快就自己玩玩得了。（商家付费开发的不在此范围内）

jdjingdian, 2020-08-10:
> > 小娱c1/c5什么时候才能用上呢
> 
> 已经支持了。我这已经测试完成了，不过看最近小娱涨的挺快。算了，就不提交了。也建议各位大佬，看哪个设备涨得快就自己玩玩得了。（商家付费开发的不在此范围内）

大大可以悄悄更新吗，现在用着c1感觉负载好高，家里宽带也就300M

coolsnowwolf, 2020-09-04:
[https://github.com/coolsnowwolf/lede/issues/5411](url)

wsltshh, 2020-10-01:
你们有没有LAN经常重连的情况？
我参考其它机型加的新型号，老源码没问题，用最新的源码又参考其它机型改了，编译比较顺利，就是刷上LAN不稳定，以为是无线闭源驱动的事，用开源的还那样。

nanchuci, 2020-10-17:
请问交换机选项该如何配置？

distance369, 2021-03-01:
你好，请问大家遇到开启HWNAT, 后通过tc+iptables   限速失败吗，你是通过什么方式解决的呢？感谢

[3 more replies]

closed 2021-06-20 by coolsnowwolf