coolsnowwolf/lede #5131: RT-N16编译最新固件报错tcping文件变量声明问题

baccoui, 2020-07-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

  RT-N16编译最新固件报错tcping文件变量声明问题

### (2) 路由器型号和固件版本

ausa rt-n16

### (3) 详细日志

make[4]: Entering directory '/home/kevin/lede/build_dir/target-mipsel_74kc_musl/tcpping-0.2'
mipsel-openwrt-linux-musl-gcc -Wall -g tcpping.c -o tcpping -Os -pipe -mno-branch-likely -mips32r2 -mtune=74kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/kevin/lede/build_dir/target-mipsel_74kc_musl/tcpping-0.2=tcpping-0.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/kevin/lede/staging_dir/target-mipsel_74kc_musl/usr/lib/libnet-1.2.x/include -L/home/kevin/lede/staging_dir/target-mipsel_74kc_musl/usr/lib/libnet-1.2.x/lib -lnet -lpcap
tcpping.c: In function 'check_device_name':
tcpping.c:181:2: error: unknown type name 'pcap_if_t'
  pcap_if_t *interface_list = NULL;
  ^~~~~~~~~
tcpping.c:182:2: error: unknown type name 'pcap_if_t'
  pcap_if_t *current_interface = NULL;
  ^~~~~~~~~
tcpping.c:183:14: error: 'PCAP_ERRBUF_SIZE' undeclared (first use in this function); did you mean 'LIBNET_ERRBUF_SIZE'?
  char errbuf[PCAP_ERRBUF_SIZE];
              ^~~~~~~~~~~~~~~~
              LIBNET_ERRBUF_SIZE
tcpping.c:183:14: note: each undeclared identifier is reported only once for each function it appears in
tcpping.c:187:6: warning: implicit declaration of function 'pcap_findalldevs' [-Wimplicit-function-declaration]
  r = pcap_findalldevs(&interface_list, errbuf);
      ^~~~~~~~~~~~~~~~
tcpping.c:202:41: error: request for member 'next' in something not a structure or union
   current_interface = current_interface -> next ) {
                                         ^~
tcpping.c:204:32: error: request for member 'name' in something not a structure or union
   if (strncmp(current_interface->name, device_name, strlen(current_interface->name)) == 0
                                ^~
tcpping.c:204:77: error: request for member 'name' in something not a structure or union
   if (strncmp(current_interface->name, device_name, strlen(current_interface->name)) == 0
                                                                             ^~
tcpping.c:205:43: error: request for member 'name' in something not a structure or union
    && device_name[strlen(current_interface->name)] == '\0') {
                                           ^~
tcpping.c:206:4: warning: implicit declaration of function 'pcap_freealldevs' [-Wimplicit-function-declaration]
    pcap_freealldevs(interface_list);
    ^~~~~~~~~~~~~~~~
tcpping.c:183:7: warning: unused variable 'errbuf' [-Wunused-variable]
  char errbuf[PCAP_ERRBUF_SIZE];
       ^~~~~~
tcpping.c: At top level:
tcpping.c:250:66: warning: 'struct pcap_pkthdr' declared inside parameter list will not be visible outside of this definition or declaration
 void show_packet(struct ip *ip, struct tcphdr *tcp, const struct pcap_pkthdr *header, const u_char *packet)
                                                                  ^~~~~~~~~~~
tcpping.c: In function 'show_packet':
tcpping.c:287:30: error: dereferencing pointer to incomplete type 'const struct pcap_pkthdr'
  printf(" Length: %u", header->caplen);
                              ^~
tcpping.c: At top level:
tcpping.c:328:44: warning: 'struct pcap_pkthdr' declared inside parameter list will not be visible outside of this definition or declaration
 void got_packet(u_char *args, const struct pcap_pkthdr *header, const u_char *packet)
                                            ^~~~~~~~~~~
tcpping.c: In function 'got_packet':
tcpping.c:356:70: error: dereferencing pointer to incomplete type 'const struct pcap_pkthdr'
  for (frame_offset = 0; (frame_offset + size_ip + size_tcp) <= header->caplen; ++frame_offset) {
                                                                      ^~
tcpping.c:368:57: warning: passing argument 3 of 'show_packet' from incompatible pointer type [-Wincompatible-pointer-types]
   show_packet(ip, ip->ip_p == IPPROTO_TCP ? tcp : NULL, header, packet);
                                                         ^~~~~~
tcpping.c:250:79: note: expected 'const struct pcap_pkthdr *' but argument is of type 'const struct pcap_pkthdr *'
 void show_packet(struct ip *ip, struct tcphdr *tcp, const struct pcap_pkthdr *header, const u_char *packet)
                                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
tcpping.c: In function 'sniff_packets':
tcpping.c:516:3: error: unknown type name 'pcap_t'; did you mean 'quad_t'?
   pcap_t *handle;
   ^~~~~~
   quad_t
tcpping.c:517:15: error: 'PCAP_ERRBUF_SIZE' undeclared (first use in this function); did you mean 'LIBNET_ERRBUF_SIZE'?
   char errbuf[PCAP_ERRBUF_SIZE];
               ^~~~~~~~~~~~~~~~
               LIBNET_ERRBUF_SIZE
tcpping.c:519:22: error: storage size of 'filter' isn't known
   struct bpf_program filter;
                      ^~~~~~
tcpping.c:520:3: error: unknown type name 'bpf_u_int32'
   bpf_u_int32 mask;
   ^~~~~~~~~~~
tcpping.c:521:3: error: unknown type name 'bpf_u_int32'
   bpf_u_int32 net;
   ^~~~~~~~~~~
tcpping.c:523:7: warning: implicit declaration of function 'pcap_lookupnet' [-Wimplicit-function-declaration]
   r = pcap_lookupnet(device_name, &net, &mask, errbuf);
       ^~~~~~~~~~~~~~
tcpping.c:529:12: warning: implicit declaration of function 'pcap_open_live'; did you mean 'libnet_open_link'? [-Wimplicit-function-declaration]
   handle = pcap_open_live(device_name, BUFSIZ, 0, 0, errbuf);
            ^~~~~~~~~~~~~~
            libnet_open_link
tcpping.c:540:7: warning: implicit declaration of function 'pcap_setnonblock' [-Wimplicit-function-declaration]
   r = pcap_setnonblock(handle, 1, errbuf);
       ^~~~~~~~~~~~~~~~
tcpping.c:557:6: warning: implicit declaration of function 'pcap_compile'; did you mean 'icmp_otime'? [-Wimplicit-function-declaration]
  r = pcap_compile(handle, &filter, filter_expression, 0, mask);
      ^~~~~~~~~~~~
      icmp_otime
tcpping.c:559:41: warning: implicit declaration of function 'pcap_geterr' [-Wimplicit-function-declaration]
   fprintf(stderr, "pcap_compile: %s\n", pcap_geterr(handle));
                                         ^~~~~~~~~~~
tcpping.c:559:35: warning: format '%s' expects argument of type 'char *', but argument 3 has type 'int' [-Wformat=]
   fprintf(stderr, "pcap_compile: %s\n", pcap_geterr(handle));
                                  ~^     ~~~~~~~~~~~~~~~~~~~
                                  %d
tcpping.c:563:6: warning: implicit declaration of function 'pcap_setfilter' [-Wimplicit-function-declaration]
  r = pcap_setfilter(handle, &filter);
      ^~~~~~~~~~~~~~
tcpping.c:565:37: warning: format '%s' expects argument of type 'char *', but argument 3 has type 'int' [-Wformat=]
   fprintf(stderr, "pcap_setfilter: %s\n", pcap_geterr(handle));
                                    ~^     ~~~~~~~~~~~~~~~~~~~
                                    %d
tcpping.c:573:6: warning: implicit declaration of function 'pcap_loop' [-Wimplicit-function-declaration]
  r = pcap_loop(handle, -1, got_packet, NULL);
      ^~~~~~~~~
tcpping.c:575:2: warning: implicit declaration of function 'pcap_close'; did you mean 'posix_close'? [-Wimplicit-function-declaration]
  pcap_close(handle);
  ^~~~~~~~~~
  posix_close
tcpping.c:519:22: warning: unused variable 'filter' [-Wunused-variable]
   struct bpf_program filter;
                      ^~~~~~
tcpping.c:517:8: warning: unused variable 'errbuf' [-Wunused-variable]
   char errbuf[PCAP_ERRBUF_SIZE];

coolsnowwolf, 2020-07-19:
make clean

baccoui, 2020-07-20:
全新环境已经是Ubuntu 64bit Ubuntu 1804，主服务的源， sudo apt-get update 最新了，全程科学上网。重新git clone，默认make menuconfig选项，只改cpu类别前三项编译。报错依旧。

1，问题点一不知道是否这些警告影响
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget swig
除了以下注意都正常
注意，选中 'zlib1g-dev' 而非 'libz-dev'
注意，根据正则表达式 'python3.5' 选中了 'libpython3.5-minimal'
注意，选中 'node-uglify' 而非 'uglifyjs'
注意，选中 'git' 而非 'git-core'
注意，选中 'upx-ucl' 而非 'upx'

2，make -j8 download V=s 有源下载文件不存在了
例1
SHELL= flock /home/kevin/lede/tmp/.wol-0.7.1.tar.gz.flock -c '  	/home/kevin/lede/scripts/download.pl "/home/kevin/lede/dl" "wol-0.7.1.tar.gz" "c2fa9d7e771134ac8c89d56b8197d4ca" "" "@SF/ahh"    '
Can't exec "curl": No such file or directory at /home/kevin/lede/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --no-check-certificate --output-document=- https://sources.cdn.openwrt.org/wol-0.7.1.tar.gz
--2020-07-20 02:51:11--  https://sources.cdn.openwrt.org/wol-0.7.1.tar.gz
Resolving sources.cdn.openwrt.org (sources.cdn.openwrt.org)... 161.202.39.238, 2a0b:4d07:702::1
Connecting to sources.cdn.openwrt.org (sources.cdn.openwrt.org)|161.202.39.238|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-07-20 02:51:13 ERROR 404: Not Found.
例2
make[2]: Entering directory '/home/kevin/lede/package/lean/luci-app-unblockmusic'
/home/kevin/lede/include/package.mk:327: warning: overriding recipe for target 'prepare-package-install'
/home/kevin/lede/include/package.mk:327: warning: ignoring old recipe for target 'prepare-package-install'
/home/kevin/lede/include/package.mk:332: warning: overriding recipe for target '/home/kevin/lede/bin/targets/bcm47xx/mips74k/packages'
/home/kevin/lede/include/package.mk:332: warning: ignoring old recipe for target '/home/kevin/lede/bin/targets/bcm47xx/mips74k/packages'
/home/kevin/lede/include/package.mk:339: warning: overriding recipe for target 'force-clean-build'
/home/kevin/lede/include/package.mk:339: warning: ignoring old recipe for target 'force-clean-build'
/home/kevin/lede/include/package.mk:344: warning: overriding recipe for target 'clean'
/home/kevin/lede/include/package.mk:344: warning: ignoring old recipe for target 'clean'
/home/kevin/lede/include/package.mk:350: warning: overriding recipe for target 'dist'
/home/kevin/lede/include/package.mk:350: warning: ignoring old recipe for target 'dist'
/home/kevin/lede/include/package.mk:353: warning: overriding recipe for target 'distcheck'
/home/kevin/lede/include/package.mk:353: warning: ignoring old recipe for target 'distcheck'
make[2]: Nothing to be done for 'download'.

3
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, char *, int, int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, char *, int, unsigned int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, char *, size_t, int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, char *, size_t, unsigned int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, char *, socklen_t, int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, char *, socklen_t, unsigned int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, char *, unsigned int, int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, char *, unsigned int, unsigned int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, void *, int, int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, void *, int, unsigned int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, void *, size_t, int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed
-- Tested: int recv(SOCKET, void *, size_t, unsigned int)
-- Performing Test curl_cv_func_recv_test
-- Performing Test curl_cv_func_recv_test - Failed

closed 2021-06-20 by coolsnowwolf