coolsnowwolf/lede #5141: 能获取ipv6 pd。 但无法分配ip给内网。

zerofo, 2020-07-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
能获取ipv6 pd。 但无法分配ip给内网。
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刚装完系统 把 wan6删除后， br-lan能获取ipv6 ::1/64的。  内网机器也都能分配 ipv6. 不知道哪一个时候 ipv6 就没了 只剩下 br-lan的。

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
/etc/config/dhcp 
```
config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '2d'
        option ra_management '1'
        option ra 'server'
        option dhcpv6 'server'
        list dns '2400:3200:baba::1'
        list dns '2400:da00::6666'
        list dns '240C::6666 240C::6644'
        list dns '2001:4860:4860::8888'
        list dns '2001:4860:4860::8844'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'
        option master '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/etc/odhcp/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'

config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option nonwildcard '1'
        option localservice '1'
        option confdir '/tmp/dnsmasq.d'
        option port '53'
        option resolvfile '/tmp/resolv.conf.auto'
        list server '127.0.0.1#5333'
        option filter_aaaa '1'
        option logqueries '1'
```
/etc/config/network
```
config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '1'
        option ula_prefix 'fd80::/128'

config interface 'lan'
        option type 'bridge'
        option ifname 'eth0 eth2 eth3 eth4 eth5'
        option proto 'static'
        option netmask '255.0.0.0'
        option ipaddr '192.168.0.1'
        option gateway '192.168.0.1'
        option stp '1'
        option igmp_snooping '1'
        option ip6assign '64'

config interface 'wan'
        option ifname 'eth1'
        option _orig_ifname 'eth1'
        option _orig_bridge 'false'
        option username 'xxxxxxxxxxx'
        option password 'xxxxxx'
        option keepalive '0'
        option ipv6 'auto'
        option metric '40'
        option proto 'pppoe'
        option peerdns '0'
```

logread|grep dhcp   没有结果
logread|grep dns  只有ddns 不时报错

minlang112, 2020-07-19:
为什么要删除WAN6？删除之后IPV6无法正常使用吧

zerofo, 2020-07-19:
> 为什么要删除WAN6？删除之后IPV6无法正常使用吧

不删除也不行

zerofo, 2020-07-19:
> 为什么要删除WAN6？删除之后IPV6无法正常使用吧

加回去也不行 dhcpv6客户端， 
 首页状态  DHCPv6 分配 
显示     
没有分配租约

zerofo, 2020-07-19:
> > 为什么要删除WAN6？删除之后IPV6无法正常使用吧
> 
> 加回去也不行 （协议dhcpv6客户端，)
> 首页状态 DHCPv6 分配
> 显示
> 没有分配租约

zerofo, 2020-07-19:
> 为什么要删除WAN6？删除之后IPV6无法正常使用吧

删除 br-lan也有ipv6

keivenliao, 2020-07-20:
贴个我的设置给你参考一下，
[image]
[image]
[image]
[image]
[image]

zerofo, 2020-07-20:
> 贴个我的设置给你参考一下，
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> 
> 
> ------------------------------------------------------------------------------------------------------------------------------------------
> 
> 
> 
> 
> 
> 
> 
> [image]
> [image]
> 

我等等逐个对一下吧。内置我都有勾了

zerofo, 2020-07-20:
> 为什么要删除WAN6？删除之后IPV6无法正常使用吧

[image]
[image]
如图

zerofo, 2020-07-20:
> 为什么要删除WAN6？删除之后IPV6无法正常使用吧

如图
[image]

[image]

zerofo, 2020-07-20:
> 贴个我的设置给你参考一下，
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> 
> [image]
> [image]

区别比较大的是 防火墙那两条  我这边不是任意是 icmp 不过我添加后还是不行。。。 还有一个是 lan 物理设置  我开启stp 和 igmp 关闭后也是不行

Twinzo1, 2020-07-20:
是不是开了负载均衡

zerofo, 2020-07-20:
> 是不是开了负载均衡

不清楚 怎么关闭？？

minlang112, 2020-07-20:
> > 是不是开了负载均衡
> 
> 不清楚 怎么关闭？？

停用mwan3和mwan3 helper

zerofo, 2020-07-20:
> helper

是在 启动项里面关闭吗？   网络-负载均衡 全局好像没有 关闭选项。。 。我之前把规则全都删除， 并在启动项里面关闭， 关闭后无法联网

Twinzo1, 2020-07-20:
负载均衡→接口 禁用

zerofo, 2020-07-20:
> 负载均衡→接口 禁用

[image]
[image]
没看见有禁用= =

zerofo, 2020-07-20:
> 负载均衡→接口 禁用

/etc/init.d/mwan3 stop 后直接断网

Remonli, 2020-07-20:
OpenWRT的ipv6好像一直挺有问题的,我也折腾了好几天没搞好。

Twinzo1, 2020-07-20:
点击修改，启用改为否

Twinzo1, 2020-07-21:
> > 点击修改，启用改为否
> 
> 否…然后 上不了网页了。dns……

只能说，你的设置有问题吧

zerofo, 2020-07-21:
> > > 点击修改，启用改为否
> > 
> > 否…然后 上不了网页了。dns……
> 
> 只能说，你的设置有问题吧

………而且最主要是我没动负载均衡，但是至少ipv6能分配

Cathgao, 2020-07-26:
用ipv6的话建议不要编译也不要安装mwan，这东西装上问题很多

zerofo, 2020-07-31:
> 用ipv6的话建议不要编译也不要安装mwan，这东西装上问题很多

额..这样.. 就排除了未来对自己发财了装双千兆的可能性.....的憧憬....的梦(白天的)

Benjamin23333, 2021-01-02:
> > 用ipv6的话建议不要编译也不要安装mwan，这东西装上问题很多
> 
> 额..这样.. 就排除了未来对自己发财了装双千兆的可能性.....的憧憬....的梦(白天的)

把负载均衡策略那里改成默认使用主路由就好了

hcym, 2021-01-02:
官方的没什么事，无论光猫拨号还是路由拨号，大雕的最好不用ipv6

closed 2021-06-20 by coolsnowwolf