coolsnowwolf/lede #5144: ssr libev 编译不通过

z826540272, 2020-07-20:
cc1: all warnings being treated as errors
Makefile:777: recipe for target 'libshadowsocks_libev_la-acl.lo' failed
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/src'
make[6]: *** [libshadowsocks_libev_la-acl.lo] Error 1
make[5]: *** [all-recursive] Error 1
Makefile:479: recipe for target 'all-recursive' failed
make[4]: *** [all] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543'
Makefile:388: recipe for target 'all' failed
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543'
Makefile:67: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocksr-libev//shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/.built' failed
make[3]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocksr-libev//shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/lean/shadowsocksr-libev'
time: package/lean/shadowsocksr-libev/compile#0.40#0.06#0.48
package/Makefile:111: recipe for target 'package/lean/shadowsocksr-libev/compile' failed

设备，mt7621，请求revert 11848e8b4309434786e95a289f37df309061fac4
此提交。

z826540272, 2020-07-20:
使用github ubuntu18 云编译

z826540272, 2020-07-20:
@AmadeusGhost 
lienol的openwrt编译ssr libev是通过的，lean的不通过，每次报错在这里
请求撤销ssr libev编译器选项优化

AmadeusGhost, 2020-07-20:
这个编译优化也不是我发的吧，你可以发个 pr revert 回去

closed 2020-07-20 by coolsnowwolf