coolsnowwolf/lede #5145: wr720n

natserver, 2020-07-20:
1.我编译了tp的wr720n的固件，提示成功却不能用。

2.编译完成全程没有报错不管云编译还是本地编译，完成后只有一个4m的bin但是用breed也刷不进去路由。测试了v3和v4版本编译完成都是没有固件。

路由器型号：tp-link wr720nv4版本

编译完成所有的文件：
[image]

asushugo, 2020-07-20:
固件太大啦，需要精简一下

jyy5206, 2020-07-20:
720n 我买过一个
发现只有 4M flash
只能装 Openwrt 的官方固件
装完后已经几乎 4M 占满了
啥也加不了了

而这里编译的出来的固件都很大
都超出了 720n 的缺省 4M 设定
所以要么出错
要么干脆就是不生成固件了。。。

后来淘宝上找了个店铺
硬改到 16+64

这里编译的固件
编译选项里已经有了 720n 16M 的选项
编译出来刷机即可

不过 720n 马力实在是太肉了
启动需要 3 分钟
而且 V2Ray 也拖不动
只能用 SS - SSR - Trojan
勉强达到 9 Mbps

不过出差时酒店房间里用用倒是不错~

供参考

xhqppp, 2020-07-22:
缩到3M试试

mondhu, 2020-07-23:
@jyy5206 你好，编译选项里找不到720n 16M选项，在哪里找到的？

jyy5206, 2020-07-23:
@mondhu 
刚才进编译机update了下，发现 720n 16M 这个选项没有了。。。汗。。。
可能管理员移除了

你就用我以前编译好的吧：
https://github.com/jyy5206/TP-link-WR720N-16M-Flush-Firmware

closed 2021-06-20 by coolsnowwolf