coolsnowwolf/lede #5149: openwrt清爽material风格主题分享

garypang13, 2020-07-20:
修改自luci-theme-argon,做了大量细节优化.
截图自Luci19.07,  Luci18.06因为其本分的局限性,显示效果有些许不同,

https://github.com/garypang13/luci-theme-edge/releases

[image]

[image]

[image]

[image]

[image]
<br/>
<br />
[image]

[image]

dolphinpaopao, 2020-07-20:
截图概览cpu信息不对呀，i3 7100u是双核四线程不是四核四线程。哈哈，强迫症max。

garypang13, 2020-07-20:
> 截图概览cpu信息不对呀，i3 7100u是双核四线程不是四核四线程。哈哈，强迫症max。

可能是我套了个PVE的原因吧,

EnnawYang, 2020-07-20:
18.06浏览日志页面还有点问题，不能上下翻滚

dolphinpaopao, 2020-07-20:
> > 截图概览cpu信息不对呀，i3 7100u是双核四线程不是四核四线程。哈哈，强迫症max。
> 
> 可能是我套了个PVE的原因吧,

我现在去编译一个试一下。感谢分享。

OleStore, 2020-07-20:
讲真？花里胡哨的，不喜欢

yubanmeiqin9048, 2020-07-20:
啊，这，为什么要放到Issues

xiaoguoyi, 2020-07-20:
系统日志没法翻页
[image]

xiaoguoyi, 2020-07-20:
退出没有图标
[image]

github-xihe, 2020-07-21:
这主题挺好看的，不过在18.06还需要打磨打磨，期待完美版

OPnb, 2020-07-21:
如果是bootstrap那样就完美了

keivenliao, 2020-07-22:
装一个体验一下

chuansao-258, 2020-07-22:
> 如果是bootstrap那样就完美了

恋丑你就默认就好了

mysteryman2045, 2020-07-22:
Newifi D2，06.30那天编译的L大的固件，直接路由器内部文件上传luci-theme-edge_1.6-20200722_19.07.ipk
到路由器安装，后直接页面全白，连登录页面都是。Google Chrome，Firefox，IE浏览器都不好使，恰恰自己路由器SSH没配置，想SSH进路由器指令修改过来也没机会。不得不刚刚重新刷了固件。😭

garypang13, 2020-07-22:
> Newifi D2，06.30那天编译的L大的固件，直接路由器内部文件上传luci-theme-edge_1.6-20200722_19.07.ipk
> 到路由器安装，后直接页面全白，连登录页面都是。Google Chrome，Firefox，IE浏览器都不好使，恰恰自己路由器SSH没配置，想SSH进路由器指令修改过来也没机会。不得不刚刚重新刷了固件。😭

要确认你用的luci版本,L大的分为luci17,luci18.06,luci19.07,该主题只有luci18.06和luci19.07版本,如果你是Luci17的话就用不了,

mysteryman2045, 2020-07-22:
> > Newifi D2，06.30那天编译的L大的固件，直接路由器内部文件上传luci-theme-edge_1.6-20200722_19.07.ipk
> > 到路由器安装，后直接页面全白，连登录页面都是。Google Chrome，Firefox，IE浏览器都不好使，恰恰自己路由器SSH没配置，想SSH进路由器指令修改过来也没机会。不得不刚刚重新刷了固件。😭
> 
> 要确认你用的luci版本,L大的分为luci17,luci18.06,luci19.07,该主题只有luci18.06和luci19.07版本,如果你是Luci17的话就用不了,

看了下 luci版本git-20.180.30743-de9fdad-1

lloyd18, 2020-07-24:
楼主你好，docker图标有吗
[image]

garypang13, 2020-07-24:
> 楼主你好，docker图标有吗
> [image]

有

stvber, 2020-07-24:
背景换个颜色会好点，太白了

closed 2021-06-20 by coolsnowwolf