coolsnowwolf/lede #5151: k2p重启后5G Wi-Fi停用

DreamWalkerXZ, 2020-07-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：k2p在重启后5G Wi-Fi停用，如图
[image]
要按“重启无线“才行

### (2) 路由器型号和固件版本

A：k2p OpenWrt Development Snapshot / LuCI Master (git-20.191.36863-eee6bae) linux 5.4.51

### (3) 详细日志

A：
[系统日志.log](https://github.com/coolsnowwolf/lede/files/4952530/default.log)
[内核日志.log](https://github.com/coolsnowwolf/lede/files/4952534/default.log)

系统日志
```
Tue Jul 21 17:12:30 2020 kern.notice kernel: [    0.000000] Linux version 5.4.51 (DreamWalkerXZ@dreamwalkerxz.ski) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2831-289e39376)) #0 SMP Mon Jul 20 14:49:03 2020
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] MIPS: machine is Phicomm K2P
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Tue Jul 21 17:12:30 2020 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Jul 21 17:12:30 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Jul 21 17:12:30 2020 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Zone ranges:
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000]   HighMem  empty
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Early memory node ranges
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jul 21 17:12:30 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Tue Jul 21 17:12:30 2020 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Tue Jul 21 17:12:30 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue Jul 21 17:12:30 2020 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
Tue Jul 21 17:12:30 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
Tue Jul 21 17:12:30 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Tue Jul 21 17:12:30 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00060810
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00060810
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] Memory: 120840K/131072K available (5675K kernel code, 248K rwdata, 1232K rodata, 1280K init, 238K bss, 10232K reserved, 0K cma-reserved, 0K highmem)
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Tue Jul 21 17:12:30 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] CPU Clock: 1000MHz
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3822520893 ns
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.000007] sched_clock: 32 bits at 500MHz, resolution 2ns, wraps every 4294967295ns
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.015477] Calibrating delay loop... 663.55 BogoMIPS (lpj=1327104)
Tue Jul 21 17:12:30 2020 kern.info kernel: [    0.059825] pid_max: default: 32768 minimum: 301
[... 1367 lines cut ...]
[  280.819048] b7
[  280.822351] e4
[  280.825655] f9
[  280.828959] 47
[  280.832263] 40
[  280.835557] 01
[  280.838861] c4
[  280.842168] 55
[  280.845471] 24
[  280.848779] 9d
[  280.852072] 58
[  280.855384] eb
[  280.858701] 8c
[  280.862014] 6b
[  280.865318] 15
[  280.868622] da
[  280.871913] 99
[  280.875220] f3
[  280.878524] 77
[  280.881833] a2
[  280.885137] 25
[  280.888441] 
[  281.153823] Rcv Wcid(3) AddBAReq
[  281.160272] Start Seq = 00000002
```

EnnawYang, 2020-07-21:
接口lan里面物理设置有没有添加ra0？

DreamWalkerXZ, 2020-07-21:
> 接口lan里面物理设置有没有添加ra0？

加完之后ok了，谢谢

closed 2020-07-21 by DreamWalkerXZ