coolsnowwolf/lede #5154: IPV6 地址获取不了

Guxing0321, 2020-07-21:
Hi Team,

我使用默认的配置获取不到ipv6 地址

[image]

但是这个接口又产生了流量
[image]

请问，可以提供下ipv6 相关的配置步骤吗，我在晚上找了很多，都没有效果
不知道是不是我的步骤有问题

BR
Robin

hyy-666, 2020-07-21:
地址应该在LAN口，但是你这个是内网用的地址，可以的话试一下编译的时候在network里选中6rd

xiehuayi75, 2020-07-21:
IPV6设置
1  接口：删除ipv6前缀
2  wan口：高级;ipv6-address:手动
3  新建ipv6接口: 协议:DHCPv6客户端   物理接口：@wan
4  DHCP-DNS ： 高级设置：取消过滤ipv6
5  防火墙： 转发-接受   区域转发： WAN-全接受    LAN覆盖网络：ipv6 lan  端口转发：目标区域wan 源区域wan
6  负载均衡： 策略：balanced  修改备用成员：使用主路由表

ldgwfyf, 2020-07-22:
> IPV6设置
> 1 接口：删除ipv6前缀
> 2 wan口：高级;ipv6-address:手动
> 3 新建ipv6接口: 协议:DHCPv6客户端 物理接口：@wan
> 4 DHCP-DNS ： 高级设置：取消过滤ipv6
> 5 防火墙： 转发-接受 区域转发： WAN-全接受 LAN覆盖网络：ipv6 lan 端口转发：目标区域wan 源区域wan
> 6 负载均衡： 策略：balanced 修改备用成员：使用主路由

软路由可以，硬路由不见得好用，至少我的K2P上就不灵了

lqhorochi, 2020-08-02:
> IPV6设置
> 1 接口：删除ipv6前缀
> 2 wan口：高级;ipv6-address:手动
> 3 新建ipv6接口: 协议:DHCPv6客户端 物理接口：@wan
> 4 DHCP-DNS ： 高级设置：取消过滤ipv6
> 5 防火墙： 转发-接受 区域转发： WAN-全接受 LAN覆盖网络：ipv6 lan 端口转发：目标区域wan 源区域wan
> 6 负载均衡： 策略：balanced 修改备用成员：使用主路由表

如果是旁路由，如何设置ipv6呢？请教一下

hcym, 2020-08-02:
跟旁路没关系，大雕的wan6只能在网络里面看到，概览看不到，不影响使用ipv6节点，节点状态只有ping值，连接失败，kcp的节点好像也有这个情况，显示失败使用正常就是啦

closed 2021-06-20 by coolsnowwolf