coolsnowwolf/lede #5158: Kernel 5.4 后自带的 dnsmasq resolvfile 默认路径已改，相关app得做下判断

niphor, 2020-07-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新的Kernel 5.4 版本 dnsmasq 也做了相应的修改：`/tmp/resolv.conf.auto` 不再使用，改为了 `/tmp/resolv.conf.d/resolv.conf.auto`,并增加了相应脚本`package/network/services/dnsmasq/files/50-dnsmasq-migrate-resolv-conf-auto.sh`做初始化修正

但是 相应的包
luci-app-sfe
luci-app-flowoffload
dnsforwarder
没有对此做适配，启用、禁用DNS优化，会直接设置为`/tmp/resolv.conf.auto` 导致路由器dnsmasq 没有上游resolver，使得DNS解析异常

### (2) 路由器型号和固件版本

A：ER-X, OpenWrt R20.7.20

### (3) 详细日志

A：无

ostrichii, 2020-07-23:
https://github.com/coolsnowwolf/lede/issues/5122
R2S都用5.40+，所以反馈得多，正好自己也有，求大雕修

niphor, 2020-07-23:
@ostrichii 昨天已经有人改了 也merge了，你可以重新编译看看，或者 直接 对应commit 改下现有的脚本，也就替换的事情，简单的

ostrichii, 2020-07-23:
> @ostrichii 昨天已经有人改了 也merge了，你可以重新编译看看，或者 直接 对应commit 改下现有的脚本，也就替换的事情，简单的

多谢，我看看记录，那我昨晚重新编译的应该是已经解决了的

ostrichii, 2020-07-23:
> > @ostrichii 昨天已经有人改了 也merge了，你可以重新编译看看，或者 直接 对应commit 改下现有的脚本，也就替换的事情，简单的
> 
> 多谢，我看看记录，那我昨晚重新编译的应该是已经解决了的

看昨晚编译的源码是在最新merge的代码后面，应该没这个问题了

closed 2020-07-23 by coolsnowwolf