coolsnowwolf/lede #5160: 新版固件   选择NODEJS  UnblockNeteaseMusic 偶尔未运行

tcsr200722, 2020-07-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
无
## 2. 详细叙述

### (1) 具体问题

A：NODEJS  UnblockNeteaseMusic 偶尔未运行

### (2) 路由器型号和固件版本

A：X86    OpenWrt R20.7.7 built by Lean 

### (3) 详细日志

A：Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.777221] pci 0000:00:17.5: PCI bridge to [bus 18]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.777422] pci 0000:00:17.5:   bridge window [mem 0xfbe00000-0xfbefffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.777662] pci 0000:00:17.5:   bridge window [mem 0xe6900000-0xe69fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.777970] pci 0000:00:17.6: PCI bridge to [bus 19]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.778184] pci 0000:00:17.6:   bridge window [mem 0xfba00000-0xfbafffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.778425] pci 0000:00:17.6:   bridge window [mem 0xe6500000-0xe65fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.778733] pci 0000:00:17.7: PCI bridge to [bus 1a]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.778933] pci 0000:00:17.7:   bridge window [mem 0xfb600000-0xfb6fffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.779173] pci 0000:00:17.7:   bridge window [mem 0xe6100000-0xe61fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.779522] pci 0000:00:18.0: PCI bridge to [bus 1b]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.779704] pci 0000:00:18.0:   bridge window [io  0x7000-0x7fff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.779934] pci 0000:00:18.0:   bridge window [mem 0xfd100000-0xfd1fffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.780174] pci 0000:00:18.0:   bridge window [mem 0x40800000-0x409fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.780481] pci 0000:00:18.1: PCI bridge to [bus 1c]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.780660] pci 0000:00:18.1:   bridge window [io  0xa000-0xafff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.780891] pci 0000:00:18.1:   bridge window [mem 0xfcd00000-0xfcdfffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.781131] pci 0000:00:18.1:   bridge window [mem 0xe7800000-0xe78fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.781438] pci 0000:00:18.2: PCI bridge to [bus 1d]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.781618] pci 0000:00:18.2:   bridge window [io  0xe000-0xefff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.781849] pci 0000:00:18.2:   bridge window [mem 0xfc900000-0xfc9fffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.782088] pci 0000:00:18.2:   bridge window [mem 0xe7400000-0xe74fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.782412] pci 0000:00:18.3: PCI bridge to [bus 1e]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.782613] pci 0000:00:18.3:   bridge window [mem 0xfc500000-0xfc5fffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.782852] pci 0000:00:18.3:   bridge window [mem 0xe7000000-0xe70fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.783161] pci 0000:00:18.4: PCI bridge to [bus 1f]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.783362] pci 0000:00:18.4:   bridge window [mem 0xfc100000-0xfc1fffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.783601] pci 0000:00:18.4:   bridge window [mem 0xe6c00000-0xe6cfffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.783913] pci 0000:00:18.5: PCI bridge to [bus 20]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.784114] pci 0000:00:18.5:   bridge window [mem 0xfbd00000-0xfbdfffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.784354] pci 0000:00:18.5:   bridge window [mem 0xe6800000-0xe68fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.784664] pci 0000:00:18.6: PCI bridge to [bus 21]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.784863] pci 0000:00:18.6:   bridge window [mem 0xfb900000-0xfb9fffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.785102] pci 0000:00:18.6:   bridge window [mem 0xe6400000-0xe64fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.785410] pci 0000:00:18.7: PCI bridge to [bus 22]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.785611] pci 0000:00:18.7:   bridge window [mem 0xfb500000-0xfb5fffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.785851] pci 0000:00:18.7:   bridge window [mem 0xe6000000-0xe60fffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.786174] pci_bus 0000:00: resource 4 [mem 0x000a0000-0x000bffff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.786395] pci_bus 0000:00: resource 5 [mem 0x000cc000-0x000cffff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.786614] pci_bus 0000:00: resource 6 [mem 0x000d0000-0x000d3fff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.786832] pci_bus 0000:00: resource 7 [mem 0x000d4000-0x000d7fff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.787052] pci_bus 0000:00: resource 8 [mem 0x000d8000-0x000dbfff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.787271] pci_bus 0000:00: resource 9 [mem 0x40000000-0xfebfffff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.787490] pci_bus 0000:00: resource 10 [io  0x0000-0x0cf7 window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.787692] pci_bus 0000:00: resource 11 [io  0x0d00-0xfeff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.787896] pci_bus 0000:02: resource 0 [io  0x2000-0x3fff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.788078] pci_bus 0000:02: resource 1 [mem 0xfd500000-0xfdffffff]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.788281] pci_bus 0000:02: resource 2 [mem 0xe7b00000-0xe7ffffff 64bit pref]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.788525] pci_bus 0000:02: resource 4 [mem 0x000a0000-0x000bffff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.788743] pci_bus 0000:02: resource 5 [mem 0x000cc000-0x000cffff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.790116] pci_bus 0000:02: resource 6 [mem 0x000d0000-0x000d3fff window]
Wed Jul 22 13:30:22 2020 kern.info kernel: [    0.790116] pci_bus 0000:02: resource 7 [mem 0x000d4000-0x000d7fff window]
[... 557 lines cut ...]
Wed Jul 22 13:30:41 2020 daemon.notice procd: /etc/rc.d/S99serverchan: serverchan is starting now ...
Wed Jul 22 13:30:41 2020 daemon.notice ttyd[8264]: ttyd 1.6.0-c0f1faf (libwebsockets 3.2.0)
Wed Jul 22 13:30:41 2020 daemon.notice ttyd[8264]: tty configuration:
Wed Jul 22 13:30:41 2020 daemon.notice ttyd[8264]:   start command: /usr/libexec/login.sh
Wed Jul 22 13:30:41 2020 daemon.notice ttyd[8264]:   close signal: SIGHUP (1)
Wed Jul 22 13:30:41 2020 daemon.notice ttyd[8264]:   terminal type: xterm-256color
Wed Jul 22 13:30:41 2020 daemon.notice ttyd[8264]:  Using foreign event loop...
Wed Jul 22 13:30:41 2020 daemon.notice ttyd[8264]:  Listening on port: 7681
Wed Jul 22 13:30:42 2020 user.info wsdd2[8422]: starting.
Wed Jul 22 13:30:42 2020 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Wed Jul 22 13:30:42 2020 daemon.info procd: - init complete -
Wed Jul 22 13:30:53 2020 cron.err crond[10490]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 13:30:53 2020 cron.err crond[10605]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:04:25 2020 cron.err crond[28863]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:04:25 2020 cron.err crond[28973]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:04:55 2020 cron.err crond[29428]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:04:55 2020 cron.err crond[29539]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:05:24 2020 cron.err crond[29859]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:05:24 2020 cron.err crond[29969]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:05:54 2020 cron.err crond[30873]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:05:54 2020 cron.err crond[30983]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:06:23 2020 cron.err crond[31396]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:06:24 2020 cron.err crond[31506]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:06:53 2020 cron.err crond[2902]: crond (busybox 1.31.1) started, log level 8
Wed Jul 22 15:06:53 2020 cron.err crond[3129]: crond (busybox 1.31.1) started, log level 8

tcsr200722, 2020-07-22:
补充一下  golang版本没有问题

jasonseye, 2020-08-09:
编译了一样的x86 64bit版本，出现了一样的问题，Nodejs无法运行，但是Golang就可以正常运行。
以下是运行Nodejs后的错误信息
root@OpenWrt:/usr/share/UnblockNeteaseMusic# node ./app.js
/usr/share/UnblockNeteaseMusic/src/hook.js:107
                                // console.log(netease.path, netease.param)

SyntaxError: Unexpected end of input
    at wrapSafe (internal/modules/cjs/loader.js:1072:16)
    at Module._compile (internal/modules/cjs/loader.js:1122:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1178:10)    at Module.load (internal/modules/cjs/loader.js:1002:32)
    at Function.Module._load (internal/modules/cjs/loader.js:901:14)
    at Module.require (internal/modules/cjs/loader.js:1044:19)
    at require (internal/modules/cjs/helpers.js:77:18)
    at Object.<anonymous> (/usr/share/UnblockNeteaseMusic/src/app.js:56:14)
    at Module._compile (internal/modules/cjs/loader.js:1158:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1178:10)
root@OpenWrt:/usr/share/UnblockNeteaseMusic#

closed 2021-06-20 by coolsnowwolf

Kisoul, 2021-09-15:
> 补充一下 golang版本没有问题

你这个问题有解决吗？