coolsnowwolf/lede #5164: 通过Softether VPN连接到路由器的设备无ipv6地址

hcwang, 2020-07-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
按照#5054 介绍的方式设置了IPV6，所有无线客户端均可以正常得到IPV6地址，并且可以通过ipv6测试，正常上网。但使用softether VPN搭建了服务器（tap device方式, [搭建步骤](https://www.iloft.xyz/archives/softether-l2tp-openwrt.html))，通过VPN方式连上路由器的客户端只能获得ipv4地址，但无法得到ipv6地址（无论是L2TP还是OpenVPN模式)。
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：小米路由器R3P

### (3) 详细日志

A：

keivenliao, 2020-07-23:
[image]
[image]

理论上不会有你说的问题。

hcwang, 2020-07-23:
> [image]
> [image]
> 
> 理论上不会有你说的问题。

我使用的是iOS设备，使用无线方式连接的时候会有IPv6地址，但使用OpenVPN或者L2TP的时候就没有。

hcwang, 2020-07-23:
> [image]
> [image]
> 
> 理论上不会有你说的问题。

我使用Windows也测试了一下，仍然没有IPv6地址，建立的是L2TP VPN。

请问你使用提tap 方式建立的Softether VPN吗？还是直接使用的物理网卡？我看到你的网络连接信息中有Mac地址，而我的没有。

keivenliao, 2020-07-23:
我用tap桥接的方式

keivenliao, 2020-07-23:
[image]

keivenliao, 2020-07-23:
不过话说回来，我目前只是在家里的设备分配了IPV6地址，方便外网访问，但整个网络我都禁用了ipv6的解析的，因为有些IPV6的访问真的很慢
[image]

hcwang, 2020-07-23:
和我有一个不一样的地方，我在桥接LAN上还有无线，难道是因为这个问题？
[image]

hcwang, 2020-07-23:
另外，我在softether vpn的github上发现了同样的[issue](https://github.com/SoftEtherVPN/SoftEtherVPN/issues/244):。

coolsnowwolf, 2020-07-23:
这部分是 softether 管理的，和 op 无关

hcwang, 2020-07-24:
> 这部分是 softether 管理的，和 op 无关

谢谢，但softether 也是按照标准进行设置的，感觉没有错过哪个步骤。所以看到楼上老兄竟然配成功了，想请教一下。

sanheho, 2021-02-25:
Salam aalikom moataz
I've been trying to complete a project for my end of study on the institute of OFPPT in morocco and i realised that vpn is the best project i can make in this period, i am a networking training and i found that softether vpn can do me more than i was wish 
I liked to idee of this solution and i start working on it since 3 month ago i have create a remote access by L2. Now i'm trying to make a layer 3 connection between different sites and completed most of it,
1. I'm using an OVH cloud vps on a linux OS as the main softether vpn server 
2. On my local site (SITE1) I'm using a windows 10 as a server manager and another pc as a bridge that cascading to a hub that I create on the server and the same configuration on the other site (SITE2).

Here is my configuration that i make so far 

On The server side: 
1. Virtual hubs :

1.1 for the virtual hub (SITE1)

* SECURENAT :Enabled
* ip address : 192.168.30.1/24
* DHCP range : 192.168.30.10/24 TO 200
* Lease time : 7200
* default Gateway : 192.168.30.1
* DNS server : 192.168.30.1
* MTU             : 1500
* TCP session : 1800
* UDP session : 60
And for static route table to push i did add  192.168.40.0/255.255.255.0/192.168.30.254

1.2 for the virtual hub (SITE2)
* SECURENAT :Enabled
* ip address : 192.168.40.1/24
* DHCP range : 192.168.40.10/24 TO 200
* Lease time : 7200
* default Gateway : 192.168.40.1
* DNS server : 192.168.40.1
* MTU             : 1500
* TCP session : 1800
* UDP session : 60
And for static route table to push i did add  192.168.30.0/255.255.255.0/192.168.40.254, 198.168.1.0/255.255.255.0/192.168.40.253

2. In Layer 3 switching setting i created one virtual layer 3 switch with two virtual interfaces for each virtual hub

* Virtual interface site 1 : 192.168.30.254/24
* Virtual interface site 2 : 192.168.40.254/24  

with no routing table

3. for the local bridge setting i haven't add anything,

4. VPN azure is disable,

5. DDNS it enable,

6. IPsec / L2TP are enable

On the bridges side:
ON SITE1 :

1. Cascading :

On the bridge one virtual hub is created by default and i cascade from the virtual hub on bridge to the virtual hub that i create already on the server side 
the configuration like this
* Setting name : site012hq
* Host name     : SERVER IP
* Port  number  : 443
* virtual hub name : HQ
* user authentication : user created on the server site

  The status is online

2. SECURENAT setting :

SECURENAT :Disabled
* ip address : 192.168.30.1/24
* DHCP range : 192.168.30.10/24 TO 200
* Lease time : 7200
* default Gateway : 192.168.30.1
* DNS server : 192.168.30.1
* MTU             : 1500
* TCP session : 1800
* UDP session : 60

And for static route table to push i haven't add any static route table

3. LOCAL BRIDGE settings: 

For the local bridge settings I add an usb network adapter and connect it to the internet, and the integrated adapter on the pc is connected from the pc to a switch POE and from that switch to the clients.

That all the configuration that i make i haven't add any port forward or static route on the router or the clients also on the firewall.
i think a least it should ping between two sites and i work for 1 day but sometimes i get ip address of my ISP not VPN IP i don't know why that happened and other problem is when the clients are connect to vpn i can't access to my local device like i can't access to my router or printer ... but when i activate SECURENAT on bridge i can access to it but the IP address i get ISP address.
I really need your help. I spent a lot of time im soo close.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

REGARDS

closed 2021-06-20 by coolsnowwolf