coolsnowwolf/lede #5165: R2S编译老报错

dl250, 2020-07-22:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt.c: In function ‘fdt_offset_ptr’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt.c:137:18: warning: comparison of integer expressions of different signedness: ‘unsigned int’ and ‘int’ [-Wsign-compare]
  137 |   if ((absoffset < offset)
      |                  ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt.c:143:23: warning: comparison of integer expressions of different signedness: ‘unsigned int’ and ‘int’ [-Wsign-compare]
  143 |   if (((offset + len) < offset)
      |                       ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt.c: In function ‘fdt_move’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt.c:307:25: warning: comparison of integer expressions of different signedness: ‘uint32_t’ {aka ‘unsigned int’} and ‘int’ [-Wsign-compare]
  307 |  if (fdt_totalsize(fdt) > bufsize)
      |                         ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c: In function ‘fdt_get_string’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c:56:16: warning: comparison of integer expressions of different signedness: ‘uint32_t’ {aka ‘unsigned int’} and ‘int32_t’ {aka ‘int’} [-Wsign-compare]
   56 |  if (absoffset >= totalsize)
      |                ^~
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c:64:18: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
   64 |    if (stroffset >= fdt_size_dt_strings(fdt))
      |                  ^~
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c:71:21: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
   71 |       || (stroffset < -fdt_size_dt_strings(fdt)))
      |                     ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c:73:20: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
   73 |   if ((-stroffset) < len)
      |                    ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c: In function ‘fdt_mem_rsv’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c:164:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  164 |   if (absoffset < fdt_off_mem_rsvmap(fdt))
      |                 ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c:166:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘long unsigned int’ [-Wsign-compare]
  166 |   if (absoffset > fdt_totalsize(fdt) -
      |                 ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c: In function ‘fdt_node_offset_by_phandle’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_ro.c:682:33: warning: comparison of integer expressions of different signedness: ‘uint32_t’ {aka ‘unsigned int’} and ‘int’ [-Wsign-compare]
  682 |  if ((phandle == 0) || (phandle == -1))
      |                                 ^~
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_wip.c: In function ‘fdt_setprop_inplace_namelen_partial’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_wip.c:26:14: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
   26 |  if (proplen < (len + idx))
      |              ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c: In function ‘fdt_grab_space_’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c:105:20: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
  105 |  if ((offset + len < offset) || (offset + len > spaceleft))
      |                    ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c:105:47: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
  105 |  if ((offset + len < offset) || (offset + len > spaceleft))
      |                                               ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c: In function ‘fdt_create_with_flags’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c:118:14: warning: comparison of integer expressions of different signedness: ‘int’ and ‘long unsigned int’ [-Wsign-compare]
  118 |  if (bufsize < hdrsize)
      |              ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c: In function ‘fdt_resize’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c:164:28: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
  164 |  if ((headsize + tailsize) > bufsize)
      |                            ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c: In function ‘fdt_add_string_’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_sw.c:258:34: warning: comparison of integer expressions of different signedness: ‘uint32_t’ {aka ‘unsigned int’} and ‘int’ [-Wsign-compare]
  258 |  if (fdt_totalsize(fdt) + offset < struct_top)
      |                                  ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_strerror.c: In function ‘fdt_strerror’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_strerror.c:51:18: warning: comparison of integer expressions of different signedness: ‘int’ and ‘long unsigned int’ [-Wsign-compare]
   51 |  else if (errval > -FDT_ERRTABSIZE) {
      |                  ^
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_overlay.c: In function ‘overlay_update_local_node_references’:
/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2020.04/scripts/dtc/libfdt/fdt_overlay.c:253:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘long unsigned int’ [-Wsign-compare]
  253 |   for (i = 0; i < (fixup_len / sizeof(uint32_t)); i++) {
      |                 ^
scripts/dtc/pylibfdt/libfdt_wrap.c:154:11: fatal error: Python.h: No such file or directory
  154 | # include <Python.h>
      |           ^~~~~~~~~~
compilation terminated.
error: command 'gcc' failed with exit status 1
make[7]: *** [scripts/dtc/pylibfdt/Makefile:27: scripts/dtc/pylibfdt/_libfdt.so] Error 1
make[6]: *** [scripts/Makefile.build:432: scripts/dtc/pylibfdt] Error 2
make[5]: *** [scripts/Makefile.build:432: scripts/dtc] Error 2
make[4]: *** [Makefile:551: scripts] Error 2
[... cut ...]
make[3]: Leaving directory '/home/ubuntu/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r2s-rk3328/compile#1.95#0.43#2.41
make[2]: *** [package/Makefile:113: package/boot/uboot-rockchip/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:107: /home/ubuntu/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:224：world] 错误 2

dl250, 2020-07-22:
make clean 也不行

ostrichii, 2020-07-23:
`sudo apt install python3-dev`

dl250, 2020-07-23:
解决了，就是少sudo apt install python3-dev的原因

closed 2020-07-23 by coolsnowwolf