coolsnowwolf/lede #5170: IPV6 DDNS异异常

yubanmeiqin9048, 2020-07-23:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
A：第一次启动可以获取到IP，但是之后就无法获取到IP了，使用的是阿里云域名解析，IPV6正常使用

### (2) 路由器型号和固件版本

A：N1 R20.7.20

### (3) 详细日志

Thu Jul 23 21:19:09 2020 user.warn ddns-scripts[18523]: IPV6: NO valid IP found
Thu Jul 23 21:19:09 2020 user.warn ddns-scripts[18523]: IPV6: Get registered/public IP for 'XXX.xyz' failed - retry 15/0 in 60 seconds
Thu Jul 23 21:20:10 2020 user.warn ddns-scripts[18523]: IPV6: NO valid IP found
Thu Jul 23 21:20:10 2020 user.warn ddns-scripts[18523]: IPV6: Get registered/public IP for 'XXX.xyz' failed - retry 16/0 in 60 seconds
Thu Jul 23 21:21:10 2020 user.warn ddns-scripts[18523]: IPV6: NO valid IP found
Thu Jul 23 21:21:11 2020 user.warn ddns-scripts[18523]: IPV6: Get registered/public IP for 'XXX.xyz' failed - retry 17/0 in 60 seconds

keivenliao, 2020-07-23:
你是不是屏蔽了ipv6的域名解析？
不过这个脚本获取域名IP的方式确实需要优化

keivenliao, 2020-07-24:
075901  note : PID '6744' terminated by 'SIGTERM' at 2020-07-24 07:59
 075905       : ************ ************** ************** **************
 075905  note : PID '6877' started at 2020-07-24 07:59
 075905       : ddns version  : 2.7.8-21
 075905       : uci configuration:
ddns.ipv6.domain='x.xxx.com'
ddns.ipv6.enabled='1'
ddns.ipv6.interface='wan'
ddns.ipv6.ip_interface='pppoe-wan'
ddns.ipv6.ip_source='interface'
ddns.ipv6.lookup_host='xxx.com'
ddns.ipv6.password='***PW***'
ddns.ipv6.service_name='aliyun.com'
ddns.ipv6.use_ipv6='1'
ddns.ipv6.username='****************'
ddns.ipv6=service
 075905       : verbose mode  : 0 - run normal, NO console output
 075905       : check interval: 600 seconds
 075905       : force interval: 259200 seconds
 075905       : retry interval: 60 seconds
 075905       : retry counter : 0 times
 075905       : No old process
 075905       : last update: 2020-07-23 09:55
 075905       : Detect registered/public IP
 075905       : #> /usr/bin/nslookup xxx.com  >/var/run/ddns/ipv6.dat 2>/var/run/ddns/ipv6.err
 075905  WARN : NO valid IP found
 075905  WARN : Error in 'expand_ipv6()' - invalid IPv6 found: '' expanded: ''
 075905  info : Starting main loop at 2020-07-24 07:59
 075905       : Detect local IP on 'interface'
 075905       : #> ip -o addr show dev pppoe-wan scope global >/var/run/ddns/ipv6.dat 2>/var/run/ddns/ipv6.err
 075905       : Local IP '240e:f8:b8:b0ca' detected on interface 'pppoe-wan'
 075905       : Update needed - L: '240e:00f8:00b8:b0ca' <> R: 'invalid'
 075905       : parsing script '/usr/lib/ddns/update_aliyun_com.sh'
 075906       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/ipv6.dat -o /var/run/ddns/ipv6.err --no-proxy 'http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=x.xxx.com&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=***********&SignatureMethod=HMAC-SHA1&Timestamp=2020-07-23T23%3A59%3A05Z&SignatureVersion=1.0&SignatureNonce=c944ee9b-a807-43c3-a22d-d0150c27b294&Signature=IHsYK3r26%2BSYVkPZ67Kt8QViEQ%3D%3D'
 075906 ERROR : wget 

: '8'
 075906       : http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=x.xxx.com&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=***********&SignatureMethod=HMAC-SHA1&Timestamp=2020-07-23T23%3A59%3A05Z&SignatureVersion=1.0&SignatureNonce=c944ee9b-a807-43c3-a22d-d0150c27b294&Signature=IHsYK3r26%2BSYVkPZ67Kt8QViEQ%3D%3D:
2020-07-24 07:59:06 ERROR 400: Bad Request.
 075906  WARN : 

 - 1/0 

 60

yubanmeiqin9048, 2020-07-24:
> 你是不是屏蔽了ipv6的域名解析？
> 不过这个脚本获取域名IP的方式确实需要优化
> 

没有屏蔽啊，就是第一次启动DDNS的时候可以获取，之后就不行了，然后计划任务里添加每10分钟让DDNS重启一次就可以正常获取了。但是我还是希望可以不通过这种方式直接开箱即用。

keivenliao, 2020-07-27:
[image]

问题出在/usr/lib/ddns/dynamic_dns_functions.sh 这个文件的获取域名地址的方法上，请大佬优化吧。

yubanmeiqin9048, 2020-07-27:
> [image]
> 
> 问题出在/usr/lib/ddns/dynamic_dns_functions.sh 这个文件的获取域名地址的方法上，请大佬优化吧。

ε=(´ο｀*)))唉，现在DDNS基本处于废用状态，只能通过脚本定时重启使用

closed 2021-06-20 by coolsnowwolf