coolsnowwolf/lede #5172: netdata开启后日志大量报错 PROCFILE: Cannot open file '/proc/sysvipc/shm'

minlang112, 2020-07-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：netdata开启后日志大量报错 PROCFILE: Cannot open file '/proc/sysvipc/shm'，但是netdata使用无异常。
参考他人经验，在/etc/netdata/netdata.conf内添加以下内容，并杀死进程重新开启后，报错日志消失。该问题在4.19内核的时候似乎没有出现过，不知道是否为该插件自身的问题。

[plugin:proc]
        ipc = no
        /proc/net/rpc/nfsd = no
        /proc/net/rpc/nfs = no
        /sys/devices/system/edac/mc = no
        /sys/devices/system/node = no
        /proc/net/sockstat = no
        /proc/net/netstat = no
        /proc/net/snmp = no
        /proc/net/sctp/snmp = no
        /proc/net/softnet_stat = no
        /proc/net/ip_vs/stats = no
        /proc/net/stat/synproxy = no
        /proc/net/rpc/nfs = no
        /sys/kernel/mm/ksm = no

[plugin:proc:/proc/diskstats]
        path to /dev/vx/dsk  =
        path to /dev/disk/by-label =

### (2) 路由器型号和固件版本

A：编译版本 OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)
内核版本 5.4.51
平台 X64

### (3) 详细日志

A：Thu Jul 23 21:07:02 2020 daemon.err netdata[4316]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
（太多了，都是重复这个内容，不复制那么多了）

yysw, 2020-08-19:
可以把ipc相关的报警关掉，像这样：
https://koolshare.cn/thread-169101-1-1.html

hzx505, 2021-02-09:
Since there is no such file on your system, you can disable shared memory totals in the [plugin:proc:ipc] section of netdata.conf to get rid of these messages.

[plugin:proc:ipc]
	shared memory totals = no

YeautyYE, 2021-02-25:
遇到同样的问题

whfweb, 2021-03-14:
Sun Mar 14 13:55:19 2021 daemon.info dnsmasq[19954]: Connected to system UBus
Sun Mar 14 13:55:20 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:21 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Mar 14 13:55:21 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Mar 14 13:55:22 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:24 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:24 2021 daemon.info dnsmasq[20195]: Connected to system UBus
Sun Mar 14 13:55:26 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:28 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:28 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Mar 14 13:55:28 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Mar 14 13:55:30 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:31 2021 daemon.info dnsmasq[20636]: Connected to system UBus
Sun Mar 14 13:55:32 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:34 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:36 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:38 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:40 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:42 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:44 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:46 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:48 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:50 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:52 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:54 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:56 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sun Mar 14 13:55:58 2021 daemon.err netdata[4382]: PROCFILE: Cannot open file '/proc/sysvipc/shm'

whfweb, 2021-03-14:
OpenWrt R21.2.1 / LuCI Master (git-21.035.34289-13f96d1)

closed 2021-06-20 by coolsnowwolf