coolsnowwolf/lede #5180: 小米R3G最新源码编译网口不能用

pipihu, 2020-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
https://github.com/coolsnowwolf/lede/issues/5002#issuecomment-660797266
https://github.com/coolsnowwolf/lede/issues/5002#issuecomment-661555214
## 2. 详细叙述

### (1) 具体问题
最新源码git clone 编译.
之前用 R20.7.7 版本没问题, R20.7.20 不行
lan 接口无法获取ip, wan 口也无法使用.wifi可以连上路由器
ssh 登录路由器 brctl show 也只能看到wlan0 wlan1 ,没有lan接口
[image]
[image]
[image]
[image]

A：

### (2) 路由器型号和固件版本
小米R3G MT7621 R20.7.20
A：

### (3) 详细日志

A：
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router 3G
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Thu Jul 23 22:40:48 2020 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Thu Jul 23 22:40:48 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Thu Jul 23 22:40:48 2020 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Zone ranges:
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000]   HighMem  empty
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Early memory node ranges
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Thu Jul 23 22:40:48 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Thu Jul 23 22:40:48 2020 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Thu Jul 23 22:40:48 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Thu Jul 23 22:40:48 2020 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
Thu Jul 23 22:40:48 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
Thu Jul 23 22:40:48 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Thu Jul 23 22:40:48 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00008260
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00008260
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] Memory: 250660K/262144K available (5678K kernel code, 249K rwdata, 1236K rodata, 1272K init, 238K bss, 11484K reserved, 0K cma-reserved, 0K highmem)
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Thu Jul 23 22:40:48 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.007792] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
Thu Jul 23 22:40:48 2020 kern.info kernel: [    0.042001] pid_max: default: 32768 minimum: 301
[... 309 lines cut ...]
Thu Jul 23 22:40:56 2020 kern.info kernel: [   27.549394] br-lan: port 2(wlan1) entered disabled state
Thu Jul 23 22:40:56 2020 kern.info kernel: [   27.555218] device wlan1 entered promiscuous mode
Thu Jul 23 22:40:56 2020 kern.info kernel: [   27.560209] br-lan: port 2(wlan1) entered blocking state
Thu Jul 23 22:40:56 2020 kern.info kernel: [   27.565569] br-lan: port 2(wlan1) entered forwarding state
Thu Jul 23 22:40:56 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Jul 23 22:40:56 2020 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Thu Jul 23 22:40:56 2020 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Thu Jul 23 22:40:56 2020 daemon.notice hostapd: wlan0: AP-ENABLED
Thu Jul 23 22:40:56 2020 daemon.notice netifd: radio1 (1540): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Thu Jul 23 22:40:56 2020 daemon.notice netifd: radio0 (1538): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Thu Jul 23 22:40:56 2020 daemon.notice netifd: radio0 (1538): Command failed: Invalid argument
Thu Jul 23 22:40:56 2020 daemon.notice netifd: radio1 (1540): Command failed: Invalid argument
Thu Jul 23 22:40:57 2020 kern.info kernel: [   28.184277] br-lan: port 2(wlan1) entered disabled state
Thu Jul 23 22:40:57 2020 kern.info kernel: [   28.553871] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Thu Jul 23 22:40:57 2020 kern.info kernel: [   28.560713] br-lan: port 2(wlan1) entered blocking state
Thu Jul 23 22:40:57 2020 kern.info kernel: [   28.566100] br-lan: port 2(wlan1) entered forwarding state
Thu Jul 23 22:40:57 2020 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Thu Jul 23 22:40:57 2020 daemon.notice hostapd: wlan1: AP-ENABLED
Thu Jul 23 22:40:57 2020 daemon.notice netifd: Network device 'wlan1' link is up
Thu Jul 23 22:40:57 2020 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jul 23 22:41:00 2020 daemon.info dnsmasq[2321]: Connected to system UBus
Thu Jul 23 22:41:09 2020 daemon.info hostapd: wlan1: STA ac:bc:32:xx:xx:xx IEEE 802.11: authenticated
Thu Jul 23 22:41:09 2020 daemon.info hostapd: wlan1: STA ac:bc:32:xx:xx:xx IEEE 802.11: associated (aid 1)
Thu Jul 23 22:41:09 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED ac:bc:32:xx:xx:xx
Thu Jul 23 22:41:09 2020 daemon.info hostapd: wlan1: STA ac:bc:32:xx:xx:xx RADIUS: starting accounting session BB609A098C1648C9

AmadeusGhost, 2020-07-25:
照着 pr #5178 把这一行去掉应该就行了
```
---	xiaomi,mir3g|\
	xiaomi,mir3g-v2|\
	xiaomi,mir4)
		ucidef_set_interfaces_lan_wan "lan1 lan2" "wan"
		;;
```

pipihu, 2020-07-26:
> 照着 pr #5178 把这一行去掉应该就行了
> 
> ```
> ---	xiaomi,mir3g|\
> 	xiaomi,mir3g-v2|\
> 	xiaomi,mir4)
> 		ucidef_set_interfaces_lan_wan "lan1 lan2" "wan"
> 		;;
> ```

参照您提供的方法,问题解决.
同时也用上了无线闭源驱动.
万分感谢

PR #5178 (2020-07-26): ramips/mtwifi: disable bind apcli interface to lan by default
The apcli interface can be bound to either lan or wan, also fix #5180
closed 2020-07-26 by pipihu