coolsnowwolf/lede #5182: youku l1 无法刷新软件包列表

Gokou-Ruri, 2020-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：点击软件包页面的“刷新列表”按钮，等待数分钟后跳转到Bad Gateway页面。页面上显示：
`Bad Gateway`
`The process did not produce any response`
虽然“刷新列表”出错，但在网页上可以卸载软件。
在ssh里可以正常`opkg update`，也可以正常安装软件。但即使在ssh里`opkg update`以后，在网页端依旧无法看到更新后的软件包列表，而且“刷新列表”按钮依旧存在，点击后依旧跳转到报错页面。

### (2) 路由器型号和固件版本

A：优酷 L1（YOUKU YK1） OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)

### (3) 详细日志

A：none

Gokou-Ruri, 2020-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：内存卡无论是单分区还是多分区，均无法挂载。
`# mount /dev/mmcblk0p1 /mnt/sdcard0`
`mount: mounting /dev/mmcblk0p1 on /mnt/sdcard0 failed: Invalid argument`
使用管理页面上的“挂载点”功能也无法挂载，添加并启用以后，并没有挂载上去。
使用DiskMan也无法挂载（和通过SSH使用mount命令是一样的报错结果）。
系统内有`kmod-sdhci-mt7620`这个软件包（编译在内）。

### (2) 路由器型号和固件版本

A：优酷 L1（YOUKU YK1） OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)

### (3) 详细日志

A：none

AmadeusGhost, 2020-07-27:
刷新软件包这个把软件源地址的 openwrt.proxy.ustclug.org 换成 https://mirrors.cloud.tencent.com/lede

Gokou-Ruri, 2020-07-28:
> 刷新软件包这个把软件源地址的 [openwrt.proxy.ustclug.org](http://openwrt.proxy.ustclug.org) 换成 https://mirrors.cloud.tencent.com/lede

谢谢大佬，不过这个方法对我没有用，我在ssh里是可以使用中科大源的，应该是luci那边有问题，导致无法正常调用opkg update或者无法收到返回值（猜测）。我试过更换成腾讯源，在网页上点“刷新列表”，和中科大源是一样的结果。

Gokou-Ruri, 2020-08-05:
#5232

closed 2021-06-20 by coolsnowwolf