coolsnowwolf/lede #5183: Tx timeout

ostrichii, 2020-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：大流量下载时，会不定时出现Tx timeout，usb 5-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd这种错误，CPU和流量下跌后又恢复

### (2) 路由器型号和固件版本

A：型号：R2S 

版本OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)

### (3) 详细日志

A：
[image]
[image]

`at Jul 25 15:12:12 2020 kern.warn kernel: [ 4574.594652] r8152 5-1:1.0 eth1: Tx timeout
Sat Jul 25 15:12:14 2020 kern.warn kernel: [ 4575.717780] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:12:14 2020 kern.warn kernel: [ 4575.718277] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:12:14 2020 kern.warn kernel: [ 4575.718783] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:12:14 2020 kern.warn kernel: [ 4575.719219] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:12:14 2020 kern.info kernel: [ 4576.390823] usb 5-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Sat Jul 25 15:12:14 2020 kern.info kernel: [ 4576.421528] r8152 5-1:1.0 eth1: DT mac addr da:56:e2:6d:13:ec
Sat Jul 25 15:12:14 2020 kern.notice kernel: [ 4576.424667] r8152 5-1:1.0 eth1: Promiscuous mode enabled
Sat Jul 25 15:12:15 2020 daemon.err netdata[2726]: heartbeat missed 1999946 monotonic microseconds
Sat Jul 25 15:13:35 2020 user.info mwan3track[3608]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Sat Jul 25 15:13:40 2020 user.info mwan3track[3608]: Lost 1 ping(s) on interface wan (pppoe-wan)
Sat Jul 25 15:13:52 2020 user.info mwan3track[3608]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Sat Jul 25 15:13:58 2020 user.info mwan3track[3608]: Lost 1 ping(s) on interface wan (pppoe-wan)
Sat Jul 25 15:14:41 2020 user.info mwan3track[3608]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Sat Jul 25 15:14:47 2020 user.info mwan3track[3608]: Lost 1 ping(s) on interface wan (pppoe-wan)
Sat Jul 25 15:15:09 2020 user.info mwan3track[3608]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Sat Jul 25 15:15:14 2020 user.info mwan3track[3608]: Lost 1 ping(s) on interface wan (pppoe-wan)
Sat Jul 25 15:16:50 2020 kern.warn kernel: [ 4852.605887] r8152 5-1:1.0 eth1: Tx timeout
Sat Jul 25 15:16:50 2020 kern.warn kernel: [ 4852.609153] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:16:50 2020 kern.warn kernel: [ 4852.609702] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:16:50 2020 kern.warn kernel: [ 4852.610237] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:16:50 2020 kern.warn kernel: [ 4852.610713] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:16:53 2020 kern.info kernel: [ 4854.757383] usb 5-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Sat Jul 25 15:16:53 2020 kern.info kernel: [ 4854.789730] r8152 5-1:1.0 eth1: DT mac addr da:56:e2:6d:13:ec
Sat Jul 25 15:16:53 2020 kern.notice kernel: [ 4854.793198] r8152 5-1:1.0 eth1: Promiscuous mode enabled
Sat Jul 25 15:19:08 2020 kern.warn kernel: [ 4990.585630] r8152 5-1:1.0 eth1: Tx timeout
Sat Jul 25 15:19:08 2020 kern.warn kernel: [ 4990.588928] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:19:08 2020 kern.warn kernel: [ 4990.589420] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:19:08 2020 kern.warn kernel: [ 4990.589953] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:19:08 2020 kern.warn kernel: [ 4990.590694] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:19:11 2020 kern.info kernel: [ 4992.741313] usb 5-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Sat Jul 25 15:19:11 2020 kern.info kernel: [ 4992.773548] r8152 5-1:1.0 eth1: DT mac addr da:56:e2:6d:13:ec
Sat Jul 25 15:19:11 2020 kern.notice kernel: [ 4992.776789] r8152 5-1:1.0 eth1: Promiscuous mode enabled
Sat Jul 25 15:20:14 2020 user.info mwan3track[3608]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Sat Jul 25 15:20:19 2020 user.info mwan3track[3608]: Lost 1 ping(s) on interface wan (pppoe-wan)
Sat Jul 25 15:20:26 2020 user.info mwan3track[3608]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Sat Jul 25 15:20:32 2020 user.info mwan3track[3608]: Lost 1 ping(s) on interface wan (pppoe-wan)
Sat Jul 25 15:21:15 2020 user.info mwan3track[3608]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Sat Jul 25 15:21:21 2020 user.info mwan3track[3608]: Lost 1 ping(s) on interface wan (pppoe-wan)
Sat Jul 25 15:26:12 2020 daemon.notice netifd: iptv (1190): udhcpc: sending renew to 10.206.128.1
Sat Jul 25 15:26:12 2020 daemon.notice netifd: iptv (1190): udhcpc: lease of 10.206.143.106 obtained, lease time 5400
Sat Jul 25 15:30:07 2020 user.info mwan3track[3608]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Sat Jul 25 15:30:13 2020 user.info mwan3track[3608]: Lost 1 ping(s) on interface wan (pppoe-wan)
Sat Jul 25 15:31:51 2020 kern.warn kernel: [ 5753.432465] r8152 5-1:1.0 eth1: Tx timeout
Sat Jul 25 15:31:51 2020 kern.warn kernel: [ 5753.436083] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:31:51 2020 kern.warn kernel: [ 5753.436623] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:31:51 2020 kern.warn kernel: [ 5753.437079] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:31:51 2020 kern.warn kernel: [ 5753.437516] r8152 5-1:1.0 eth1: Tx status -2
Sat Jul 25 15:31:53 2020 kern.info kernel: [ 5755.573379] usb 5-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Sat Jul 25 15:31:53 2020 kern.info kernel: [ 5755.605624] r8152 5-1:1.0 eth1: DT mac addr da:56:e2:6d:13:ec
Sat Jul 25 15:31:53 2020 kern.notice kernel: [ 5755.608583] r8152 5-1:1.0 eth1: Promiscuous mode enabled
`

hxlls, 2020-07-25:
明显就是网卡掉了把。。USB的，正常了，看开点

ostrichii, 2020-07-25:
> 明显就是网卡掉了把。。USB的，正常了，看开点

这个不是插上去的

closed 2021-06-20 by coolsnowwolf