coolsnowwolf/lede #5184: IPSec VPN 无法使用

tporwx, 2020-07-25:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：IPSec 无法连接

### (2) 路由器型号和固件版本

A：k3 最新源码编译

### (3) 详细日志

A：Sat Jul 25 18:41:15 2020 authpriv.info ipsec_starter[14885]: Starting strongSwan 5.8.4 IPsec [starter]...
Sat Jul 25 18:41:15 2020 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.8.4, Linux 4.19.131, armv7l)
Sat Jul 25 18:41:15 2020 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Sat Jul 25 18:41:15 2020 daemon.info : 00[KNL] failed to create TUN device
Sat Jul 25 18:41:15 2020 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Sat Jul 25 18:41:15 2020 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Sat Jul 25 18:41:15 2020 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Sat Jul 25 18:41:15 2020 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Sat Jul 25 18:41:15 2020 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Sat Jul 25 18:41:15 2020 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Sat Jul 25 18:41:15 2020 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Sat Jul 25 18:41:15 2020 daemon.info : 00[CFG]   loaded IKE secret for %any
Sat Jul 25 18:41:15 2020 daemon.info : 00[CFG]   loaded EAP secret for lxb
Sat Jul 25 18:41:15 2020 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Sat Jul 25 18:41:15 2020 daemon.info : 00[JOB] spawning 16 worker threads
Sat Jul 25 18:41:15 2020 daemon.info : 01[DMN] thread 1 received 11
Sat Jul 25 18:41:15 2020 daemon.info : 01[LIB] no support for capturing backtraces
Sat Jul 25 18:41:15 2020 daemon.info : 01[DMN] killing ourself, received critical signal
Sat Jul 25 18:41:15 2020 authpriv.info ipsec_starter[14885]: charon has died -- restart scheduled (5sec)
Sat Jul 25 18:41:15 2020 authpriv.info ipsec_starter[14885]: charon refused to be started
Sat Jul 25 18:41:20 2020 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.8.4, Linux 4.19.131, armv7l)
Sat Jul 25 18:41:20 2020 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Sat Jul 25 18:41:20 2020 daemon.info : 00[KNL] failed to create TUN device
Sat Jul 25 18:41:20 2020 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Sat Jul 25 18:41:20 2020 daemon.info : 00[KNL] unable to create IPv4 routing table rule
Sat Jul 25 18:41:20 2020 daemon.info : 00[KNL] unable to create IPv6 routing table rule
Sat Jul 25 18:41:20 2020 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Sat Jul 25 18:41:20 2020 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Sat Jul 25 18:41:20 2020 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Sat Jul 25 18:41:20 2020 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Sat Jul 25 18:41:20 2020 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Sat Jul 25 18:41:20 2020 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Sat Jul 25 18:41:20 2020 daemon.info : 00[CFG]   loaded IKE secret for %any
Sat Jul 25 18:41:20 2020 daemon.info : 00[CFG]   loaded EAP secret for lxb
Sat Jul 25 18:41:20 2020 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Sat Jul 25 18:41:20 2020 daemon.info : 00[JOB] spawning 16 worker threads
Sat Jul 25 18:41:20 2020 authpriv.info ipsec_starter[14885]: charon (14959) started after 40 ms
Sat Jul 25 18:41:20 2020 daemon.info : 05[CFG] received stroke: add connection 'xauth_psk'
Sat Jul 25 18:41:20 2020 daemon.info : 05[CFG] adding virtual IP address pool 192.168.1.250/24
Sat Jul 25 18:41:20 2020 daemon.info : 05[CFG] added configuration 'xauth_psk'
Sat Jul 25 18:41:20 2020 daemon.info : 05[DMN] thread 5 received 11
Sat Jul 25 18:41:20 2020 daemon.info : 05[LIB] no support for capturing backtraces
Sat Jul 25 18:41:20 2020 daemon.info : 05[DMN] killing ourself, received critical signal
Sat Jul 25 18:41:20 2020 authpriv.info ipsec_starter[14885]: charon has died -- restart scheduled (5sec)

cnzd, 2020-07-27:
我也是，手机连接不上vpn

mingxiaoyu, 2020-07-27:
Mon Jul 27 22:53:14 2020 authpriv.info ipsec_starter[1785]: charon (21256) started after 20 ms
Mon Jul 27 22:53:14 2020 authpriv.info ipsec_starter[1785]: reading stroke response failed
Mon Jul 27 22:53:14 2020 authpriv.info ipsec_starter[1785]: charon has died -- restart scheduled (5sec)

CoolZxp, 2020-09-07:
plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL

strongswan-mod-kernel-libipsec_5.8.4-1_mipsel_24kc

CoolZxp, 2020-09-07:
plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL

strongswan-mod-kernel-libipsec
看一下是不是没装这个软件包 或者 版本有问题

lonelysoul, 2020-09-09:
我也无法使用，连不上。报错和第一位相同
已有这个包：strongswan-mod-kernel-libipsec

Hey-Leo, 2020-09-19:
同，报错
```
Sat Sep 19 16:18:55 2020 authpriv.info ipsec_starter[7889]: charon has died -- restart scheduled (5sec)
Sat Sep 19 16:18:55 2020 authpriv.info ipsec_starter[7889]: charon refused to be started
Sat Sep 19 16:19:00 2020 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.8.4, Linux 5.4.61, mips)
Sat Sep 19 16:19:00 2020 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Sat Sep 19 16:19:00 2020 daemon.info : 00[KNL] failed to create TUN device
Sat Sep 19 16:19:00 2020 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Sat Sep 19 16:19:00 2020 daemon.info : 00[KNL] unable to create IPv4 routing table rule
Sat Sep 19 16:19:00 2020 daemon.info : 00[KNL] unable to create IPv6 routing table rule
Sat Sep 19 16:19:00 2020 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Sat Sep 19 16:19:00 2020 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Sat Sep 19 16:19:00 2020 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Sat Sep 19 16:19:00 2020 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Sat Sep 19 16:19:00 2020 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Sat Sep 19 16:19:00 2020 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Sat Sep 19 16:19:00 2020 daemon.info : 00[CFG]   loaded IKE secret for %any
Sat Sep 19 16:19:00 2020 daemon.info : 00[CFG]   loaded EAP secret for 12
Sat Sep 19 16:19:00 2020 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Sat Sep 19 16:19:00 2020 daemon.info : 00[JOB] spawning 16 worker threads
Sat Sep 19 16:19:00 2020 authpriv.info ipsec_starter[7889]: charon (14354) started after 40 ms
Sat Sep 19 16:19:00 2020 daemon.info : 04[DMN] thread 4 received 11
Sat Sep 19 16:19:00 2020 daemon.info : 05[CFG] received stroke: add connection 'xauth_psk'
Sat Sep 19 16:19:00 2020 daemon.info : 09[DMN] thread 9 received 11
Sat Sep 19 16:19:00 2020 daemon.info : 09[LIB] no support for capturing backtraces
Sat Sep 19 16:19:00 2020 daemon.info : 09[DMN] killing ourself, received critical signal
Sat Sep 19 16:19:00 2020 authpriv.info ipsec_starter[7889]: charon has died -- restart scheduled (5sec)
```

Mr-dbo, 2020-10-21:
编译出来就错误了……应该是包的问题

fchen22, 2020-12-25:
刚刚碰到这个问题，是因为没有配置CONFIG_PACKAGE_kmod-tun

kkxkx, 2020-12-28:
CONFIG_PACKAGE_kmod-tun 这个问题如何处理的 重新编译了吗？还是去哪里配置 ？

josonwang, 2021-01-13:
5.4内核这个IPSEC即使使用手机端 连接上了也访问不了外网。

closed 2021-06-20 by coolsnowwolf