coolsnowwolf/lede #5188: 32M flash ，固件大小超过16M 报 sysupgrade.bin is too big 问题

Aes64X, 2020-07-26:
此问题发生在 Newifi D1 和 小娱C5上，两个设备都使用 32M Flash。
新三尚不明确，昨晚见 https://www.right.com.cn/forum/thread-4044574-1-1.html 据发帖描述其首次打包正常，之后报 sysupgrade.bin is too big，但也见坛子新三分区有最近的固件发出。

主要表现为 设备使用 32M Flash，DTS中也表现其拥有32M空间，但在生成固件时大小无法超过 16M，超过即报 sysupgrade.bin is too big

经多次尝试，内存映像在大小17.5M时，无法生成固件，报 sysupgrade.bin is too big ，继续删减插件，在固件大小14M左右时，固件生成正常。

曾尝试将其还原到主线Op还存在 ralink_default_fw_size_32M=33226752 定义时尝试打包，仍然提示 too big，也检查了DTS中存储描述，都正常。

各位路过的大佬劳烦分析一下问题
附有log文件

[log.txt](https://github.com/coolsnowwolf/lede/files/4976994/log.txt)

Ljzkirito, 2020-07-26:
我刚刚编译了两次同配置小娱C5固件，22M，没有问题。
这是我的配置文件
[XY-C5.config.txt](https://github.com/coolsnowwolf/lede/files/4977081/XY-C5.config.txt)

Aes64X, 2020-07-26:
> 我刚刚编译了两次同配置小娱C5固件，22M，没有问题。
> 这是我的配置文件
> [XY-C5.config.txt](https://github.com/coolsnowwolf/lede/files/4977081/XY-C5.config.txt)

多谢，我将再次检查一些改动
关闭。

closed 2020-07-26 by Aes64X