coolsnowwolf/lede #5189: 最新源码 make disclean 之后重新编译，好多 patch 都失效了，编译不通过

caicaicai21, 2020-07-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
最新源码 make disclean 之后重新编译，好多 patch 都失效了，编译不通过
A：

### (2) 路由器型号和固件版本
x86_64，同步至日期 Jul 23, 2020 版本
A：

### (3) 详细日志
. /home/caicaicai21/lede/include/shell.sh; bzcat /home/caicaicai21/lede/dl/u-boot-2020.04.tar.bz2 | tar -C /home/caicaicai21/lede/build_dir/host/u-boot-2020.04/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/caicaicai21/lede/build_dir/host/u-boot-2020.04

Applying ./patches/010-freebsd-ulong-fix.patch using plaintext: 
patching file include/image.h
Hunk #1 succeeded at 53 (offset 2 lines).

Applying ./patches/020-include_compile_fix.patch using plaintext: 
patching file include/u-boot/rsa-checksum.h
Hunk #1 succeeded at 6 (offset -1 lines).

Applying ./patches/030-allow-to-use-different-magic.patch using plaintext: 
patching file tools/mkimage.c
patching file tools/default_image.c
patching file tools/imagetool.h

Applying ./patches/050-Add-compatibility-with-non-Linux-hosts.patch using plaintext: 
patching file include/image.h
patching file include/imx8image.h
patching file include/linux/posix_types.h
patching file include/linux/types.h
patching file lib/rsa/rsa-sign.c

Applying ./patches/050-image_h_portability.patch using plaintext: 
patching file include/image.h
Hunk #1 FAILED at 17.
Hunk #2 FAILED at 308.
2 out of 2 hunks FAILED -- saving rejects to file include/image.h.rej
Patch failed!  Please fix ./patches/050-image_h_portability.patch!
make[3]: *** [Makefile:57: /home/caicaicai21/lede/build_dir/host/u-boot-2020.04/.prepared2472840a6e142acfa17d8fa515767f9a_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/caicaicai21/lede/tools/mkimage'
time: tools/mkimage/compile#1.44#1.68#2.41
make[2]: *** [tools/Makefile:165: tools/mkimage/compile] Error 2
make[2]: Leaving directory '/home/caicaicai21/lede'
make[1]: *** [tools/Makefile:161: /home/caicaicai21/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyynyyyyynyynnyyyynyyyyyyyyyyyyyyyyynynnyyyynynyyyy] Error 2
make[1]: Leaving directory '/home/caicaicai21/lede'
make: *** [/home/caicaicai21/lede/include/toplevel.mk:224: world] Error 2
(日志显示的只是其中的一个)
A：

caicaicai21, 2020-07-26:
估计重新生成patch会是一个大工程哦。

liyufan, 2020-07-26:
+1一样的错误日志

closed 2020-07-27 by caicaicai21

caicaicai21, 2020-07-27:
可能是我的问题，我 git reset 后应该用 git clean 命令清理一下多余的文件。

caicaicai21, 2020-07-27:
> +1一样的错误日志

如果你也是和我一样，用 git reset 命令的话，试着用 git clean 命令清理一下多余的文件。