coolsnowwolf/lede #5191: 使用一段时间后会出现 /usr/lib/lua/luci/dispatcher.lua:510: Fatal: Indexcache is not sane!

Cc5192, 2020-07-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [] 没有类似的issue
  搜索过，解决方法都是临时的
## 2. 详细叙述

### (1) 具体问题
正常使用一段时间以后进入web管理页面会出现
/usr/lib/lua/luci/dispatcher.lua:510: Fatal: Indexcache is not sane!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:510: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:553: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:238: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
这个错误，可以使用ssh清除缓存或重启，但是这个是什么导致的呢？是否有办法永久解决

A：

### (2) 路由器型号和固件版本
ESXI虚拟的Openwrt
A：

### (3) 详细日志
/usr/lib/lua/luci/dispatcher.lua:510: Fatal: Indexcache is not sane!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:510: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:553: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:238: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
A：

VaguelyCloud, 2020-08-01:
没错我也是遇到同样情况，也是7月底左右。解决办法只有重启

Cc5192, 2020-08-03:
> 没错我也是遇到同样情况，也是7月底左右。解决办法只有重启

方便加个QQ沟通一下咱俩的使用环境的共同点吗  好像很多人没有这个问题。。

VaguelyCloud, 2020-08-03:
可以，695022329 除此问题外。我还遇到ppoe 拨号不稳定问题。但是类似问题只出现在主路由上。旁路由未出现问题

lxnmail, 2020-08-25:
有解决办法了么？最近我也被这个问题困扰了。差不多一天左右就会出现这个问题，只能重启解决。我是再pve虚拟op的

closed 2021-06-20 by coolsnowwolf

ChenYuJie1987, 2022-07-04:
<html>
<body>

rm -r /tmp/luci-indexcache
--

</body>
</html>