coolsnowwolf/lede #5192: 模糊问题

smalldickman, 2020-07-26:
模糊问题，请问各位刷5.4内核的有没有出现操作时会重启的现象，同一台设备，老版内核没有出现这种现象，cpu温度正常50多度。
至今出现四次（每次操作也都不同，不可再现），没有操作稳定运行一天也是没问题，，

chengjyang, 2020-07-26:
我的也是,wan口还经常重连

smalldickman, 2020-07-26:
> 我的也是,wan口还经常重连
我刷下旧版看下，或许硬件出问题了，日志也看不懂

xiehuayi75, 2020-07-26:
我的一换公网ip就不能拨号了。要手动连接才行

shenyz, 2020-07-27:
5.4.50，R20.7.7随便折腾，没有出现不良反应，已经运行10天

smalldickman, 2020-07-27:
> 5.4.50，R20.7.7随便折腾，没有出现不良反应，已经运行10天

smalldickman, 2020-07-28:
今天出现三次重启，第一次满载重启（用owncloud生成图片预览），第二次第三次负载不到50%重启（这两次是相同操作过程，用jellyfin扫描媒体库），这次是重新编译的5.4内核版本，虽然重启但不像之前卡死。。这是系统日志搜索reboot显示相关，一点不懂，不知道有没有用，求老师上课
[image]

（三次重启应该都不是温度问题，最高70，特意加了风扇）

bin20088, 2020-07-30:
确实遇到了同样的问题，我这平均每天一到两次不定时重启，暂时没抓到规律

bin20088, 2020-07-30:
> 确实遇到了同样的问题，我这平均每天一到两次不定时重启，暂时没抓到规律

刷回R20.6.18稳定运行一个礼拜了

smalldickman, 2020-07-30:
今天又无故重启，这次没有任何操做。这几次重启时间都是不固定的

smalldickman, 2020-08-06:
> > 确实遇到了同样的问题，我这平均每天一到两次不定时重启，暂时没抓到规律
> 
> 
> 
> 刷回R20.6.18稳定运行一个礼拜了

不知道算不算规律，新内核，接口流量lan口或者wan口累计150g左右就会重启，换回老版本4.x lan口流量累计到1.5T出现过重启，猜是不是缓存的缘故

closed 2020-08-08 by smalldickman