coolsnowwolf/lede #5195: SSR一直重启

wangyang8284, 2020-07-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：SSR无法启动

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：------------end------------
2020-07-27 03:22:05 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:22:05 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:21:35 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:21:34 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:21:04 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:21:03 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:20:33 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:20:33 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:20:02 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:20:02 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:19:32 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:19:18 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:19:05 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:19:04 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:18:34 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:18:24 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:18:15 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:18:14 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:17:44 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:17:43 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:17:13 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:17:13 ssr redir tcp error.restart!
------------end------------
2020-07-27 03:17:11 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
2020-07-27 03:17:03 订阅失败, 恢复服务
2020-07-27 03:17:02 更新失败，没有可用的节点信息
2020-07-27 03:17:02 https://losadhwselfff2332dasd.xyz/link/R5h8q6AD1wx7tcZJ?mu=2: 获取内容为空
------------end------------
2020-07-27 03:17:03 Main node:Can't find /usr/bin/ssr-redir program, can't start!
/etc/rc.common: line 69: resolveip: not found
-----------start-----------
2020-07-27 03:16:28 服务正在暂停
------------end------------
2020-07-27 03:16:13 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:16:05 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:15:52 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:15:52 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:15:47 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:15:30 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
2020-07-27 03:15:19 Main node:Can't find /usr/bin/ssr-redir program, can't start!
-----------start-----------
------------end------------
-----------start-----------
2020-07-27 03:12:41 订阅失败, 恢复服务
2020-07-27 03:12:41 更新失败，没有可用的节点信息
2020-07-27 03:12:41 https://losadhwselfff2332dasd.xyz/link/R5h8q6AD1wx7tcZJ?mu=2: 获取内容为空
------------end------------
-----------start-----------
2020-07-27 03:12:18 订阅失败, 恢复服务
2020-07-27 03:12:18 更新失败，没有可用的节点信息
2020-07-27 03:12:18 https://losadhwselfff2332dasd.xyz/link/R5h8q6AD1wx7tcZJ?mu=2: 获取内容为空
------------end------------
-----------start-----------
2020-07-27 03:12:07 服务正在暂停
2020-07-27 03:11:44 服务正在暂停
------------end------------
-----------start-----------

forever2019, 2020-07-27:
Sun Jul 26 23:37:58 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Jul 26 23:37:59 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Sun Jul 26 23:38:01 2020 daemon.notice netifd: Interface 'wan_6' is now up
Sun Jul 26 23:38:02 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sun Jul 26 23:38:34 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jul 26 23:38:34 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jul 26 23:38:35 2020 daemon.info dnsmasq[2805]: Connected to system UBus
Sun Jul 26 23:39:17 2020 daemon.info pppd[1896]: System time change detected.
Mon Jul 27 05:00:00 2020 cron.err crond[1296]: USER root pid 10561 cmd /usr/share/shadowsocksr/ssrplusupdate.sh >>/tmp/ssrplus.log 2>&1
Mon Jul 27 05:00:11 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 27 05:00:11 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 27 05:00:11 2020 daemon.info dnsmasq[10851]: Connected to system UBus
Mon Jul 27 05:00:11 2020 cron.err crond[10885]: crond (busybox 1.31.1) started, log level 5
Mon Jul 27 05:00:15 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 27 05:00:15 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 27 05:00:16 2020 daemon.info dnsmasq[11316]: Connected to system UBus
Mon Jul 27 05:00:16 2020 cron.err crond[11350]: crond (busybox 1.31.1) started, log level 5
Mon Jul 27 05:00:17 2020 kern.info /usr/bin/ssr-redir[11449]: initializing ciphers... none
Mon Jul 27 05:00:17 2020 kern.info /usr/bin/ssr-redir[11451]: initializing ciphers... none
Mon Jul 27 05:00:17 2020 kern.info /usr/bin/ssr-redir[11453]: initializing ciphers... none
Mon Jul 27 05:00:17 2020 kern.info /usr/bin/ssr-redir[11455]: initializing ciphers... none
Mon Jul 27 05:00:19 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul 27 05:00:19 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul 27 05:00:20 2020 daemon.info dnsmasq[11576]: Connected to system UBus
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11453]: tcp port reuse enabled
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11453]: listening at 0.0.0.0:1234
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11453]: running from root user
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11451]: tcp port reuse enabled
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11451]: listening at 0.0.0.0:1234
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11451]: running from root user
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11449]: tcp port reuse enabled
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11449]: listening at 0.0.0.0:1234
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11449]: running from root user
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11455]: tcp port reuse enabled
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11455]: listening at 0.0.0.0:1234
Mon Jul 27 05:00:20 2020 kern.info /usr/bin/ssr-redir[11455]: running from root user
Mon Jul 27 05:11:25 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 05:22:32 2020 kern.err /usr/bin/ssr-redir[11453]: server recv: Connection reset by peer
Mon Jul 27 05:33:40 2020 kern.err /usr/bin/ssr-redir[11455]: send: Connection reset by peer
Mon Jul 27 05:43:05 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 05:44:47 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 05:55:55 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 06:07:02 2020 kern.err /usr/bin/ssr-redir[11449]: server recv: Connection reset by peer
Mon Jul 27 06:13:05 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 06:18:10 2020 kern.err /usr/bin/ssr-redir[11451]: server recv: Connection reset by peer
Mon Jul 27 06:29:18 2020 kern.err /usr/bin/ssr-redir[11449]: server recv: Connection reset by peer
Mon Jul 27 06:40:25 2020 kern.err /usr/bin/ssr-redir[11449]: server recv: Connection reset by peer
Mon Jul 27 06:43:05 2020 kern.err /usr/bin/ssr-redir[11453]: server recv: Connection reset by peer
Mon Jul 27 06:51:33 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 07:13:48 2020 kern.err /usr/bin/ssr-redir[11449]: server recv: Connection reset by peer
Mon Jul 27 07:24:55 2020 kern.err /usr/bin/ssr-redir[11453]: server recv: Connection reset by peer
Mon Jul 27 07:36:03 2020 kern.err /usr/bin/ssr-redir[11449]: server recv: Connection reset by peer
Mon Jul 27 07:43:05 2020 kern.err /usr/bin/ssr-redir[11451]: server recv: Connection reset by peer
Mon Jul 27 07:47:10 2020 kern.err /usr/bin/ssr-redir[11453]: server recv: Connection reset by peer
Mon Jul 27 07:58:18 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 08:09:25 2020 kern.err /usr/bin/ssr-redir[11453]: server recv: Connection reset by peer
Mon Jul 27 08:20:33 2020 kern.err /usr/bin/ssr-redir[11451]: server recv: Connection reset by peer
Mon Jul 27 08:31:40 2020 kern.err /usr/bin/ssr-redir[11453]: server recv: Connection reset by peer
[... cut ...]
Mon Jul 27 16:07:51 2020 kern.err /usr/bin/ssr-redir[11451]: server recv: Connection reset by peer
Mon Jul 27 16:13:05 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 16:18:59 2020 kern.err /usr/bin/ssr-redir[11449]: server recv: Connection reset by peer
Mon Jul 27 16:30:06 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 16:52:21 2020 kern.err /usr/bin/ssr-redir[11455]: server recv: Connection reset by peer
Mon Jul 27 17:25:44 2020 kern.err /usr/bin/ssr-redir[11453]: server recv: Connection reset by peer

coolsnowwolf, 2020-07-28:
luci 找不到主程序

uosX, 2020-11-14:
------------end------------
2020-11-14 20:46:37 Main node: Started!
-----------start-----------
2020-11-14 20:46:34 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:45:57 Main node: Started!
-----------start-----------
2020-11-14 20:45:54 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:45:17 Main node: Started!
-----------start-----------
2020-11-14 20:45:14 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:44:38 Main node: Started!
-----------start-----------
2020-11-14 20:44:34 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:43:57 Main node: Started!
-----------start-----------
2020-11-14 20:43:54 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:43:17 Main node: Started!
-----------start-----------
2020-11-14 20:43:14 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:42:37 Main node: Started!
-----------start-----------
2020-11-14 20:42:34 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:41:57 Main node: Started!
-----------start-----------
2020-11-14 20:41:54 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:41:17 Main node: Started!
-----------start-----------
2020-11-14 20:41:14 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:40:37 Main node: Started!
-----------start-----------
2020-11-14 20:40:34 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:39:57 Main node: Started!
-----------start-----------
2020-11-14 20:39:54 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:39:17 Main node: Started!
-----------start-----------
2020-11-14 20:39:14 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:38:37 Main node: Started!
-----------start-----------
2020-11-14 20:38:34 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:37:57 Main node: Started!
-----------start-----------
2020-11-14 20:37:54 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:37:17 Main node: Started!
-----------start-----------
2020-11-14 20:37:14 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:36:37 Main node: Started!
-----------start-----------
2020-11-14 20:36:34 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:35:57 Main node: Started!
-----------start-----------
2020-11-14 20:35:53 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:35:17 Main node: Started!
-----------start-----------
2020-11-14 20:35:17 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:34:43 Main node: Started!
-----------start-----------
------------end------------
2020-11-14 20:34:24 Main node: Started!
-----------start-----------
2020-11-14 20:34:21 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:33:44 Main node: Started!
-----------start-----------
2020-11-14 20:33:41 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:33:05 Main node: Started!
-----------start-----------
2020-11-14 20:33:01 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:32:24 Main node: Started!
-----------start-----------
2020-11-14 20:32:21 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:31:45 Main node: Started!
-----------start-----------
2020-11-14 20:31:41 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:31:05 Main node: Started!
-----------start-----------
2020-11-14 20:31:01 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:30:24 Main node: Started!
-----------start-----------
2020-11-14 20:30:20 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:29:47 Main node: Started!
-----------start-----------
2020-11-14 20:29:43 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:29:06 Main node: Started!
-----------start-----------
2020-11-14 20:29:03 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:28:26 Main node: Started!
-----------start-----------
2020-11-14 20:28:22 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:27:46 Main node: Started!
-----------start-----------
2020-11-14 20:27:43 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:27:06 Main node: Started!
-----------start-----------
2020-11-14 20:27:03 ssr redir tcp error.restart!
------------end------------
2020-11-14 20:26:27 Main node: Started!
-----------start-----------
------------end------------
2020-11-14 20:25:32 Main node:V2Ray 4.32.1 (OpenWrt) Lean (go1.15.2 linux/amd64) Started!
-----------start-----------

closed 2021-06-20 by coolsnowwolf