coolsnowwolf/lede #5196: OP刻到U盘可以启动，刻到固态不能。。。

adaaaaaa, 2020-07-27:
固态是没有问题的，可以识别，可以存放，修改数据。。。
最常见报错：Error:Unknown Filesystem
少见报错: error:disk `hd0,gpt1' not found.（但是这种报错一般重启之后就变成第一种报错）
我给固态擦除数据了再克隆硬盘，从可以启动的U盘到固态，还是不行。。。

martin0607, 2020-07-27:
你的固态坏掉了

adaaaaaa, 2020-07-27:
固态还可以擦写啊。。。怎么就坏了？
[image]

adaaaaaa, 2020-07-27:
@martin0607 固态还可以擦写啊。。。怎么算坏了？

adaaaaaa, 2020-07-27:
@martin0607 看s.m.a.r.t信息也是没有问题啊？

martin0607, 2020-07-27:
不要用UEFI看看，用legacy

adaaaaaa, 2020-07-27:
@martin0607 不知道哪里设置，是打开CSM兼容模块吗？之前默认是关闭的。。。
可是这样依然不行，甚至连GRUB都报错了。。。

adaaaaaa, 2020-07-27:
@martin0607 奇怪，我用7月20的版本就是不行，用4月8号的，立刻就能启动了。。。
新版本对NVME固态硬盘有什么不兼容的成分吗？？

loydcheung, 2020-07-27:
固态我记得要主板支持UEFI启动才行。

adaaaaaa, 2020-07-27:
@loydcheung 
4月和7月固件的差异就在这里吗？

Nadleeh1988, 2020-07-27:
从这次内核升级以后，我用了各种IMG写入工具，写入SSD都到99%的时候报错。。。但是还是能使用。。。

adaaaaaa, 2020-07-27:
@Nadleeh1988 不行，我只能用4月份编译好的固件，7月份的怎么都不能启动啊。。。
报错你看得懂吗？？😭

Nadleeh1988, 2020-07-27:
你刷完报错以后  接屏幕启动下会发现 停在一个选择界面   LUCI SAFE啥的  只能按上下选择  其他按钮都无效，这个时候再重启下就能正常使用了。我是这样的情况。

xiexiaosheng, 2020-07-27:
https://github.com/OPisthebest/OP-is-the-best/releases  下载这个

Nadleeh1988, 2020-07-27:
> https://github.com/OPisthebest/OP-is-the-best/releases 下载这个

谢谢，我去试下

adaaaaaa, 2020-07-27:
@Nadleeh1988 
你确定和我是一样的报错吗？
我还没试过多次重启是否可行。。。不够，报错之后也不是你说的那个界面。。。还是返回了grub菜单。。。

adaaaaaa, 2020-07-27:
@xiexiaosheng 我在另外一天电脑上试write image to那一格是空白的？？

adaaaaaa, 2020-07-27:
发现一个怪事。。。4月固件有4个分区。。。7月的有时候看起来是3个分区，有时候又是2个。。。
怎么最近变动那么大吗？

xiexiaosheng, 2020-07-27:
我今天才刷过OP

adaaaaaa, 2020-07-27:
@xiexiaosheng 
你那边刷好之后，有几个分区？
我这边看镜像文件，PE里看是3分区，WIN10里看是2分区。。。都是diskgenius看的。。。

xiexiaosheng, 2020-07-27:
两个

adaaaaaa, 2020-07-27:
@xiexiaosheng 你用哪种镜像文件的格式写的，img,iso,vdi,vmdk还是其他？

xiexiaosheng, 2020-07-27:
img吧，我记着是

adaaaaaa, 2020-07-27:
表现那么怪异。。。没有信心啊。。。再刷，怕还是同样的报错啊。。。😭

xiexiaosheng, 2020-07-27:
没事，失败是成功他妈

adaaaaaa, 2020-07-27:
diskgenius5.2.0版本打开镜像有2个分区，5.2.1版本打开是3个分区。。。所以，镜像文件到底有多少个分区？？

xiexiaosheng, 2020-07-27:
不知道，哈哈哈，你看看能不能用，能用就是好事

Nadleeh1988, 2020-07-27:
我试了 最新的编译 用楼上的1.2写入成功了   重启后报HD0引导丢失，进GNU GRUB目录。按CTRL ALT DELETE  重启后正常使用

adaaaaaa, 2020-07-28:
我直接在PE里面刻录就fail了。。。diskimage里面提示fail。。。
@Nadleeh1988 @xiexiaosheng

adaaaaaa, 2020-07-28:
你那种报错就是我前面说的第二种报错咯？我重启之后变成第一种报错。。。
@Nadleeh1988

Nadleeh1988, 2020-07-28:
> 你那种报错就是我前面说的第二种报错咯？我重启之后变成第一种报错。。。
> @Nadleeh1988

是的，是你上面写的那种少见报错 。然后敲几下回车进到GNU GRUB列表。然后只能上下选择，按其他没反应，就直接ctrl alt delete重启了，重启以后正常了

adaaaaaa, 2020-07-28:
@Nadleeh1988 嗯，我重启之后是第一种报错。。。求救。。。
[image]

Nadleeh1988, 2020-07-28:
是不是你的镜像问题，我是昨天自己编译的EFI引导的镜像

adaaaaaa, 2020-07-28:
我发现刻到U盘也不是100%能启动成功。。。直接刻录的可以。。。从ventoy里面启动就会再加载几秒之后卡住。。。
怎么那么多状况呢。。。

adaaaaaa, 2020-07-28:
嗯，我也算EFI镜像。。。以前4月的镜像也是UEFI-GPT的镜像。。。

adaaaaaa, 2020-07-28:
[配置文件](http://netcut.cn/x86_64.config)

adaaaaaa, 2020-07-29:
其实只要一行命令就可以暂时解决了。。。
是我配置有问题，还是其他哪里有问题？
@Nadleeh1988 @xiexiaosheng

qiuzhongqiang, 2020-11-08:
我编译的固件也是刻到固态U盘后不能启动，卡在SUING XHCI_HCD这个位置。百度无果。

winkee01, 2021-01-30:
请问这种问题怎么修复？我的系统是正常使用的，只是突然有一天重启后，就变这样了。
注：
我不是刻录后用不上的，我是已经正常安装正常使用了很长时间，系统一切都正常。只是突然有一天重启之后变这样的。

[image]

EnnawYang, 2021-01-30:
> 请问这种问题怎么修复？我的系统是正常使用的，只是突然有一天重启后，就变这样了。
> 注：
> 我不是刻录后用不上的，我是已经正常安装正常使用了很长时间，系统一切都正常。只是突然有一天重启之后变这样的。
> 
> [image]

不如问下eSir？

winkee01, 2021-01-30:
这个问题的原因我找到了，是因为多个外接存储器（比如多个 U 盘）插到电脑上，其中有一个 U 盘是你刻了 openwrt 用来启动，里面有 uefi 引导分区。但是其它 U 盘不是启动盘，缺乏引导分区，所以 grub 检测时报错 hd0.gpt1 找不到，这个是 grub 的 bug，跟 openwrt 关系不大，我个人认为可以升级 grub 解决（注：openwrt 中默认编译的是 grub，很多流行的 Linux 发行版用的是 grub2）

EnnawYang, 2021-01-30:
> 这个问题的原因我找到了，是因为多个外接存储器（比如多个 U 盘）插到电脑上，其中有一个 U 盘是你刻了 openwrt 用来启动，里面有 uefi 引导分区。但是其它 U 盘不是启动盘，缺乏引导分区，所以 grub 检测时报错 hd0.gpt1 找不到，这个是 grub 的 bug，跟 openwrt 关系不大，所以要解决这个问题，拔掉那些不是用来启动的 u 盘即可。另外，要解决多个 u 盘插在电脑上是不影响启动，我个人认为可以升级 grub 来根本性解决（注：openwrt 中默认编译的是 grub，很多流行的 Linux 发行版用的是 grub2）。

那您要不重新开个issue说明下具体情况看下有没有大佬修？我之前也遇到过UEFI不能启动，但传统bios无问题，U盘启动无问题

winkee01, 2021-01-30:
> 那您要不重新开个issue说明下具体情况看下有没有大佬修？我之前也遇到过UEFI不能启动，但传统bios无问题，U盘启动无问题

这确实是一个通用的问题，各种 openwrt 编译版其实都会遇到这个问题。

qiuzhongqiang, 2021-01-30:
> 这个问题的原因我找到了，是因为多个外接存储器（比如多个 U 盘）插到电脑上，其中有一个 U 盘是你刻了 openwrt 用来启动，里面有 uefi 引导分区。但是其它 U 盘不是启动盘，缺乏引导分区，所以 grub 检测时报错 hd0.gpt1 找不到，这个是 grub 的 bug，跟 openwrt 关系不大，所以要解决这个问题，拔掉那些不是用来启动的 u 盘即可。另外，要解决多个 u 盘插在电脑上是不影响启动，我个人认为可以升级 grub 来根本性解决（注：openwrt 中默认编译的是 grub，很多流行的 Linux 发行版用的是 grub2）。

你好，我试过了，就算只留这个固态U盘，我的还是启动不了。
两三年前是可以的，后来的固件就不行了，放弃了。
现在使用SD卡和读卡器使用，就是读写时明显感觉慢了，运行DOCKER好难受。

closed 2021-06-20 by coolsnowwolf