coolsnowwolf/lede #5206: odhcp6c导致DNS故障

LoogCP, 2020-07-28:
最近编译了一个固件，WAN6里面提示安装odhcp6c，用opkg安装之后（可能是版本不对），在DHCP/DNS里面分配主机静态地址的时候如果填写了ipv6后缀那么就无法联网，
具体表现为  
1. dhcp下各客户端无法解析DNS，而自带DNS设置的设备则能正常使用，
2. 软路由能联网，*ping*和*nslookup*均正常。  
**PC（网线直链软路由LAN口）国内网页无法访问，国外走$$R的网页能访问**   
删除静态地址的ipv6后缀，软路由及其下设备均能正常联网。    
关于odhcp6c这个包，我在*.config*文件里面搜索，并没有同名软件包/mod，不知道是不是在old-package里面

minlang112, 2020-07-28:
如果要使用IPV6，请在编译时候勾选Extra packages —> ipv6helper，此选项代表选择了以下四个↓
Network —> odhcp6c
Network —> odhcpd-ipv6only
LuCI —> Protocols —> luci-proto-ipv6
LuCI —> Protocols —> luci-proto-ppp

LoogCP, 2020-07-29:
> 如果要使用IPV6，请在编译时候勾选Extra packages —> ipv6helper，此选项代表选择了以下四个↓
> Network —> odhcp6c
> Network —> odhcpd-ipv6only
> LuCI —> Protocols —> luci-proto-ipv6
> LuCI —> Protocols —> luci-proto-ppp

加了ipv6-helper之后重新编译了几次固件，出现了更严重的问题
adbyby、shadowsocksr任一开启，都无法连接网络，不管国内还是国外都不行
我现在试试重新git clone 项目编译看看

LoogCP, 2020-07-31:
这两天又编译了好几个固件，发现都存在本机DNS能解析但是电脑有线连接和手机无线连接无法解析DNS的问题，必须要手动设置DNS才能上网
这个问题貌似会纵向影响到以前的固件（包括不保留配置更新固件），

LoogCP, 2020-08-03:
找到原因了，是因为设置里面有个“网络--DHCP/DNS--忽略空域名解析”被选上了，去掉勾选就正常了。

closed 2020-08-03 by LoogCP