coolsnowwolf/lede #5207: 内核和官网的不一样，无法安装软件包

share18x, 2020-07-28:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
错误：无法安装软件
ip6tables无法使用nat

### (2) 路由器型号和固件版本

A：
设备：树莓派2B
官网软件源内核：5.4.52-1
固件内核：5.4.51

### (3) 详细日志

A：

`Configuring kmod-ipt-nat6.
 Collected errors:

 satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ipt-nat6:
 kernel (= 5.4.52-1-5d03e1b9a6c93a939ae64a45d3c91ab2)`

ricksuzade-maker, 2020-07-28:
你这个是安装的内核模块，不能用这种方法来安装，需要在编译前把这个组件勾上，和固件一起编译才能用。

chr09670, 2020-07-31:
现在是否有对应的源啊？无法用opkg安装软件包耶

fanck0605, 2020-08-02:
版本一样也装不上。要 commit hash 也一样才能装上。老老实实编译吧。

mysteryman2045, 2020-08-15:
https://downloads.openwrt.org/snapshots/targets/x86/64/openwrt-x86-64.manifest 里找到官方kernel - 5.4.52-1的数字摘要  9eab9fa91f175ea5bc835e03eb11867b。去自己路由器usr/lib/opkg下把自己的status文件里的所有数字摘要全部替换成跟官方一样的。再安装跟内核相关的东西，就OK了。

mysteryman2045, 2020-08-15:
当然我这个地方只是举例，你得找到跟自己版本、内核一致的官方kernel - 5.4.52-1的数字摘要。

closed 2021-06-20 by coolsnowwolf