coolsnowwolf/lede #5217: 无法编译luci-app-vnstat

Gokou-Ruri, 2020-07-28:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：添加了`luci-app-vnstat`以后就会编译失败，然后`make menuconfig`删除`luci-app-vnstat`，再次编译就会编译成功（固件大小24.5M）。

### (2) 路由器型号和固件版本

A：Newifi D2 

### (3) 详细日志

```
Collected errors:
 * check_data_file_clashes: Package libgd wants to install file /home/bbbbbbb/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/libgd.so
        But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/bbbbbbb/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/libgd.so.3
        But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/bbbbbbb/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/libgd.so.3.0.8
        But that file is already provided by package  * libgd-full
 * opkg_install_cmd: Cannot install package luci-app-vnstat.
 * check_data_file_clashes: Package libgd wants to install file /home/bbbbbbb/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/libgd.so
        But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/bbbbbbb/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/libgd.so.3
        But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/bbbbbbb/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/libgd.so.3.0.8
        But that file is already provided by package  * libgd-full
 * opkg_install_cmd: Cannot install package libgd.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/bbbbbbb/lede'
package/Makefile:108: recipe for target '/home/bbbbbbb/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/bbbbbbb/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
```

.config文件我添加了txt后缀，不然无法上传。
[.config.txt](https://github.com/coolsnowwolf/lede/files/4991610/default.config.txt)

minlang112, 2020-07-29:
看报错内容是 luci-app-vnstat 需要安装的依赖libgd 和已有的libgd-full冲突

Gokou-Ruri, 2020-08-31:
> 看报错内容是 luci-app-vnstat 需要安装的依赖libgd 和已有的libgd-full冲突

终于查到原因了，是kodbox和vnstat需要的依赖包冲突导致的，现在不知道要怎么同时编译这两个软件包。
另外我用Actions可以编译成功，本地编译却会报这个错误。

closed 2021-06-20 by coolsnowwolf