coolsnowwolf/lede #5222: ShadowSocksR Plus+ 更新节点失败

yh1272000, 2020-07-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新订阅服务器节点失败
2020-07-29 19:50:13 订阅失败, 恢复服务
2020-07-29 19:50:12 更新失败，没有可用的节点信息
2020-07-29 19:50:12 https://xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx 获取内容为空
订阅地址没有问题，我的电脑客户端和手机能正常更新。地址在浏览器也能打开，就是不能订阅。

### (2) 路由器型号和固件版本

A：
主机型号 | Intel(R) Core(TM) i3-4010U CPU @ 1.70GHz : 2 Core 4 Thread
CPU 信息 | 1396.802 MHz +40.0°C (high = +100.0°C, crit = +100.0°C)
固件版本 | OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)
内核版本 | 5.4.51

### (3) 详细日志

A：

minlang112, 2020-07-29:
昨天一朋友遇到该问题，经测试可能存在的情况：①该订阅地址已经上墙，需要在SSR已经正常运行并使用绕过中国大陆IP模式下才能订阅；②作为所谓旁路由模式下，主路由（某型TPLINK路由AC一体机）没有设置DNS服务器，导致无法解析。

dingpengyu, 2020-07-29:
i36100U v2正常使用

dingpengyu, 2020-07-29:
连接方式：
软路由连在光猫上，redmi ac2100做维 ap。

xiaocenter, 2020-07-30:
anti-ad订阅地址更换了 现在无法更新AD数据源

yh1272000, 2020-07-30:
我手动输入了一个节点，可以正常翻墙了。设为全局模式，试用了另外两个机场，也无法订阅，看来不是订阅地址的问题。

我是从半年前用软路由的，OPENWRT作的主路由，没有旁路由。当时可以订阅，也是同一个机场，之后一直是自动更新
（也没注意是否能更新成功）。前几天升级了固件，手贱删除了节点，再点更新就无法更新节点了，只能手动输入。是这次更新固件的问题？固件是我自己编译的，.config也没有改动。

yh1272000, 2020-07-30:
> anti-ad订阅地址更换了现在无法更新AD数据源

anti-ad更换订阅地址可以更新。

xiaocenter, 2020-07-30:
> > anti-ad订阅地址更换了现在无法更新AD数据源
> 
> anti-ad更换订阅地址可以更新。

最好SSR 这个工具源码更新下   不用手动更改了

minlang112, 2020-07-30:
> > > anti-ad订阅地址更换了现在无法更新AD数据源
> > 
> > 
> > anti-ad更换订阅地址可以更新。
> 
> 最好SSR 这个工具源码更新下 不用手动更改了

已经有人在helloworld项目提交

yh1272000, 2020-07-30:
试了几十次，有一次意外更新了两个节点（一共53个）。
2020-07-30 23:10:00 重启服务成功
2020-07-30 23:10:00 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:448: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:350: in function </usr/share/shadowsocksr/subscribe.lua:297>
	/usr/share/shadowsocksr/subscribe.lua:112: in function 'processData'
	/usr/share/shadowsocksr/subscribe.lua:31: in function 'split'
	/usr/share/shadowsocksr/subscribe.lua:450: in function </usr/share/shadowsocksr/subscribe.lua:448>
2020-07-30 23:10:00 stack traceback:
2020-07-30 23:10:00 /usr/share/shadowsocksr/subscribe.lua:31: attempt to index local 'full' (a nil value)
2020-07-30 23:10:00 成功解析: ssr 节点, [BosLife] 台湾 - 中转 - 02
2020-07-30 23:10:00 成功解析: ssr 节点, [BosLife] 台湾 - 中转 - 01
2020-07-30 23:09:19 服务正在暂停

yh1272000, 2020-07-31:
已经好了。

是MWAN的问题，我是双WAN设置了负载均衡，电信的WAN口不知从何时起ping任何地址（电信DNS、BAIDU等）都ping不通，一直离线。设置了跟踪网关本身，就显示在线，更新订阅就成功了。

测试了多次都正常，真是太奇怪了。之前上网一直正常，也没发现MWAN的问题，更奇怪的是电信的光猫是可以上网的，IPTV每天都正常播放，只是ping不通。

十分感谢大家的帮忙！

-----------start-----------
2020-07-31 12:42:52 订阅更新成功
2020-07-31 12:42:52 新增节点数量: 0 删除节点数量: 0
2020-07-31 12:42:52 维持当前主服务器节点。
2020-07-31 12:42:52 成功解析节点数量: 53
2020-07-31 12:42:52 成功解析: ssr 节点, [XXXXX] 英国 - 中转 - 03
2020-07-31 12:42:52 成功解析: ssr 节点, [XXXXX] 英国 - 中转 - 02
2020-07-31 12:42:52 成功解析: ssr 节点, [XXXXX] 英国 - 中转 - 01
......
......
......
2020-07-31 12:42:04 成功解析: ssr 节点, [XXXXX] 台湾 - 中转 - 01
2020-07-31 12:42:02 服务正在暂停
------------end------------

gyj1109, 2020-08-04:
> 已经好了。
> 
> 是MWAN的问题，我是双WAN设置了负载均衡，电信的WAN口重新更改了设置，并添加了新的规则，更新订阅就成功了。
> 
> 测试了多次都成功更新，真是太奇怪了。之前上网一直正常，也没发现MWAN的问题。
> 
> 十分感谢大家的帮忙！
> 
> -----------start-----------
> 2020-07-31 12:42:52 订阅更新成功
> 2020-07-31 12:42:52 新增节点数量: 0 删除节点数量: 0
> 2020-07-31 12:42:52 维持当前主服务器节点。
> 2020-07-31 12:42:52 成功解析节点数量: 53
> 2020-07-31 12:42:52 成功解析: ssr 节点, [XXXXX] 英国 - 中转 - 03
> 2020-07-31 12:42:52 成功解析: ssr 节点, [XXXXX] 英国 - 中转 - 02
> 2020-07-31 12:42:52 成功解析: ssr 节点, [XXXXX] 英国 - 中转 - 01
> ......
> ......
> ......
> 2020-07-31 12:42:04 成功解析: ssr 节点, [XXXXX] 台湾 - 中转 - 01
> 2020-07-31 12:42:02 服务正在暂停
> ------------end------------

已解决请关闭issue，谢谢！

closed 2020-08-04 by yh1272000

caliban511, 2020-09-18:
我也遇到了这个问题，但我没有使用负载均衡之类的东西，，，这可怎么办？

cyd1st, 2020-10-19:
> 我也遇到了这个问题，但我没有使用负载均衡之类的东西，，，这可怎么办？

我也没用，但我发现运行模式里选择GFW模式，再更新重新订阅就可以了。

wuyz320, 2022-08-06:
我也有这问题，并没有得到解决啊

madneal, 2024-03-02:
> > 我也遇到了这个问题，但我没有使用负载均衡之类的东西，，，这可怎么办？
> 
> 我也没用，但我发现运行模式里选择GFW模式，再更新重新订阅就可以了。

还是不行，订阅地址应该没问题，手机里面能更新，但是软路由也不能连接手机流量哇

edgexie, 2024-04-15:
更新节点失败可能是因为软路由自身不能上网。
1. 判断软路由本身是否能上网，可以ssh 进入到软路由控制台后，使用ping 一个网站看是否能ping通。如果不通，那就需要调整lan口的连接。
2. 在网络->接口->基本设置中，一定要设置自定义的DNS服务器（我是设置了这个以后订阅地址成功的）。我的设置如下。
[image]

yongdinghui1, 2024-04-30:
> 更新节点失败可能是因为软路由自身不能上网。
>
> 1. 判断软路由本身是否能上网，可以ssh 进入到软路由控制台后，使用ping 一个网站看是否能ping通。如果不通，那就需要调整lan口的连接。
> 2. 在网络->接口->基本设置中，一定要设置自定义的DNS服务器（我是设置了这个以后订阅地址成功的）。我的设置如下。 [...]

正解，困扰了我很久的问题，终于解决了，非常感谢。。。。。。。。。。。。