coolsnowwolf/lede #5225: 统计插件 /cgi-bin/luci/admin/statistics/graph 出错

klever1988, 2020-07-30:
[image]

goldkeyber112, 2020-07-30:
主题footer权限问题，赋予你用的主题footer权限就好了。

klever1988, 2020-07-30:
> 主题footer权限问题，赋予你用的主题footer权限就好了。

[image]

一样的呀

goldkeyber112, 2020-08-02:
> > 主题footer权限问题，赋予你用的主题footer权限就好了。
> 
> [image]
> 
> 一样的呀

chmod +x /usr/lib/lua/luci/view/themes/argon/footer.htm

goldkeyber112, 2020-08-02:
> > > 主题footer权限问题，赋予你用的主题footer权限就好了。
> > 
> > 
> > [image]
> > 一样的呀
> 
> chmod +x /usr/lib/lua/luci/view/themes/argon/footer.htm

单独加footer的权限可能还有问题,如果有
你直接
chmod -R +x /usr/lib/lua/luci/view/themes/argon

klever1988, 2020-08-02:
这是argon主题代码的错误，作者已经修正

https://github.com/jerrykuku/luci-theme-argon/commit/1d0c40c85c7a3c4673ea45bfb23383d4ac6d6d67#diff-4be01be5457439e9191946ae431eef02

closed 2020-08-02 by klever1988