coolsnowwolf/lede #5226: Usb error

9000000, 2020-07-30:
Can not mount 

```
    4.899471] UBI: auto-attach mtd9
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    4.902841] ubi0: attaching mtd9
Thu Jul 30 10:56:07 2020 kern.info kernel: [    5.463458] ralink_soc_eth 1e100000.ethernet eth0: port 2 link up
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.205845] ubi0: scanning is finished
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.228045] ubi0: attached mtd9 (name "ubi", size 117 MiB)
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.233579] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.240447] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.247226] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.254177] ubi0: good PEBs: 940, bad PEBs: 0, corrupted PEBs: 0
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.260175] ubi0: user volume: 2, internal volumes: 1, max. volumes count: 128
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.267388] ubi0: max/mean erase counter: 3/1, WL threshold: 4096, image sequence number: 1593810564
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.276507] ubi0: available PEBs: 0, total reserved PEBs: 940, PEBs reserved for bad PEB handling: 20
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.285730] ubi0: background thread "ubi_bgt0d" started, PID 435
Thu Jul 30 10:56:07 2020 kern.info kernel: [    6.287806] block ubiblock0_0: created from ubi0:0(rootfs)
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    6.297248] ubiblock: device ubiblock0_0 (rootfs) set to be root filesystem
Thu Jul 30 10:56:07 2020 kern.info kernel: [    6.304211] hctosys: unable to open rtc device (rtc0)
Thu Jul 30 10:56:07 2020 kern.info kernel: [    6.316710] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
Thu Jul 30 10:56:07 2020 kern.info kernel: [    6.328299] Freeing unused kernel memory: 1288K
Thu Jul 30 10:56:07 2020 kern.warn kernel: [    6.332878] This architecture does not have kernel memory protection.
Thu Jul 30 10:56:07 2020 kern.info kernel: [    6.339311] Run /sbin/init as init process
Thu Jul 30 10:56:07 2020 kern.info kernel: [    6.394798] ralink_soc_eth 1e100000.ethernet eth0: port 1 link up
Thu Jul 30 10:56:07 2020 user.info kernel: [    6.910255] init: Console is alive
Thu Jul 30 10:56:07 2020 user.info kernel: [    6.913894] init: - watchdog -
Thu Jul 30 10:56:07 2020 user.info kernel: [    8.329353] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.457288] usbcore: registered new interface driver usbfs
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.462898] usbcore: registered new interface driver hub
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.468383] usbcore: registered new device driver usb
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    8.482548] SCSI subsystem initialized
Thu Jul 30 10:56:07 2020 kern.warn kernel: [    8.496674] xhci-mtk 1e1c0000.xhci: 1e1c0000.xhci supply vusb33 not found, using dummy regulator
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.505714] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.511085] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 1
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.526195] xhci-mtk 1e1c0000.xhci: hcc params 0x01401198 hci version 0x96 quirks 0x0000000000210010
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.535393] xhci-mtk 1e1c0000.xhci: irq 20, io mem 0x1e1c0000
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.542543] hub 1-0:1.0: USB hub found
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.546415] hub 1-0:1.0: 2 ports detected
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.551159] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.556508] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 2
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.563996] xhci-mtk 1e1c0000.xhci: Host supports USB 3.0 SuperSpeed
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.570558] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.579610] hub 2-0:1.0: USB hub found
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.583486] hub 2-0:1.0: 1 port detected
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.596620] usbcore: registered new interface driver usb-storage
Thu Jul 30 10:56:07 2020 kern.info kernel: [    8.604662] usbcore: registered new interface driver uas
Thu Jul 30 10:56:07 2020 user.info kernel: [    8.610436] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Thu Jul 30 10:56:07 2020 user.info kernel: [    8.622512] init: - preinit -
Thu Jul 30 10:56:07 2020 kern.info kernel: [    9.360535] ralink_soc_eth 1e100000.ethernet: PPE started
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    9.546877] random: jshn: uninitialized urandom read (4 bytes read)
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    9.609647] random: jshn: uninitialized urandom read (4 bytes read)
Thu Jul 30 10:56:07 2020 kern.notice kernel: [    9.787536] random: jshn: uninitialized urandom read (4 bytes read)
Thu Jul 30 10:56:07 2020 kern.warn kernel: [   11.801245] urandom_read: 4 callbacks suppressed
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   11.801259] random: procd: uninitialized urandom read (4 bytes read)
Thu Jul 30 10:56:07 2020 kern.info kernel: [   11.930153] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci-mtk
Thu Jul 30 10:56:07 2020 kern.info kernel: [   11.959260] scsi host0: uas
Thu Jul 30 10:56:07 2020 kern.err kernel: [   11.963643] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 4
Thu Jul 30 10:56:07 2020 kern.err kernel: [   11.972064] xhci-mtk 1e1c0000.xhci: @000000000f0bb2a0 0f0b0000 00000000 05000000 01058001
Thu Jul 30 10:56:07 2020 kern.err kernel: [   11.980204] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 6
Thu Jul 30 10:56:07 2020 kern.err kernel: [   11.988603] xhci-mtk 1e1c0000.xhci: @000000000f0bb2b0 0f0b0100 00000000 05000000 01078001
Thu Jul 30 10:56:07 2020 user.info kernel: [   15.971483] mount_root: loading kmods from internal overlay
Thu Jul 30 10:56:07 2020 user.info kernel: [   16.010563] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Thu Jul 30 10:56:07 2020 user.info kernel: [   16.018506] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.326282] UBIFS (ubi0:1): Mounting in unauthenticated mode
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.332182] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 609
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.420270] UBIFS (ubi0:1): recovery needed
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.572634] UBIFS (ubi0:1): recovery completed
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.577228] UBIFS (ubi0:1): UBIFS: mounted UBI device 0, volume 1, name "rootfs_data"
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.585051] UBIFS (ubi0:1): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.594951] UBIFS (ubi0:1): FS size: 97263616 bytes (92 MiB, 766 LEBs), journal size 4825088 bytes (4 MiB, 38 LEBs)
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.605360] UBIFS (ubi0:1): reserved for root: 4594000 bytes (4486 KiB)
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.611973] UBIFS (ubi0:1): media format: w4/r0 (latest is w5/r0), UUID AA1986D7-C6D0-4386-B7D9-A9AE29EFC711, small LPT model
Thu Jul 30 10:56:07 2020 user.info kernel: [   16.626472] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
Thu Jul 30 10:56:07 2020 user.info kernel: [   16.645254] block: extroot: not configured
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.649722] UBIFS (ubi0:1): un-mount UBI device 0
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.654464] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" stops
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.667077] UBIFS (ubi0:1): Mounting in unauthenticated mode
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.673015] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 612
[... cut ...]
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.841144] UBIFS (ubi0:1): reserved for root: 4594000 bytes (4486 KiB)
Thu Jul 30 10:56:07 2020 kern.notice kernel: [   16.847759] UBIFS (ubi0:1): media format: w4/r0 (latest is w5/r0), UUID AA1986D7-C6D0-4386-B7D9-A9AE29EFC711, small LPT model
Thu Jul 30 10:56:07 2020 user.info kernel: [   17.089831] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
Thu Jul 30 10:56:07 2020 user.info kernel: [   17.106606] block: extroot: not configured
Thu Jul 30 10:56:07 2020 user.info kernel: [   17.114087] mount_root: switching to ubifs overlay
```

9000000, 2020-07-30:
```
root@OpenWrtMT76:~# block info                                                  /dev/mtdblock9: UUID="1593810564" VERSION="1" TYPE="ubi"
/dev/ubiblock0_0: UUID="116f0f59-9b4c8b56-47240c1c-db7a79f9" VERSION="4.0" MOUNT="/rom" TYPE="squashfs"
/dev/ubi0_1: UUID="aa1986d7-c6d0-4386-b7d9-a9ae29efc711" VERSION="w4r0" MOUNT="/overlay" TYPE="ubifs"
root@OpenWrtMT76:~# lsusb -t                                                    /:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M
root@OpenWrtMT76:~# block info                                                  /dev/mtdblock9: UUID="1593810564" VERSION="1" TYPE="ubi"
/dev/ubiblock0_0: UUID="116f0f59-9b4c8b56-47240c1c-db7a79f9" VERSION="4.0" MOUNT="/rom" TYPE="squashfs"
/dev/ubi0_1: UUID="aa1986d7-c6d0-4386-b7d9-a9ae29efc711" VERSION="w4r0" MOUNT="/overlay" TYPE="ubifs"
root@OpenWrtMT76:~#
```

9000000, 2020-07-30:
```
32.550244] scsi 0:0:0:0: tag#0 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD IN 
[   32.557852] scsi 0:0:0:0: tag#0 CDB: opcode=0x12 12 00 00 00 24 00
[   32.564397] xhci-mtk 1e1c0000.xhci: Mismatch between completed Set TR Deq Ptr command & xHCI internal state.
[   32.574243] xhci-mtk 1e1c0000.xhci: ep deq seg = 90a042f4, deq ptr = ef6ead1c
[   32.658266] scsi host0: uas_eh_device_reset_handler start
[   32.663915] xhci-mtk 1e1c0000.xhci: Mismatch between completed Set TR Deq Ptr command & xHCI internal state.
[   32.673752] xhci-mtk 1e1c0000.xhci: ep deq seg = af90e189, deq ptr = ceea2792
[   32.834510] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-mtk
[   32.862166] scsi host0: uas_eh_device_reset_handler success
[   32.867991] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 4
[   32.876451] xhci-mtk 1e1c0000.xhci: @000000000f0bb420 0f0b0200 00000000 05000000 01058001
[   32.890895] scsi 0:0:0:0: tag#0 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD 
[   32.898249] scsi 0:0:0:0: tag#0 CDB: opcode=0x0 00 00 00 00 00 00
[   32.904441] scsi host0: uas_eh_device_reset_handler start
[   32.910061] xhci-mtk 1e1c0000.xhci: Mismatch between completed Set TR Deq Ptr command & xHCI internal state.
[   32.919910] xhci-mtk 1e1c0000.xhci: ep deq seg = 71c164cd, deq ptr = eef8f2dd
[   33.062766] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-mtk
[   33.099832] scsi host0: uas_eh_device_reset_handler success
[   33.105517] scsi 0:0:0:0: Device offlined - not ready after error recovery
[   
```

AmadeusGhost, 2020-07-31:
opkg remove kmod-usb-storage-uas

9000000, 2020-07-31:
Thank you, I changed the USB 3.0 cable to 2.0, and it worked

closed 2020-07-31 by 9000000